pyvisa/pyvisa-py #302: Error when running executable generated by pyinstaller [pyvisa-py]

gamer388, 2021-04-21:
I'm trying to build an executable from a simple python script using pyvisa-py but I'm running into error after I run the executable generated by pyinstaller.

Here what my small python code looks like

> 
> > import pyvisa as visa
> > import tkinter as tk
> > 
> > root = tk.Tk()
> > root.title("SCPI test")
> > canvas1 = tk.Canvas(root, width=200, height=100, bg='lightsteelblue2', relief='raised')
> > canvas1.pack()
> > 
> > 
> > def test_1():
> >     rm = visa.ResourceManager("@py")
> >     res_list = rm.list_resources()
> >     print('res_list :', res_list)
> >     len_list = len(res_list)
> >     print('len_list :', len_list)
> >     if not len_list:
> >         print("No equipment found.")
> >     try:
> >         inst = rm.open_resource('USB0::0x1AB1::0x0588::DS1K00005888::INSTR')
> >         print(inst.query("*IDN?"))
> >     except ValueError:
> >         print("No device found.")
> > 
> > 
> > Launch_prgm = tk.Button(text="Device detect", command=test_1, bg='green', fg='white', font=('helvetica', 12, 'bold'))
> > canvas1.create_window(100, 50, window=Launch_prgm)
> > 
> > root.mainloop()
> > 
> > When I run this code in Pycharm or by directly running the .py file from a terminal outside Pycharm it is working well. But when I build the executable using pyinstaller I got the following error.
> > 
> > Exception in Tkinter callback
> > Traceback (most recent call last):
> >   File "pyvisa/highlevel.py", line 2833, in get_wrapper_class
> >   File "importlib/__init__.py", line 126, in import_module
> >   File "<frozen importlib._bootstrap>", line 994, in _gcd_import
> >   File "<frozen importlib._bootstrap>", line 971, in _find_and_load
> >   File "<frozen importlib._bootstrap>", line 953, in _find_and_load_unlocked
> > ModuleNotFoundError: No module named 'pyvisa_py'

During handling of the above exception, another exception occurred:

```
Traceback (most recent call last):
  File "pyvisa/highlevel.py", line 2838, in get_wrapper_class
  File "importlib/__init__.py", line 126, in import_module
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 953, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'pyvisa-py'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "tkinter/__init__.py", line 1705, in __call__
  File "test_ea_psu.py", line 17, in test_1
  File "pyvisa/highlevel.py", line 3015, in __new__
  File "pyvisa/highlevel.py", line 2926, in open_visa_library
  File "pyvisa/highlevel.py", line 2849, in get_wrapper_class
ValueError: Wrapper not found: No package named pyvisa_py

Apparently I'm not the only one having problems with pyvisa and pyinstaller. Many people on github had this issue as well. https://github.com/pyvisa/pyvisa-py/issues/216

I'm using python 3.6 on Zorin OS (an ubuntu like OS). When I do python3 -m visa info I got the following info showing

python3 -m visa info
/usr/local/lib/python3.6/dist-packages/visa.py:23: FutureWarning: The visa module provided by PyVISA is being deprecated. You can replace `import visa` by `import pyvisa as visa` to achieve the same effect.

The reason for the deprecation is the possible conflict with the visa package provided by the https://github.com/visa-sdk/visa-python which can result in hard to debug situations.
  FutureWarning,
Machine Details:
   Platform ID:    Linux-5.4.0-67-generic-x86_64-with-Zorin-15-bionic
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.6.9
   Compiler:       GCC 8.4.0
   Bits:           64bit
   Build:          Jan 26 2021 15:33:00 (#default)
   Unicode:        UCS4

PyVISA Version: 1.11.3

Backends:
   ivi:
      Version: 1.11.3 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.5.2
      ASRL INSTR: Available via PySerial (3.5)
      USB INSTR: Available via PyUSB (1.1.1). Backend: libusb1
      USB RAW: Available via PyUSB (1.1.1). Backend: libusb1
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
         No module named 'gpib'
```

Also I installed pyvisa and pyvisa-py using pycharm package installer builtin function (work the same as pip3)

If I do list pip3 I got

```
Package                   Version
------------------------- ----------------------
altgraph                  0.17
apturl                    0.5.2
asn1crypto                0.24.0
Brlapi                    0.6.6
certifi                   2018.1.18
chardet                   3.0.4
chrome-gnome-shell        0.0.0
colorama                  0.4.4
command-not-found         0.3
configparser              5.0.2
crayons                   0.4.0
cryptography              2.1.4
cupshelpers               1.0
cycler                    0.10.0
dataclasses               0.8
defer                     1.0.6
defusedxml                0.7.1
distro-info               0.18ubuntu0.18.04.1
ea-psu-controller         1.1.0
et-xmlfile                1.0.1
httplib2                  0.9.2
idna                      2.6
importlib-metadata        4.0.0
iso8601                   0.1.14
keyring                   10.6.0
keyrings.alt              3.0
language-selector         0.1
launchpadlib              1.10.6
lazr.restfulclient        0.13.5
lazr.uri                  1.0.3
louis                     3.5.0
lxml                      4.6.2
m3u8                      0.8.0
macaroonbakery            1.1.3
Mako                      1.0.7
MarkupSafe                1.0
matplotlib                2.1.1
netifaces                 0.10.4
numpy                     1.13.3
oauth                     1.0.1
olefile                   0.45.1
openpyxl                  3.0.7
pexpect                   4.2.1
Pillow                    5.1.0
pip                       21.0.1
power                     1.4
protobuf                  3.0.0
psutil                    5.4.2
pycairo                   1.16.2
pycrypto                  2.6.1
pycups                    1.9.73
pygobject                 3.26.1
pyinstaller               4.3
pyinstaller-hooks-contrib 2021.1
pymacaroons               0.13.0
PyNaCl                    1.1.2
pyparsing                 2.2.0
pyRFC3339                 1.0
pyserial                  3.5
python-apt                1.6.5-ubuntu0.5-zorin1
python-dateutil           2.6.1
python-debian             0.1.32
pytz                      2018.3
pyusb                     1.1.1
PyVISA                    1.11.3
PyVISA-py                 0.5.2
pyxdg                     0.25
PyYAML                    3.12
reportlab                 3.4.0
requests                  2.18.4
requests-unixsocket       0.1.5
SecretStorage             2.3.1
selenium                  3.141.0
setuptools                56.0.0
simplejson                3.13.2
six                       1.11.0
system-service            0.3
typing-extensions         3.7.4.3
ubuntu-drivers-common     0.0.0
ufw                       0.36
urllib3                   1.22
wadllib                   1.3.2
webdriver-manager         3.3.0
wheel                     0.30.0
xkit                      0.0.0
zipp                      3.4.1
zope.interface            4.3.2
zorin-appearance          3.0
zorin-connect             1.0
zorin-exec-guard          1.0
```

I have experience using pyinstaller with other python code I have written in the past but I'm a beginner with pyvisa. I spent my whole last night trying to figure out what was the problem but I coudn't so that's why I'm asking here for help. Sorry if the same question have been posted before I searched everywhere a solution but didn't find any.
https://stackoverflow.com/questions/67168199/error-when-running-executable-generated-by-pyinstaller-pyvisa-py?noredirect=1#comment118755247_67168199
I also posted my questions about the problem on stackoverflow but got no useful answer there so I hope to get help here.

MatthieuDartiailh, 2021-04-21:
Have a look at this issue (https://github.com/pyvisa/pyvisa-py/issues/216), which describes your issue and its solution (basically adding pyvisa_py to the list of hidden files). I never used PyInstaller myself and offered to people using it to make a PR to clarify the situation but it never happened. You are welcome to make a PR to update the docs if you feel so inclined.

gamer388, 2021-04-21:
Ok I'll try this when building the executable with pyinstaller when I come back at home this evening.
But I already tried to do add pyvisa and pyvisa-py but maybe the typo wasn't exactly correct.

MatthieuDartiailh, 2021-04-21:
Note that the installed python package is `pyvisa_py` while the name of the project on PyPI is `pyvisa-py` (this follows the same convention used for example by pytest extensions).

jaydoherty, 2021-07-07:
I just posted this in #216, but in case you are still working on it, you need to add "pyvisa_py" as a hidden import. 

You can do this either with the "hidden-import" command line option:
```pyinstaller --hidden-import=pyvisa_py example.py```

Or (my preferred way), by modifying your .spec file:
```
block_cipher = None
a = Analysis(['minimal.py'],
     pathex=['/Developer/PItests/minimal'],
     binaries=None,
     datas=None,
     hiddenimports=['pyvisa_py'],   # <----- this is the key line
     hookspath=None,
     runtime_hooks=None,
     excludes=None,
     cipher=block_cipher)
pyz = PYZ(a.pure, a.zipped_data,
     cipher=block_cipher)
exe = EXE(pyz,... )
coll = COLLECT(...)
```