pyvisa/pyvisa-py #303: Absoulte beginner - some basic help with pyvisa

jlozanol, 2021-04-28:
Hi Team,

Sorry if this is not the right forum to put this question. I'm brand new to general programming and I'm trying to understand how to communicate with some devices via Python. My short term goal is to create a python program to communicate a Thorlabs Spectrometer CCS175 and I just installed the Instrumental library.

I just tried to run the list of instruments to see if the spectrometer appeared and I just got the following:

```
(base) PS C:\> python
Python 3.9.1 (default, Dec 11 2020, 09:29:25) [MSC v.1916 64 bit (AMD64)] :: Anaconda, Inc. on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import visa
>>> from instrumental import instrument, list_instruments
>>> paramsets = list_instruments()
Traceback (most recent call last):
  File "C:\Users\s3853339\Miniconda3\lib\site-packages\instrumental\drivers\__init__.py", line 677, in list_instruments
    inst_list.extend(list_visa_instruments())
  File "C:\Users\s3853339\Miniconda3\lib\site-packages\instrumental\drivers\__init__.py", line 621, in list_visa_instruments
    return list(gen_visa_instruments())
  File "C:\Users\s3853339\Miniconda3\lib\site-packages\instrumental\drivers\__init__.py", line 567, in gen_visa_instruments
    rm = visa.ResourceManager()
AttributeError: module 'visa' has no attribute 'ResourceManager'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\s3853339\Miniconda3\lib\site-packages\instrumental\drivers\__init__.py", line 678, in list_instruments
    except visa.VisaIOError:
AttributeError: module 'visa' has no attribute 'VisaIOError'
```

I then tried:

```
>>> rm = visa.ResourceManager(r'C:\WINDOWS\system32\visa64.dll')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: module 'visa' has no attribute 'ResourceManager'

>>> import pyvisa as visa
>>> rm = visa.ResourceManager(r'C:\WINDOWS\system32\visa64.dll')
>>> rm.list_resources()
()
# rm is empty, I think no device is detected...
```

I try this in both a normal cmd prompt and in an admin prompt with same results. Running Win10.

I'm not sure what I'm missing at the moment... I imagine there's something wrong with pyvisa that cannot see any devices connected to the computer?
BTW, I have also connected a NI myDAQ device that I have successfully made it work via nidaqmx library.
It seems an issue with pyvisa, maybe?

Thanks heaps for your help!

Regards,

jlozanol, 2021-04-28:
Also... I run this:

```
(base) PS C:\> pyvisa-info
Machine Details:
   Platform ID:    Windows-10-10.0.18362-SP0
   Processor:      Intel64 Family 6 Model 142 Stepping 12, GenuineIntel

Python:
   Implementation: CPython
   Executable:     c:\users\s3853339\miniconda3\python.exe
   Version:        3.9.1
   Compiler:       MSC v.1916 64 bit (AMD64)
   Bits:           64bit
   Build:          Dec 11 2020 09:29:25 (#default)
   Unicode:        UCS4

PyVISA Version: 1.11.3

Backends:
   ivi:
      Version: 1.11.3 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: National Instruments
         Spec. Version: National Instruments
      #2: C:\WINDOWS\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: National Instruments
         Spec. Version: National Instruments
```

However, this one is not working:

```
(base) PS C:\WINDOWS\system32> python -m visa info
C:\Users\s3853339\Miniconda3\python.exe: No module named visa.__main__; 'visa' is a package and cannot be directly executed
```

jlozanol, 2021-04-28:
Also, I know from NI MAX that this spectrometer has the following VISA resource name...

[image]

MatthieuDartiailh, 2021-04-28:
By default the resource manager only list INSTR resources. To get it to list all present resources, you can do `rm.list_resources('?*')`. But since you know the address you can directly pass the address to `open_resource`

jlozanol, 2021-04-28:
Thx @MatthieuDartiailh !

I tried both things and still no success:

```
>>> rm.list_resources('?*')
()
>>> test=rm.open_resource('USB0::0x1313::0x8087::M00601006::RAW')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\s3853339\Miniconda3\envs\instruments\lib\site-packages\pyvisa\highlevel.py", line 3304, in open_resource
    res.open(access_mode, open_timeout)
  File "C:\Users\s3853339\Miniconda3\envs\instruments\lib\site-packages\pyvisa\resources\resource.py", line 297, in open
    self.session, status = self._resource_manager.open_bare_resource(
  File "C:\Users\s3853339\Miniconda3\envs\instruments\lib\site-packages\pyvisa\highlevel.py", line 3232, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "C:\Users\s3853339\Miniconda3\envs\instruments\lib\site-packages\pyvisa_py\highlevel.py", line 167, in open
    sess = cls(session, resource_name, parsed, open_timeout)
  File "C:\Users\s3853339\Miniconda3\envs\instruments\lib\site-packages\pyvisa_py\sessions.py", line 323, in __init__
    self.after_parsing()
  File "C:\Users\s3853339\Miniconda3\envs\instruments\lib\site-packages\pyvisa_py\usb.py", line 81, in after_parsing
    self.interface = self._intf_cls(
  File "C:\Users\s3853339\Miniconda3\envs\instruments\lib\site-packages\pyvisa_py\protocols\usbraw.py", line 37, in __init__
    super(USBRawDevice, self).__init__(vendor, product, serial_number, **kwargs)
  File "C:\Users\s3853339\Miniconda3\envs\instruments\lib\site-packages\pyvisa_py\protocols\usbtmc.py", line 199, in __init__
    raise ValueError("No device found.")
ValueError: No device found.
```

MatthieuDartiailh, 2021-04-28:
Since you are running on windows with NI-VISA installed you have no reason to run through pyvisa-py do not pass any argument to the ResourceManager and it will pick the NI-VISA which should be able to connect since NI-MAX manage to do so.

jlozanol, 2021-04-28:
Legend!
Thanks so much @MatthieuDartiailh for your help!

Regards,

jlozanol, 2021-04-29:
Hey @MatthieuDartiailh,

Sorry again, re this same device. After opening it I'm not able to get some information form the spectrometer:

```
test.read_termination = '\n'
>>> test.write_termination = '\n'
>>> test.query('*IDN?')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\s3853339\Miniconda3\envs\instruments\lib\site-packages\pyvisa\resources\messagebased.py", line 638, in query
    self.write(message)
  File "C:\Users\s3853339\Miniconda3\envs\instruments\lib\site-packages\pyvisa\resources\messagebased.py", line 197, in write
    count = self.write_raw(message.encode(enco))
  File "C:\Users\s3853339\Miniconda3\envs\instruments\lib\site-packages\pyvisa\resources\messagebased.py", line 157, in write_raw
    return self.visalib.write(self.session, message)[0]
  File "C:\Users\s3853339\Miniconda3\envs\instruments\lib\site-packages\pyvisa\ctwrapper\functions.py", line 2796, in write
    ret = library.viWrite(session, data, len(data), byref(return_count))
  File "C:\Users\s3853339\Miniconda3\envs\instruments\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 222, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
  File "C:\Users\s3853339\Miniconda3\envs\instruments\lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

```

Tracking what was performed on #397 and #513 but I don't get what could it be wrong. Tried with NI VISA Test Panel although I think it's sending me weird information after the `*IDN?`

[image]

Just checked in the CCS175 manual and unfortunately cannot see any command list or parameter list to follow along.

Thoughts?

Thanks again!

MatthieuDartiailh, 2021-04-29:
USB RAW devices do not use the usual text based interface but a lower level vendor specific protocol. However Throlabs does not describe in the manual, since they want you to use their library (see section 5 of the manual). You can have a look at the C source code of the NI-VISA based driver (C:\Program Files\IVI Foundation\VISA\WinNT\ThorlabsCCSseries\TLCCS.c according to the manual) and infer the protocol from there. If you share the file I may be able to provide some guidance.

jlozanol, 2021-04-30:
Attaching the file as txt.

Thanks heaps!

[TLCCS.txt](https://github.com/pyvisa/pyvisa-py/files/6402293/TLCCS.txt)

MatthieuDartiailh, 2021-04-30:
I will try to come back to you early next week. Ping me if you do not hear from me.

jlozanol, 2021-05-01:
Hi @MatthieuDartiailh,

No need for that anylonger, I have successfully communicated with the spectrometer via Instrumental library.
Thanks heaps for all the help!

closed 2021-05-01 by jlozanol