pyvisa/pyvisa-py #305: Timeout error after hours of operation

n-benchoubane, 2021-07-23:
Hello !

I recently started operating the Rigol instruments through remote control with _pyvisa_ library. Yesterday I left the Rigol multimeter running for over 22 hours and it worked wonderfully. Then all of a sudden I got a timeout error in the query to measure the voltage. What I found strange about the timeout error is that my laptop no longer recognized the instrument, which did not happen with previous timeout error I experienced.
I had to manually restart the instrument. I typically press the local key to exit remote control mode, but that didn't help. 

Has anyone had a similar situation? Are there limitations to the maximum remote control time?

MatthieuDartiailh, 2021-07-25:
Hi, 
Could you post the output of pyvisa info, the model of your instrument and possibly your script ? Since the instrument becomes completely non-responsive you may have it a bug in the firmware. It could be worth checking if there is a firmware update available.

n-benchoubane, 2021-07-26:
Hello!

### pyvisa-info

This is the output for the pyvisa info
```
Machine Details:
   Platform ID:    Windows-10-10.0.19041-SP0
   Processor:      AMD64 Family 23 Model 24 Stepping 1, AuthenticAMD
Python:
   Implementation: CPython
   Executable:     C:\Users\choub\anaconda3\python.exe
   Version:        3.8.8
   Compiler:       MSC v.1916 64 bit (AMD64)
   Bits:           64bit
   Build:          Apr 13 2021 15:08:03 (#default)
   Unicode:        UCS4
PyVISA Version: 1.11.3
Backends:
   ivi:
      Version: 1.11.3 (bundled with PyVISA)
      #1: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: National Instruments
         Spec. Version: National Instruments
      #2: C:\Windows\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: National Instruments
         Spec. Version: National Instruments
```

### Instrument

- Rigol Multimeter DM3068
- Connection USB
- [Documentation] ftp://host-212-18-29-48.customer.m-online.net/ftp/pub/rigol/programming_guides/DMM/DM3000_ProgrammingGuide_EN.pdf

### Code
```
file : dm3000.py
```

```
from pyvisa import *
import time

_delay  = 0.5           #small delay 500ms sec

class DM3068:
    def __init__(self, usb_or_serial='USB0'):
        try:
            self.rm = ResourceManager()
            self.instrument_list = self.rm.list_resources()
            self.address = [elem for elem in self.instrument_list if (elem.find('USB') != -1 and elem.find(
                usb_or_serial) != -1)]  # Search a instrument with USB and serial number in the instrument list

            if self.address.__len__() == 0:
                self.status = "Not Connected"
                print("Could not connect to device")
            else:
                self.address = self.address[0]
                print("Address ", self.address)
                self.device = self.rm.open_resource(self.address)
                print("Connected to " + self.address)
                self.status = "Connected"
                self.connected_with = 'USB'

        except VisaIOError:
            self.status = "Not Connected"
            print("PyVISA is not able to find any devices")

    def configure_voltage_range(self, range):
        command = ':CONFigure:VOLTage:DC %s' % range
        self.device.query(command)
        time.sleep(_delay)

    def setup_voltage_mode(self, mode):
        command = ':FUNCtion:VOLTage:%s' % mode
        self.device.query(command)
        time.sleep(_delay)

    def setup_current_mode(self, mode):
        command = ':FUNCtion:CURRent:%s' % mode
        self.device.query(command)
        time.sleep(_delay)

    def measure_voltage(self, mode):
        # define a MEASURE VOLTAGE function
        command = ':MEASure:VOLTage:%s?' % mode
        volt = self.device.query(command)
        volt = float(volt)
        time.sleep(_delay)
        return volt

    def measure_current(self, mode):
        # define a MEASURE CURRENT function
        command = ':MEASure:CURRent:%s?' % mode
        curr = self.device.query(command)
        curr = float(curr)
        time.sleep(_delay)
        return curr
```

```
file : main.py
```
The user inputs the mode and range for the voltage mode for the multimeter as well as sampling time. The script also writes the data into a file.
```
...
DMM = DM3068()
DMM.setup_voltage_mode(args.mode)
DMM.configure_voltage_range(args.range)

while True:
        voltage = DMM.measure_voltage(args.mode)
        time.sleep(sampling)
```
I'm working on a script to scan the devices available with last status state and last time connected. 

As I mentioned, the error appeared after over 22 hours of operation all of sudden. I will redo the experiment today leaving it all night and add to this thread if it appears again. I reckon I should add a try-except clause for the timeout_error. Any advice is appreciated! 

One thing I failed to mentioned after I reset the instrument is that it didn't connect right away, it took about five minutes before it started to measure the voltage.

Thank you truly for the fast reply!

MatthieuDartiailh, 2021-07-26:
Adding a try except is reasonable but if somehow the instrument is in a broken state (you mentioned having to power cycle the instrument) it may not help. If your instrument support communication over TCPIP or serial it may be worth checking if you see the same issue using those since if it is a firmware issue it may specific to the transport layer.

n-benchoubane, 2021-07-26:
Thank you very much! I will try that and let you know.

closed 2021-08-17 by n-benchoubane