pyvisa/pyvisa-py #306: pyvisa VI_ERROR_LIBRARY_NFOUND

SashaBrownsberger, 2021-09-22:
Hello! 
I am attempting to use pyvisa to communicate with some KeySight instruments for some data measurements.  I am running on a new Windows 10 machine. 

I am encountering a bug with pyvisa that I have not seen before. When I attempt to open a ResourceManager, I see an error about a missing library that is required for VISA (though I am not told what the missing library is).  Here are the steps required to produce the code (commands are run from python environment in Command Prompt): 

```
import pyvisa as visa 
rm = visa.ResourceManager() 
```
The following error is then thrown: ```

Traceback (most recent call last): 
    File "<stdin>", line 1, in <module> 
    File "C:\Users\LabUser\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa\highlevel.py", line 3024, in __new__ 
         obj.session, err = visa_library.open_default_resource_manager() 
    File "C:\Users\LabUser\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1871, in open_default_resource_manager 
       ret = library.viOpenDefaultRM(byref(session)) 
    File "C:\Users\LabUser\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 222, in _return_handler 
      return self.handle_return_value(session, ret_value) # type: ignore 
    File "C:\Users\LabUser\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value 
        raise errors.VisaIOError(rv) 
pyvisa.errors.VisaIOError: VI_ERROR_LIBRARY_NFOUND (-1073807202): A code library required by VISA could not ve located or loaded. 
```

This does not appear to be a commonly encountered error.  Online searches don't turn up much.  Any help would be much appreciated.

SashaBrownsberger, 2021-09-22:
Hi again, 
I restarted my machine and this issue appears to have been solved. 

I do not know why.  

What is the appropriate protocol here, given I accidentally solved my own query with a hard restart?  Should we leave this up for a future user who encounters the same issue, or delete it as it may have been a fluke.  

Regardless, I apologize for asking what turned out to be a self-solving problem.

closed 2021-09-22 by SashaBrownsberger

MaheshTidke1, 2023-03-14:
Hi All,
I am also facing same issue.

import pyvisa
rm = pyvisa.ResourceManager()
rm.list_resources()
inst = rm.open_resource('GPIB0::1::INSTR')
print(inst.query("*IDN?"))

Traceback (most recent call last):
  File "C:\Users\MaheshNamdevTidke\ROHM Co., Ltd\Documenets\Lab_automation_project\Python_coding\Oscilloscop and python code\Instruments test code.py", line 6, in <module>
    inst = rm.open_resource('GPIB0::1::INSTR')
  File "C:\Users\MaheshNamdevTidke\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\pyvisa\highlevel.py", line 3247, in open_resource
    info = self.resource_info(resource_name, extended=True)
  File "C:\Users\MaheshNamdevTidke\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\pyvisa\highlevel.py", line 3174, in resource_info
    ret, err = self.visalib.parse_resource_extended(self.session, resource_name)
  File "C:\Users\MaheshNamdevTidke\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\pyvisa\ctwrapper\functions.py", line 2077, in parse_resource_extended
    ret = library.viParseRsrcEx(
  File "C:\Users\MaheshNamdevTidke\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\pyvisa\ctwrapper\highlevel.py", line 226, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
  File "C:\Users\MaheshNamdevTidke\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_LIBRARY_NFOUND (-1073807202): A code library required by VISA could not be located or loaded.

please help to solve this issue.?
in USB and LAN working fine...only GPIB connection this issue.
I am using:
-Keysight 82357B USB/Gpib Interface High-Speed USB 2.0
-Keysight IO suit
-Python 3.11
-NI MAX
-window 10 64 bit

MatthieuDartiailh, 2023-03-14:
You are using a Keysight adapter, and since you have NIMax, I assume you have both NI VISA and Keysight VISA installed. 

The problem is likely that you are implicitly using NI-VISA, which cannot use your adapter. The easiest solution is to use Keysight VISA conflict manager. To do so open Keysight Connection expert. Then, in the tools menu in the menubar, select VISA conflict manager. Finally, select Keysight for GPIB resources (or simply disble NI-VISA).

Hope this helps.

MaheshTidke1, 2023-03-16:
Hi MattieuDartiailh ,
I tried you suggestion, but I am getting same issue...

Thanks

MatthieuDartiailh, 2023-03-16:
Can you post the output of `pyvisa-info` ?

MatthieuDartiailh, 2023-03-16:
You need to run this command in the terminal not in the Python interpreter.

MatthieuDartiailh, 2023-03-16:
Windows command prompt

MaheshTidke1, 2023-03-17:
C:\Users\MaheshNamdevTidke> pyvisa-info
'pyvisa-info' is not recognized as an internal or external command,
operable program or batch file.

C:\Users\MaheshNamdevTidke>pip -info

Usage:
  C:\Users\MaheshNamdevTidke\AppData\Local\Microsoft\WindowsApps\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\python.exe -m pip <command> [options]

no such option: -i

MatthieuDartiailh, 2023-03-17:
Try `pip list`. If you have pyvisa installed you should have pyvisa-info

MaheshTidke1, 2023-03-17:
C:\Users\MaheshNamdevTidke>pip list
Package            Version
------------------ ---------
bcrypt             4.0.1
certifi            2022.12.7
cffi               1.15.1
charset-normalizer 3.1.0
contourpy          1.0.7
cryptography       39.0.2
cycler             0.11.0
fonttools          4.38.0
future             0.18.3
idna               3.4
instrument         0.6.0
keithley2600       2.0.2
keithleygui        1.2.1
kiwisolver         1.4.4
matplotlib         3.7.0
netmiko            4.1.2
ntc-templates      3.2.0
numpy              1.24.2
packaging          23.0
pandas             1.5.3
paramiko           3.1.0
Pillow             9.4.0
Pint               0.20.1
pip                23.0.1
pycparser          2.21
PyMeasure          0.11.1
PyNaCl             1.5.0
pyparsing          3.0.9
PyQt5              5.15.9
PyQt5-Qt5          5.15.2
PyQt5-sip          12.11.1
pyqtgraph          0.13.1
pyserial           3.5
PyTektronixScope   0.2.1
python-dateutil    2.8.2
python-dotenv      1.0.0
pytz               2022.7.1
PyVISA             1.13.0
PyVISA-py          0.6.3
PyYAML             6.0
repr               0.3.1
requests           2.28.2
scp                0.14.5
six                1.16.0
tenacity           8.2.2
textfsm            1.1.2
typing_extensions  4.5.0
urllib3            1.26.14
visa               1.0.0
wheel              0.38.4

MatthieuDartiailh, 2023-03-17:
Ok that's definitively weird. Can you try re-installing pyvisa ?