pyvisa/pyvisa-py #310: [COM] Communication issue with COM port using Pyvisa

cambronnem, 2021-12-06:
Instrument details
------------------
* Model: serial COM port 
* Communication: TCPIP, GPIB, ....
* Link to the documentation (if available):

Output of `pyvisa-info`
-----------------------

C:\Users\sesa627732\Documents\SXWS_SMARTX\CI_System\CI_System_ASP\User_Interface_Test_Cases>pyvisa-info
Machine Details:
   Platform ID:    Windows-10-10.0.19043-SP0
   Processor:      Intel64 Family 6 Model 158 Stepping 10, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Users\sesa627732\AppData\Local\Programs\Python\Python39\python.exe
   Version:        3.9.9
   Compiler:       MSC v.1929 64 bit (AMD64)
   Bits:           64bit
   Build:          Nov 15 2021 18:08:50 (#tags/v3.9.9:ccb0e6a)
   Unicode:        UCS4

PyVISA Version: 1.11.1

Backends:
   ivi:
      Version: 1.11.1 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 64
         Could not get more info:
            VI_ERROR_INV_OBJECT (-1073807346): The given session or object reference is invalid.
      #2: C:\WINDOWS\system32\visa64.dll:
         found by: auto
         bitness: 64
         Could not get more info:
            VI_ERROR_INV_OBJECT (-1073807346): The given session or object reference is invalid.
[test.txt](https://github.com/pyvisa/pyvisa-py/files/7662226/test.txt)
[test.txt](https://github.com/pyvisa/pyvisa-py/files/7662228/test.txt)

MatthieuDartiailh, 2021-12-06:
This is typical of a corrupted VISA installation I suggest you uninstall and reinstall whatever is your VISA distribution.

cambronnem, 2021-12-06:
Thanks...now it working

closed 2021-12-06 by cambronnem