pyvisa/pyvisa-py #312: [COM] Communication issue with LeCroy WaveRunner oscilloscope using USB

SengerM, 2021-12-22:
Instrument details
------------------
* Model: LeCroy WaveRunner 9254M
* Communication: USB
* Link to the documentation (if available):

Output of `pyvisa-info`
-----------------------
```
Machine Details:
   Platform ID:    Linux-5.11.0-43-generic-x86_64-with-glibc2.29
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.8.10
   Compiler:       GCC 9.3.0
   Bits:           64bit
   Build:          Nov 26 2021 20:14:08 (#default)
   Unicode:        UCS4

PyVISA Version: 1.11.3

Backends:
   ivi:
      Version: 1.11.3 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.5.2
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named 'serial'
      USB INSTR: Available via PyUSB (1.2.1). Backend: libusb1
      USB RAW: Available via PyUSB (1.2.1). Backend: libusb1
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
         No module named 'gpib
```

Description
-----------------------

I have communicated with this same oscilloscope from a different Ubuntu 20.04 machine in the past (things were set up about an year ago). Now I am trying to do it in this new machine but I am getting `usb.core.USBError: [Errno 32] Pipe error` when I try to open the connection. This is my code:
```
import pyvisa

rm = pyvisa.ResourceManager()
print(rm.list_resources())

keithley = rm.open_resource('USB0::1510::9328::04481179::0::INSTR')
print(keithley.query('*IDN?'))

lecroy = rm.open_resource('USB0::1535::4131::4751N40408::0::INSTR')
print(lecroy.query('*IDN?'))
```
and the output
```
('USB0::1510::9328::04481179::0::INSTR', 'USB0::1535::4131::4751N40408::0::INSTR')
KEITHLEY INSTRUMENTS,MODEL 2470,04481179,1.6.8d

Traceback (most recent call last):
  File "deleteme.py", line 9, in <module>
    lecroy = rm.open_resource('USB0::1535::4131::4751N40408::0::INSTR')
  File "/home/tct/.local/lib/python3.8/site-packages/pyvisa/highlevel.py", line 3304, in open_resource
    res.open(access_mode, open_timeout)
  File "/home/tct/.local/lib/python3.8/site-packages/pyvisa/resources/resource.py", line 297, in open
    self.session, status = self._resource_manager.open_bare_resource(
  File "/home/tct/.local/lib/python3.8/site-packages/pyvisa/highlevel.py", line 3232, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "/home/tct/.local/lib/python3.8/site-packages/pyvisa_py/highlevel.py", line 167, in open
    sess = cls(session, resource_name, parsed, open_timeout)
  File "/home/tct/.local/lib/python3.8/site-packages/pyvisa_py/sessions.py", line 323, in __init__
    self.after_parsing()
  File "/home/tct/.local/lib/python3.8/site-packages/pyvisa_py/usb.py", line 81, in after_parsing
    self.interface = self._intf_cls(
  File "/home/tct/.local/lib/python3.8/site-packages/pyvisa_py/protocols/usbtmc.py", line 305, in __init__
    self._enable_remote_control()
  File "/home/tct/.local/lib/python3.8/site-packages/pyvisa_py/protocols/usbtmc.py", line 311, in _enable_remote_control
    self.usb_dev.ctrl_transfer(
  File "/home/tct/.local/lib/python3.8/site-packages/usb/core.py", line 1082, in ctrl_transfer
    ret = self._ctx.backend.ctrl_transfer(
  File "/home/tct/.local/lib/python3.8/site-packages/usb/backend/libusb1.py", line 893, in ctrl_transfer
    ret = _check(self.lib.libusb_control_transfer(
  File "/home/tct/.local/lib/python3.8/site-packages/usb/backend/libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 32] Pipe error
```
As seen, the communication with the Keithley works fine but fails with the oscilloscope.

SengerM, 2021-12-29:
I changed the oscilloscope to a different computer running Ubuntu 20.04 too, same configuration as the original. When I try to connect with the oscilloscope, the same error occurs. Now, with more time, I installed the NI-VISA library following [the instructions for Ubuntu](https://www.ni.com/en-us/support/documentation/supplemental/18/downloading-and-installing-ni-driver-software-on-linux-desktop.html). So now in this computer `pyvisa-info` gives this:
```
/home/myself/local/lib/python3.8/site-packages/gpib_ctypes/gpib/gpib.py:54: UserWarning: GPIB library not found. Please manually load it using _load_lib(filename). All GPIB functions will raise OSError until the library is manually loaded.
  warnings.warn(message)
Machine Details:
   Platform ID:    Linux-5.4.0-91-generic-x86_64-with-glibc2.29
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.8.10
   Compiler:       GCC 9.3.0
   Bits:           64bit
   Build:          Nov 26 2021 20:14:08 (#default)
   Unicode:        UCS4

PyVISA Version: 1.11.3

Backends:
   ivi:
      Version: 1.11.3 (bundled with PyVISA)
      #1: /usr/lib/x86_64-linux-gnu/libvisa.so.21.0.0:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: National Instruments
         Spec. Version: National Instruments
   py:
      Version: 0.5.2
      ASRL INSTR: Available via PySerial (3.5)
      USB INSTR: Available via PyUSB (1.2.1). Backend: libusb1
      USB RAW: Available via PyUSB (1.2.1). Backend: libusb1
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 
```
After this, the following strange sequence of events happened:

1.  I modified the first line of the script to use the NI backend: `rm = pyvisa.ResourceManager('@ivi')`. Then I run this script and obtained 
```
Traceback (most recent call last):
  File "test_lecroy_the_new_one.py", line 10, in <module>
    osc = rm.open_resource('USB0::1535::4131::4751N40408::0::INSTR')
  File "/home/sengerm/.local/lib/python3.8/site-packages/pyvisa/highlevel.py", line 3304, in open_resource
    res.open(access_mode, open_timeout)
  File "/home/sengerm/.local/lib/python3.8/site-packages/pyvisa/resources/resource.py", line 297, in open
    self.session, status = self._resource_manager.open_bare_resource(
  File "/home/sengerm/.local/lib/python3.8/site-packages/pyvisa/highlevel.py", line 3232, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "/home/sengerm/.local/lib/python3.8/site-packages/pyvisa/ctwrapper/functions.py", line 1851, in open
    ret = library.viOpen(
  File "/home/sengerm/.local/lib/python3.8/site-packages/pyvisa/ctwrapper/highlevel.py", line 222, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
  File "/home/sengerm/.local/lib/python3.8/site-packages/pyvisa/highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_RSRC_BUSY (-1073807246): The resource is valid, but VISA cannot currently access it.
```
Also noted that `rm.list_resources()` is now returning an empty tuple, i.e. is not seeing the oscilloscope.
2. Came back to `rm = pyvisa.ResourceManager('@py')`. As expected, the original error occurs. 
3. Now the surprising part, just because I was bored in the lab I went back to `rm = pyvisa.ResourceManager('@ivi')` and now it is working without problems.

All this happened without restarting the computer or the oscilloscope in the middle. It just happened within 30 seconds or so I was doing this random tests.

I am just reporting this here in case it is of use to somebody, be it the developers (thank you BTW) or someone running into a similar issue (me in the future?).

SengerM, 2022-02-08:
Just want to add a comment on this issue now that I have been dealing with it for a while. The problem persists, but there is a workaround. First of all, the switch between `@py`  and `@ivi` that I described before still has to be done every now and then (not sure what triggers this). To forget about it now I am connecting to the oscilloscope in this way:

```python
try:
	oscilloscope = pyvisa.ResourceManager('@ivi').open_resource(args.oscilloscope_visa_name)
except pyvisa.errors.VisaIOError:
	try:
		pyvisa.ResourceManager('@py').open_resource(args.oscilloscope_visa_name) # This I already know it won't work, but it triggers something that makes the `@ivi` to work.
	except:
		pass
	oscilloscope = pyvisa.ResourceManager('@ivi').open_resource(args.oscilloscope_visa_name) # Now this works. Don't ask me.
```
With this weird connection mechanism, it works most of the times. There is one more thing, and it is that sometimes the connection still fails due to a *busy resource error* or so (sorry I don't have it right now) and the solution to this is to go to the oscilloscope and navigate to `Utilities` → `Utilities setup` → `Remote` and here just click in `Off` and then back again to `USBTMC`. I have no idea if this fixes something within the oscilloscope or restarts some connection within Linux, but after this it works.

mwcmwc12, 2022-02-25:
Just to chime in, we are seeing this error with 2 of our Lecroy scopes as well. It won't even open resource at all and faces the same pipe error.

Models are: Lecroy HDO6104 and Lecroy WaveRunning 8000HD

MatthieuDartiailh, 2022-02-25:
Can you have a look at the discussion in https://github.com/pyvisa/pyvisa-py/issues/256 and see if any of the investigated solution works in your case ? The more feedback we get the better.