pyvisa/pyvisa-py #314: [COM] Communication issue with Keysight VXI systems using PCI-to-1394 adapter in Ubuntu 18.04

janetmayle, 2022-01-06:
I'm sorry to disturb you.  I'd like to ask whether I can use pyvisa-py to communicate with Keysight VXI systems using PCI-to-1394 adapter in Ubuntu 18.04?

If it does, please give me some guidance, such as some lib or driver that must be installed ......

If not, which pyvisa backend should I use? I notice that Keysight IO Libraries Suite do not support VXI in linux, see https://www.keysight.com/us/en/lib/software-detail/computer-software/io-libraries-suite-downloads-2175637.html.

Thank you very much for your time.

MatthieuDartiailh, 2022-01-07:
The short story is that no pyvisa-py does not support VXI devices.

I have personally never encountered such a class of device and I have no idea how if VISA implemented for them. If you figure it out and want to contribute your work back you are welcome but I do not have the bandwidth to help with that.

janetmayle, 2022-01-10:
Thanks for your reply. 

I'm a little surprised about  you said "I have personally never encountered such a class of device ". Maybe it's because I didn't describe it clearly. I want to redescribe in detail what I want to do.

My OS is Ubuntu 18.04. I want to control several instruments such as digital multimeter.  These instruments are placed in VXI mainframe backplane. The slot 0 of the mainframe is a device called Keysight E8491B which used to link the VXI mainframe backplane to the IEEE 1394 serial bus. And finally I use a PCI-to-1394 adapter to link my PC to Keysight E8491B.