pyvisa/pyvisa-py #316: [COM] Cannot set flow_control to none for serial device Gwinstek PST3202 DC Power Supply

mwcmwc12, 2022-01-21:
Instrument details
------------------
* Model: Gwinstek PST3202
* Communication: Serial
* Link to the documentation (if available): https://assets.testequity.com/te1/Documents/pdf/PST-PSS-PSH-Programming-Manual.pdf

Output of `pyvisa-info`
-----------------------
Machine Details:
   Platform ID:    Linux-5.13.0-27-generic-x86_64-with-glibc2.29
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.8.10
   Compiler:       GCC 9.3.0
   Bits:           64bit
   Build:          Nov 26 2021 20:14:08 (#default)
   Unicode:        UCS4

PyVISA Version: 1.11.3

Backends:
   ivi:
      Version: 1.11.3 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.5.2
      ASRL INSTR: Available via PySerial (3.5)
      USB INSTR: Available via PyUSB (1.2.1). Backend: libusb1
      USB RAW: Available via PyUSB (1.2.1). Backend: libusb1
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
         No module named 'gpib'

Hi I am running into a weird issue with setting flow_control on this Power Supply. I am using pyvisa-py as the backend.
run this simple code:

```
import pyvisa
from pyvisa.constants import ControlFlow
rm = pyvisa.ResourceManager('@py')
instr = rm.open_resource(
    'ASRL/dev/ttyUSB0::INSTR'
)
instr.flow_control = ControlFlow.none
print(instr.query("*IDN?"))
```

It throws this error:
Traceback (most recent call last):
  File "simpleserialtest.py", line 7, in <module>
    instr.flow_control = ControlFlow.none
  File "/home/xxx/.local/lib/python3.8/site-packages/pyvisa/attributes.py", line 184, in __set__
    instance.set_visa_attribute(
  File "/home/xxx/.local/lib/python3.8/site-packages/pyvisa/resources/resource.py", line 393, in set_visa_attribute
    return self.visalib.set_attribute(self.session, name, state)
  File "/home/xxx/.local/lib/python3.8/site-packages/pyvisa_py/highlevel.py", line 658, in set_attribute
    return self.handle_return_value(
  File "/home/xxx/.local/lib/python3.8/site-packages/pyvisa/highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_NSUP_ATTR_STATE (-1073807330): The specified state of the attribute is not valid, or is not supported as defined by the object.

However, if I set it to xon_xoff (which the instrument doesn't support) no errors occur. I have tried to separately connect to the instrument using Putty with flow control set to none and it works.

Any suggestions?

MatthieuDartiailh, 2022-01-21:
The offending line is at https://github.com/pyvisa/pyvisa-py/blob/main/pyvisa_py/serial.py#L421

0 should be considered a valid value. Care to make a PR ?

MatthieuDartiailh, 2022-01-21:
Also pyvisa should use an IntFlag in place of an Ingenue for this attribute. I never got the chance to fix it and I cannot easily test changes affecting serial resources so if you want to have a look feel free to do so.

mwcmwc12, 2022-01-22:
Sure, I will go ahead and make the changes. Just curious, without looking into it too deeply, why is flow_control's setting not as straight forward as some of the other attributes like parity where parity is just assigns the enum, while flow_control has a bitwise operation AND before the assignment.

MatthieuDartiailh, 2022-01-22:
Because the VISA standard says it is actually possible to set multiple ones at the same time and hence behave like a flag. I do not have the VISA reference handy but on can easily find it on the IVI foundation website.

mwcmwc12, 2022-01-22:
Learn something new everyday. Didn't realize you can even mix flow control since I personally never ran into an instrument that had that kind of specification. Thanks for the explanation.

Let me do a quick check in the lab on Monday with these changes and I will issue a PR if things go well.

MatthieuDartiailh, 2022-01-22:
Sounds good.

MatthieuDartiailh, 2022-01-23:
If you have the time you can also change IntEnum to IntFlag at https://github.com/pyvisa/pyvisa/blob/main/pyvisa/constants.py#L1080. It should be trivial but having somebody test it is always nice.

commit 2dd19aae (2022-01-28): Fix flow control setting issue, fix #316 (#317)

* Fix flow control setting issue, fix #316

Flow control for serial device cannot be set to none as
0 or ControlFlow.none is treated as an invalid attribute.
Changed to "0 <="

* Added changelog entry
closed 2022-01-28 by MatthieuDartiailh