pyvisa/pyvisa-py #319: list_resources does not change

Saulliu00, 2022-03-18:
I have a Keysight 34470A, I connect it by LAN. (IP is 10.10.11.61)
I also have a Agilent E3631A, I connect it by RS232. (COM 7)

I tried connect both devices, only RS232, only LAN, or neither.
my `list_resources` always show me the same thing.

`import pyvisa`
`import os`
`from pyvisa import constants`
`from pyvisa.constants import StopBits, Parity`

`rm = pyvisa.ResourceManager()`
`print(rm.list_resources())`

The console always shows me:
>('TCPIP0::10.10.11.61::inst0::INSTR',)

Could anyone help what is going on?

MatthieuDartiailh, 2022-03-18:
Can you post the output of pyvisa-info ? The behavior of this function is quite dependent on the backend and on the exact implementation of VISA for the ivi backend. For example Keysight will not scan for TCPIP devices and will not forget USB devices automatically.

Saulliu00, 2022-03-18:
Hi Matthieu,

So far, I just checked the list resources, so I didn't open/close any
COM/LAN, but I do have another keysight software run LAN to my keysight
device earlier.
I don't know it impact or not. By the way, I just posted the pyvisa-info
below.
Thank you so much for your help.
[image: image.png]

Best regards,
Saul

MatthieuDartiailh, 2022-03-18:
I cannot see the image. Try copy pasting the text instead.

Saulliu00, 2022-03-20:
Hi Matthieu,

The info is below:

OTUT-RSTAAd I CH(25e0)
Machine Details:
Platform ID:
Windows-10-10.0.18363-SPO
Processor:
Inte164 Family 6 Model 158 Stepping 13,GenuineIntel
Python:
Implementation: CPython
Executable:
C:\Users\Liusa\Anaconda3\python.exe
Version:
3.8.12
Compiler:
MSC v.1916 64 bit (AMD64)
Bits:
64bit
Build:
Oct 12 2021 03:01:40 (#default)
Unicode:
UCS4
PyVISA Version: 1.11.3
Backends:
ivi:
Version: 1.11.3 (bundled with PyVISA)
＃1:C:\WINDOWS\system32\visa32.d11:
found by: auto
bitness: 64
Vendor:Keysight Technologies
Impl. Version: Keysight Technologies
Spec.Version: Keysight Technologies
＃2:C:\WINDOWS\system32\visa64.d11:
found by: auto
bitness: 64
Vendor: Keysight Technologies
Imp1. Version: Keysight Technologies
Spec.Version: Keysight Technologies
py:
Version: 0.5.2
ASRL INSTR:Available via PySerial (3.5)
USB INSTR: Available via PyUSB (1.2.1). Backend:libusb1
USB RAW:Available via PyUSB (1.2.1). Backend: libusb1
TCPIP INSTR: Available
TCPIP SOCKET:Available
GPIB INSTR:
Please install linux-gpib (Linux) or gpib-ctypes(Windows,Linux)
o use this resource type. Note that installing gpib-ctypes will give you ac
ess to a broader range of funcionality.
No module named 'gpib'

Best regards,
Saul

MatthieuDartiailh, 2022-03-21:
For serial I do not know of the top of my head, but for TCPIP Keysight VISA does not perform any automatic discovery. Once you add manually your device in Keysight connection Expert it will show up in PyVISA.

Saulliu00, 2022-03-21:
Hi Mattieu,

I think I know the issue. I use HTML5 UI to control the instrument
sometimes, and I also use python code sometimes.
Regardless TCPIP or serial, there always got some connection issue.
The problem is that I don't know how to disconnect when I use HTML5, so
python code always has some problems.
Do you know if I can run some pyvisa code to make it reset the connection,
and then I open the port by TCPIP.

Best regards,
Saul

Saulliu00, 2022-03-21:
Hi Mattieu,

Thank you so much for helping me.
Just wondering if we can schedule a meeting sometime, so it would be easier?
I am located in California, so it is PST.

Best regards,
Saul