pyvisa/pyvisa-py #320: pyvisa_py.protocols.rpc.RPCError

18476304815, 2022-04-02:
When using pyvisa to connect to a keysight N9020A spectrometer via TCPIP for reading measurement data, the following error is reported
`self.instrument_IP = f'TCPIP0::{str(instrument_IP)}::INSTR'
        #instrument_IP = '192.168.2.4'            
        rm = visa.ResourceManager('@py')
        print(rm)
        
        self.my_instrument = rm.open_resource(self.instrument_IP)
        print(self.my_instrument.query('*IDN?'))`
`Traceback (most recent call last):
  File "C:\Users\tx\AppData\Local\Programs\Python\Python39\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\tx\AppData\Local\Programs\Python\Python39\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "c:\Users\tx\.vscode\extensions\ms-python.python-2022.4.0\pythonFiles\lib\python\debugpy\__main__.py", line 45, in <module>
    cli.main()
  File "c:\Users\tx\.vscode\extensions\ms-python.python-2022.4.0\pythonFiles\lib\python\debugpy/..\debugpy\server\cli.py", line 444, in main
    run()
  File "c:\Users\tx\.vscode\extensions\ms-python.python-2022.4.0\pythonFiles\lib\python\debugpy/..\debugpy\server\cli.py", line 285, in run_file
    runpy.run_path(target_as_str, run_name=compat.force_str("__main__"))
  File "C:\Users\tx\AppData\Local\Programs\Python\Python39\lib\runpy.py", line 268, in run_path
    return _run_module_code(code, init_globals, run_name,
  File "C:\Users\tx\AppData\Local\Programs\Python\Python39\lib\runpy.py", line 97, in _run_module_code        
    _run_code(code, mod_globals, init_globals,
  File "C:\Users\tx\AppData\Local\Programs\Python\Python39\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "d:\HG-IC\培训文档\频谱仪自动控制脚本\aglient_control_2.4G.py", line 377, in <module>
    bgn_test.request_evm_pwr()#获取星状图数据
  File "d:\HG-IC\培训文档\频谱仪自动控制脚本\aglient_control_2.4G.py", line 165, in request_evm_pwr
    data1=self.my_instrument.query(":READ:EVM?")                         #query命令可以直接访问并获取数据     
  File "C:\Users\tx\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa\resources\messagebased.py", line 644, in query
    return self.read()
  File "C:\Users\tx\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa\resources\messagebased.py", line 486, in read
    message = self._read_raw().decode(enco)
  File "C:\Users\tx\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa\resources\messagebased.py", line 442, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "C:\Users\tx\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa_py\highlevel.py", line 519, in read
    return data, self.handle_return_value(session, status_code)
  File "C:\Users\tx\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_IO (-1073807298): Could not perform operation because of I/O error.       
Error closing VISA link: wrong xid in reply 22 instead of 23
`
The following error occurs when I try to use the "try" statement to write the command repeatedly
`        while True:
            try:
                self.my_instrument.write(':READ:EVM?')
                time.sleep(1)
                data1 = self.my_instrument.read()
                break
            except visa.errors.VisaIOError:
                continue`
`Traceback (most recent call last):
  File "C:\Users\tx\AppData\Local\Programs\Python\Python39\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\tx\AppData\Local\Programs\Python\Python39\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "c:\Users\tx\.vscode\extensions\ms-python.python-2022.4.0\pythonFiles\lib\python\debugpy\__main__.py", line 45, in <module>
    cli.main()
  File "c:\Users\tx\.vscode\extensions\ms-python.python-2022.4.0\pythonFiles\lib\python\debugpy/..\debugpy\server\cli.py", line 444, in main
    run()
  File "c:\Users\tx\.vscode\extensions\ms-python.python-2022.4.0\pythonFiles\lib\python\debugpy/..\debugpy\server\cli.py", line 285, in run_file
    runpy.run_path(target_as_str, run_name=compat.force_str("__main__"))
  File "C:\Users\tx\AppData\Local\Programs\Python\Python39\lib\runpy.py", line 268, in run_path
    return _run_module_code(code, init_globals, run_name,
  File "C:\Users\tx\AppData\Local\Programs\Python\Python39\lib\runpy.py", line 97, in _run_module_code        
    _run_code(code, mod_globals, init_globals,
  File "C:\Users\tx\AppData\Local\Programs\Python\Python39\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "d:\HG-IC\培训文档\频谱仪自动控制脚本\aglient_control_2.4G.py", line 377, in <module>
    bgn_test.request_evm_pwr()#获取星状图数据
  File "d:\HG-IC\培训文档\频谱仪自动控制脚本\aglient_control_2.4G.py", line 158, in request_evm_pwr
    self.my_instrument.write(':READ:EVM?')                   #读取数据
  File "C:\Users\tx\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa_py\protocols\rpc.py", line 472, in make_call
    return super(RawTCPClient, self).make_call(proc, args, pack_func, unpack_func)
  File "C:\Users\tx\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa_py\protocols\rpc.py", line 238, in make_call
    self.do_call()
  File "C:\Users\tx\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa_py\protocols\rpc.py", line 505, in do_call
    raise RPCError(msg.format(xid, self.lastxid))
pyvisa_py.protocols.rpc.RPCError: wrong xid in reply 22 instead of 23
Error closing VISA link: wrong xid in reply 23 instead of 24
`

18476304815, 2022-04-02:
visa I am also using it for the first time, please help me with this if you can

18476304815, 2022-04-02:
This is how it looks when linking to a VISA
```
instrument_IP = input('Please enter the IP:')
self.instrument_IP = f'TCPIP0::{str(instrument_IP)}::INSTR'
#TCPIP = 'TCPIP0::192.168.2.10::INSTR'
rm = visa.ResourceManager('@py')
print(rm)
        
self.my_instrument = rm.open_resource(self.instrument_IP)
print(self.my_instrument.query('*IDN?'))
```

18476304815, 2022-04-02:
@MatthieuDartiailh

MatthieuDartiailh, 2022-04-02:
I won't have to investigate this in details. However since you are on Windows and you have a VISA implementation installed just use it by creating a resource manager without passing `"@py"`. It should work.

Out of curiosity though could you provide a minimal script to reproduce the error. In particular what exception are you catching ?

18476304815, 2022-04-06:
@MatthieuDartiailh I'm sorry I haven't returned your information for so long. How can I give you the code

MatthieuDartiailh, 2022-04-06:
You can just paste it in Github if it is a simple script or you can email me. There is a public email visible on my profile.

bobmcnamara, 2022-04-18:
This issue is part of what I fixed in PR #322:  An interrupted query leaves behind stale RX packets which should be quietly ignored rather than raising an exception.

closed 2022-04-18 by MatthieuDartiailh

18476304815, 2022-06-15:
After using the "delay" parameter, the new version still seems to have the following problems.

```
Please confirm that your test equipment has been set as follows:2412MHZ,MCS=1
<class 'str'>
RMS MAX EVM:-13.23
RMS AVG EVM:-13.81
Freq_error:-4.923 KHz
AVG_PWR:19.91 dBm

Please confirm that your test equipment has been set as follows:2417MHZ,MCS=1
Traceback (most recent call last):
  File "C:\Users\tx\AppData\Local\Programs\Python\Python39\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\tx\AppData\Local\Programs\Python\Python39\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "c:\Users\tx\.vscode\extensions\ms-python.python-2022.8.0\pythonFiles\lib\python\debugpy\__main__.py", line 45, in <module>      
    cli.main()
  File "c:\Users\tx\.vscode\extensions\ms-python.python-2022.8.0\pythonFiles\lib\python\debugpy/..\debugpy\server\cli.py", line 444, in main
    run()
  File "c:\Users\tx\.vscode\extensions\ms-python.python-2022.8.0\pythonFiles\lib\python\debugpy/..\debugpy\server\cli.py", line 285, in run_file
    runpy.run_path(target_as_str, run_name=compat.force_str("__main__"))
  File "C:\Users\tx\AppData\Local\Programs\Python\Python39\lib\runpy.py", line 268, in run_path
    return _run_module_code(code, init_globals, run_name,
  File "C:\Users\tx\AppData\Local\Programs\Python\Python39\lib\runpy.py", line 97, in _run_module_code
    _run_code(code, mod_globals, init_globals,
  File "C:\Users\tx\AppData\Local\Programs\Python\Python39\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "d:\HG-IC\培训文档\RF性能自动测试脚本\keysightN9020A_Matthieu.py", line 63, in <module>
    delayed_response()
  File "d:\HG-IC\培训文档\RF性能自动测试脚本\keysightN9020A_Matthieu.py", line 28, in delayed_response
    betry = my_instrument.query('*OPC?',delay=3)
  File "C:\Users\tx\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa\resources\messagebased.py", line 648, in query      
    return self.read()
  File "C:\Users\tx\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa\resources\messagebased.py", line 486, in read       
    message = self._read_raw().decode(enco)
  File "C:\Users\tx\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa\resources\messagebased.py", line 442, in _read_raw  
    chunk, status = self.visalib.read(self.session, size)
  File "C:\Users\tx\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa\ctwrapper\functions.py", line 2337, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "C:\Users\tx\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 226, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
  File "C:\Users\tx\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value     
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
```

18476304815, 2022-06-15:
@MatthieuDartiailh And the following error was reported even though I did not change the code.
```
Please confirm that your test equipment has been set as follows:2412MHZ,MCS=2
<class 'str'>
RMS MAX EVM:-16.14
RMS AVG EVM:-16.68
Freq_error:-5.019 KHz
    exec(code, run_globals)
  File "d:\HG-IC\培训文档\RF性能自动测试脚本\keysightN9020A_Matthieu.py", line 62, in <module>
    delayed_response()
  File "d:\HG-IC\培训文档\RF性能自动测试脚本\keysightN9020A_Matthieu.py", line 27, in delayed_response
    betry = my_instrument.query('*OPC?',delay=3)
  File "C:\Users\tx\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa\resources\messagebased.py", line 648, in query      
    return self.read()
  File "C:\Users\tx\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa\resources\messagebased.py", line 486, in read       
    message = self._read_raw().decode(enco)
  File "C:\Users\tx\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa\resources\messagebased.py", line 442, in _read_raw  
    chunk, status = self.visalib.read(self.session, size)  File "C:\Users\tx\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa_py\highlevel.py", line 519, in read
    return data, self.handle_return_value(session, status_code)
  File "C:\Users\tx\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value         raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_IO (-1073807298): Could not perform operation because of I/O error.
```

18476304815, 2022-06-15:
This baffles me

18476304815, 2022-06-15:
This is the code I have since improved @MatthieuDartiailh 
```
def delayed_response():
    '''Waiting for response'''
    while True:
        betry = my_instrument.query('*OPC?',delay=3)
        if betry.strip()=='1':
            break
for tx_mcs, tx_freq in itertools.product(tx_mcs_list, tx_freq_list):
    parameters = [
        'INST:SEL WLAN',
        'CONFigure:EVM',
        ':RAD:STAN AG',
        f'FREQ:CENT {tx_freq} MHZ',
        f'CORR:SA:GAIN {External_Gain}',
        'POW:ATT 8',
        'EVM:TIME:SLEN 50ms'
        ]
    for Set in parameters:      #Write SCPI command
        my_instrument.write(Set)
        time.sleep(1)

    print(f'Please confirm that your test equipment has been set as follows:{tx_freq}MHZ,MCS={tx_mcs}')
    command = [
    'at+test_start=1',
    f'at+lo_freq={tx_freq}',
    'at+bss_bw=20',
    'at+tx_frm_type=2',
    f'at+tx_mcs={tx_mcs}',
    'at+tx_start=1',
    ]
    for c in command:
        #Write the command to the equipment to be tested through the serial port
        com.write(c.encode('utf-8'))
        time.sleep(0.5)

    my_instrument.write(':INIT:REST')
    delayed_response()

    data1 = my_instrument.query(':FETC:EVM?',delay=15)
    print(type(data1))
    instrumentData = data1.split(',')
    '''
        11g/n:
        data1[0]:RMS MAX EVM
        data1[1]:RMS AVG EVM
        data1[7]:Freq_error avg
        data1[19]:AVG_Burst_PWR
        '''
    print('RMS MAX EVM:%.2f'%(float(instrumentData[0])))
    print('RMS AVG EVM:%.2f'%(float(instrumentData[1])))
    print('Freq_error:%.3f'%(float(instrumentData[7])/1000),'KHz')
    print('AVG_PWR:%.2f'%float(instrumentData[19]),'dBm')
    print(' ')
```

MatthieuDartiailh, 2022-06-15:
Did you try installing NI VISA, or Keysight VISA since you are on windows ? This would allow to check if your script is sound and if the issue lies only with the Python backend.

18476304815, 2022-06-16:
I have installed the Keysight VISA backend correctly, can't the following information confirm that I have installed it correctly?
```
Machine Details:
   Platform ID:    Windows-10-10.0.22000-SP0
   Processor:      Intel64 Family 6 Model 165 Stepping 3, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Users\tx\AppData\Local\Programs\Python\Python39\python.exe
   Version:        3.9.7
   Compiler:       MSC v.1929 64 bit (AMD64)
   Bits:           64bit
   Build:          Aug 30 2021 20:19:38 (#tags/v3.9.7:1016ef3)
   Unicode:        UCS4

PyVISA Version: 1.12.0

Backends:
   ivi:
      Version: 1.12.0 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: Keysight Technologies
         Impl. Version: 1376087058
         Spec. Version: 5243136
      #2: C:\WINDOWS\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: Keysight Technologies
         Impl. Version: 1376087058
         Spec. Version: 5243136
   py:
      Version: 0.5.3
      ASRL INSTR: Available via PySerial (3.5)
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      TCPIP INSTR: Available
      TCPIP SOCKET: Available
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
         No module named 'gpib'
```

MatthieuDartiailh, 2022-06-16:
Sorry my bad. If that case try running your script without passing `@py` to the resource manager.

18476304815, 2022-06-16:
This still does not work, or will report an error, and also can not explain why I can obviously use a long time, run to the middle of the error, is it and I updated the window11 related?
```
Please confirm that your test equipment has been set as follows:2412MHZ,MCS=4
<class 'str'>
RMS MAX EVM:-20.63
RMS AVG EVM:-21.63            <----As you can see here I can still read and run normally
Freq_error:-3.261 KHz
AVG_PWR:18.42 dBm

**Please confirm that your test equipment has been set as follows:2417MHZ,MCS=4
<class 'str'>
RMS MAX EVM:-20.50
RMS AVG EVM:-21.59          <----As you can see here I can still read and run normally
Freq_error:-3.305 KHz
AVG_PWR:18.56 dBm**

Please confirm that your test equipment has been set as follows:2422MHZ,MCS=4
Traceback (most recent call last):
  File "C:\Users\admin\AppData\Local\Programs\Python\Python39\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\admin\AppData\Local\Programs\Python\Python39\lib\runpy.py", line 87, in _run_code  
    exec(code, run_globals)
  File "c:\Users\admin\.vscode\extensions\ms-python.python-2022.4.1\pythonFiles\lib\python\debugpy\__main__.py", line 45, in <module>
    cli.main()
  File "c:\Users\admin\.vscode\extensions\ms-python.python-2022.4.1\pythonFiles\lib\python\debugpy/..\debugpy\server\cli.py", line 444, in main
    run()
  File "c:\Users\admin\.vscode\extensions\ms-python.python-2022.4.1\pythonFiles\lib\python\debugpy/..\debugpy\server\cli.py", line 285, in run_file
    runpy.run_path(target_as_str, run_name=compat.force_str("__main__"))
  File "C:\Users\admin\AppData\Local\Programs\Python\Python39\lib\runpy.py", line 268, in run_path  
    return _run_module_code(code, init_globals, run_name,
  File "C:\Users\admin\AppData\Local\Programs\Python\Python39\lib\runpy.py", line 97, in _run_module_code
    _run_code(code, mod_globals, init_globals,
  File "C:\Users\admin\AppData\Local\Programs\Python\Python39\lib\runpy.py", line 87, in _run_code  
    exec(code, run_globals)
  File "d:\TX410x\WiFi4自动化RF性能测试脚本\频谱仪2.4G\test\keysightN9020A_test.py", line 63, in <module>
    delayed_response()
  File "d:\TX410x\WiFi4自动化RF性能测试脚本\频谱仪2.4G\test\keysightN9020A_test.py", line 29, in delayed_response
    betry = my_instrument.query('*OPC?',delay=5)
  File "C:\Users\admin\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa\resources\messagebased.py", line 644, in query
    return self.read()
  File "C:\Users\admin\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa\resources\messagebased.py", line 486, in read
  File "C:\Users\admin\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa\ctwrapper\functions.py", line 2337, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "C:\Users\admin\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 222, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
  File "C:\Users\admin\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
```

18476304815, 2022-06-16:
Maybe I'll email you the script and you can read it

18476304815, 2022-06-20:
@MatthieuDartiailh Did you receive an email from me

MatthieuDartiailh, 2022-06-23:
I did and answered by email.