pyvisa/pyvisa-py #324: [COM] usbtmc timeouts with SPD3303C

mik1234mc, 2022-05-15:
Dear all,

We use Siglent SPD3303C over usbtmc connection (at debian11) and querying the device over pyvisa times out in most of the cases (only once I got an ID string). Device.write("XXX") works correctly. The issue must be in pyvisa because if I use the usbtmc driver directly, the queries work as expected.

versions:
pyvisa-py in ./.local/lib/python3.9/site-packages (0.5.3)
pyvisa>=1.12.0 in ./.local/lib/python3.9/site-packages (from pyvisa-py) (1.12.0)

query error with pyvisa:
```
import pyvisa
pyvisa.ResourceManager().list_resources()
('ASRL/dev/ttyS1::INSTR', 'ASRL/dev/ttyS0::INSTR', 'ASRL/dev/ttyUSB0::INSTR', 'USB0::1155::30016::SPD3EDCD3R1335::0::INSTR')
src =pyvisa.ResourceManager().open_resource("USB0::1155::30016::SPD3EDCD3R1335::0::INSTR")
src.query("*IDN?")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/michael/.local/lib/python3.9/site-packages/pyvisa/resources/messagebased.py", line 648, in query
    return self.read()
  File "/home/michael/.local/lib/python3.9/site-packages/pyvisa/resources/messagebased.py", line 486, in read
    message = self._read_raw().decode(enco)
  File "/home/michael/.local/lib/python3.9/site-packages/pyvisa/resources/messagebased.py", line 442, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "/home/michael/.local/lib/python3.9/site-packages/pyvisa_py/highlevel.py", line 519, in read
    return data, self.handle_return_value(session, status_code)
  File "/home/michael/.local/lib/python3.9/site-packages/pyvisa/highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

working queries without pyvisa:
```
>>> import os
>>> d = os.open('/dev/usbtmc0', os.O_RDWR)
>>> os.write(d, bytes('*IDN?\r\n', 'ascii'))
7
>>> os.read(d, 1024)
b'Siglent Technologies,SPD3303C,SPD3EDCD3R1335,1.02.01.01.03R2,V1.3\n'
```

Michael

MatthieuDartiailh, 2022-05-16:
Hi, could you try to mimic your os calls by passing `write_termination = "\r\n"` and `read_termination="\n"` ?

My second question is are you sure your device uses USBTMC rather than serial over USB ? I am asking cause the os calls looks suspiciously simple for USB TMC.

mik1234mc, 2022-05-19:
Unfortunately, no luck with any termination character combination. I received the ID string just once from about 10 attempts.

The device manual states USB TMC and I am opening /dev/usbtmc0 dev. The USB2serial would be registered as /dev/tty*, I guess.

MatthieuDartiailh, 2022-05-20:
Can you have a look at the discussion in #256 (starting at this comment https://github.com/pyvisa/pyvisa-py/issues/256#issuecomment-1033442276) and see if this helps in your case. I do not have the hardware on hand to explore this issue.