pyvisa/pyvisa-py #327: scope screen shot using pyvisa

MaximLMC, 2022-06-11:
I’m trying to get a screenshot of a scope using python pyvisa library but I get querry uninterrupted or interrupted using the code below 

    import pyvisa
     import sys
     #
     # Example VISA address for a USB         connection:
VISA_ADDRESS = 'USB0::0x2A8D::0x1797::CN57046145::0::INSTR'
#
# Define VISA Resource Manager
scope = pyvisa.ResourceManager('C:\\Windows\\System32\\visa32.dll')
scope.read_termination = '\n'
scope.write_termination = '\n' #or = '\r'

print (scope.query('*IDN?\n'))
scope.write('SAVE:IMAG:FILEF PNG')
scope.write('HARDCOPY START')
raw_data = scope.read_raw ()
f= open ('Shot.png' 'wb')
f.write (raw_data)
f.close

MatthieuDartiailh, 2022-06-11:
Without your instrument model, manual and a traceback I cannot really help here. Also note that if you expect binary data you should use read_binary_values as otherwise you will stop at the first termination character.

MaximLMC, 2022-06-11:
I’m using Agilent and LeCtoy, and my problem is with both, I can Communicate with them, but I can’t get a screen shot of the scope, I get massage of my Agilent scope “ querry uninterrupted ”

MaximLMC, 2022-06-13:
> I’m using Agilent and LeCtoy, and my problem is with both, I can Communicate with them, but I can’t get a screen shot of the scope, I get massage of my Agilent scope “ querry uninterrupted ”

MaximLMC, 2022-06-15:
I’m using Agilent 6000,  I can Communicate with it, but I can’t get a screen shot of the scope, I get massage of my Agilent scope “ querry uninterrupted or print canceled ”

closed 2022-06-20 by MaximLMC