pyvisa/pyvisa-py #329: [COM] Communication issue with any GPIB device using NI GPIB-USB-HS on Ubuntu Linux

nelsonov, 2022-06-25:
Instrument details
------------------
* Model: Any
* Communication: TCPIP, GPIB, .... GPIB
* Link to the documentation (if available):

Output of `pyvisa-info`
-----------------------

Machine Details:
   Platform ID:    Linux-5.15.0-40-generic-x86_64-with-glibc2.35
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.10.4
   Compiler:       GCC 11.2.0
   Bits:           64bit
   Build:          Apr  2 2022 09:04:19 (#main)
   Unicode:        UCS4

PyVISA Version: 1.12.0

Backends:
   ivi:
      Version: 1.12.0 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.5.3
      ASRL INSTR: Available via PySerial (3.5)
      USB INSTR: Available via PyUSB (1.2.1). Backend: libusb1
      USB RAW: Available via PyUSB (1.2.1). Backend: libusb1
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 
      GPIB INSTR: Available via Linux GPIB (b'4.3.4 r[2020]')
      GPIB INTFC: Available via Linux GPIB (b'4.3.4 r[2020]')

Python 3.10.4 (main, Apr  2 2022, 09:04:19) [GCC 11.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyvisa
>>> pyvisa.log_to_screen()
>>> rm = pyvisa.ResourceManager()
2022-06-25 02:33:03,203 - pyvisa - DEBUG - No visa library specified, trying to find alternatives.
2022-06-25 02:33:03,203 - pyvisa - DEBUG - Environment variable PYVISA_LIBRARY is unset.
2022-06-25 02:33:03,240 - pyvisa - DEBUG - Not loading dll_extra_paths because we are not on Windows or Python < 3.8
2022-06-25 02:33:03,505 - pyvisa - DEBUG - Automatically found library files: [None, None, None, None, None]
2022-06-25 02:33:03,506 - pyvisa - DEBUG - No user defined library files
2022-06-25 02:33:03,506 - pyvisa - DEBUG - Did not find IVI binary
2022-06-25 02:33:03,536 - pyvisa - DEBUG - SerialSession was correctly imported.
2022-06-25 02:33:03,592 - pyvisa - DEBUG - USBSession and USBRawSession were correctly imported.
2022-06-25 02:33:03,599 - pyvisa - DEBUG - TCPIPSession was correctly imported.
2022-06-25 02:33:03,602 - pyvisa - DEBUG - GPIBSession was correctly imported.
2022-06-25 02:33:03,602 - pyvisa - DEBUG - pyvisa-py is available.
2022-06-25 02:33:03,603 - pyvisa - DEBUG - Created library wrapper for py
2022-06-25 02:33:03,603 - pyvisa - DEBUG - Created ResourceManager with session 4416596
>>> print(rm.list_resources())                                                                                                                                                                
2022-06-25 02:34:15,221 - pyvisa - WARNING - Found a device whose serial number cannot be read. The partial VISA resource name is: USB0::1155::14152::???::0::RAW                             
libgpib: invalid descriptor                                                                                                                                                                   
libgpib: invalid descriptor                                                                                                                                                                   
2022-06-25 02:34:15,707 - pyvisa - DEBUG - GPIB board 1 error in _find_boards(): GpibError('ask() failed: One or more arguments to the function call were invalid.')                          
libgpib: invalid descriptor                                                                                                                                                                   
libgpib: invalid descriptor                                                                                                                                                                   
2022-06-25 02:34:15,708 - pyvisa - DEBUG - GPIB board 2 error in _find_boards(): GpibError('ask() failed: One or more arguments to the function call were invalid.')                          
libgpib: invalid descriptor                                                                                                                                                                   
libgpib: invalid descriptor
2022-06-25 02:34:15,708 - pyvisa - DEBUG - GPIB board 3 error in _find_boards(): GpibError('ask() failed: One or more arguments to the function call were invalid.')
libgpib: invalid descriptor
libgpib: invalid descriptor
2022-06-25 02:34:15,708 - pyvisa - DEBUG - GPIB board 4 error in _find_boards(): GpibError('ask() failed: One or more arguments to the function call were invalid.')
[... 58 lines cut ...]
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/pyvisa_py/gpib.py", line 401, in write
    ifc.write(data)
  File "/home/lnelson/.local/lib/python3.10/site-packages/Gpib.py", line 53, in write
    gpib.write(self.id, str)
gpib.GpibError: write() error: No such device or address (errno: 6)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.10/dist-packages/pyvisa/resources/messagebased.py", line 642, in query
    self.write(message)
  File "/usr/local/lib/python3.10/dist-packages/pyvisa/resources/messagebased.py", line 197, in write
    count = self.write_raw(message.encode(enco))
  File "/usr/local/lib/python3.10/dist-packages/pyvisa/resources/messagebased.py", line 157, in write_raw
    return self.visalib.write(self.session, message)[0]
  File "/usr/local/lib/python3.10/dist-packages/pyvisa_py/highlevel.py", line 547, in write
    return written, self.handle_return_value(session, status_code)
  File "/usr/local/lib/python3.10/dist-packages/pyvisa/highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_SYSTEM_ERROR (-1073807360): Unknown system error (miscellaneous error).

The same adapter, now connected to a Win10 box talking to the same device 5:

[image]

I seem to be able to communicate enough to scan and pick up the available devices in Linux.  All three devices on the bus show up.  However, any sort of query to them fails.

MatthieuDartiailh, 2022-06-25:
I have never this one before.

Can you check that you can use pyvisa-py on your windows machine ? And maybe check also for a more recent version of Linux GPIB.

nelsonov, 2022-06-25:
Maybe I stumbled on linux-gpib in the wrong place.  Is there a more recent source than svn://svn.code.sf.net/p/linux-gpib/code/trunk ?

```
Machine Details:
   Platform ID:    Windows-10-10.0.22000-SP0
   Processor:      Intel64 Family 6 Model 126 Stepping 5, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Program Files\Python310\python.exe
   Version:        3.10.4
   Compiler:       MSC v.1929 64 bit (AMD64)
   Bits:           64bit
   Build:          Mar 23 2022 23:13:41 (#tags/v3.10.4:9d38120)
   Unicode:        UCS4

PyVISA Version: 1.12.0

Backends:
   ivi:
      Version: 1.12.0 (bundled with PyVISA)
      #1: C:\Windows\System32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: Keysight Technologies
         Impl. Version: 1361381330
         Spec. Version: 5243136
      #2: C:\Windows\System32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: Keysight Technologies
         Impl. Version: 1361381330
         Spec. Version: 5243136
```

I've spent most of the day trying to figure out how to install libusb on Windows.  I'm not a Windows person.  Unix has served me well for almost 30 years.  There's no end of references on the Internet about WHERE to get it, but not anything even remotely useful about what to do with the files once obtained and extracted.  This isn't a concern for pyvisa.  I'm just venting my extreme frustration.  I may never be able to give a good answer about whether I can use pyvisa-py on Windows.

MatthieuDartiailh, 2022-06-26:
You do not need the usb backend to be available. NI VISA is all you need to use their adapter (however I am surprised the GPIB backend is not listed...).

nelsonov, 2022-06-26:
If I use the ivi backend everything works.  I have no problems capturing waveforms from the Tektronix TDS420A (ID::5 on my gpib). I have no luck at all using the py backend.  It can't find the GPIB library and I have no idea what it's looking for:

```
C:\Users\nelso\AppData\Local\Programs\Python\Python310\lib\site-packages\gpib_ctypes\gpib\gpib.py:54: UserWarning: GPIB library not found. Please manually load it using _load_lib(filename). All GPIB functions will raise OSError until the library is manually loaded.
```

And this still leaves me with my original problem with Ubuntu where it enumerates instruments on the bus but can't write to them.

MatthieuDartiailh, 2022-06-26:
gpib_ctypes is looking for a gpib-32.dll under system or system32, not sure if that helps.

Honestly, I think your best course of action is to reproduce the issue using linux-gpib python bindings directly (you can look at pyvisa-py code under gpib.py, it is straightforward)  and reach out to them. Sorry my idea to test on Windows only caused more trouble.

nelsonov, 2022-06-27:
I think there was an issue with uploading firmware into the gpib-usb-hs when connected.  I seem to have resolved that by following a script designed for rpi's.  There is, however, a breaking change with Python3.10 and linux_gpib that may come up:  It's referenced here in a completely unrelated topic about Gnome.  It appears in ```gpibinter.c``` in their python bindings.  I will have to take this up with them.

closed 2022-06-27 by nelsonov

MatthieuDartiailh, 2022-06-28:
Happy you figured out your issue and sorry I could not provide more relevant help.