pyvisa/pyvisa-py #333: open_resource causes SDG1032X to freeze when using USB

knuesel, 2022-09-13:
My SDG1032X freezes when I run the following:

```python
#!/usr/bin/python3

import pyvisa as visa

rm = visa.ResourceManager('@py')

addr = 'USB0::0xF4EC::0x1103::SDG1XDCX6R2773::0::INSTR'

device = rm.open_resource(addr)
```

PyVISA also hangs until I switch off the device. Then I get the following traceback:

```
Traceback (most recent call last):
  File "/home/j/lab/./bug.py", line 10, in <module>
    device = rm.open_resource(addr)
  File "/home/j/.local/lib/python3.10/site-packages/pyvisa/highlevel.py", line 3284, in open_resource
    res.open(access_mode, open_timeout)
  File "/home/j/.local/lib/python3.10/site-packages/pyvisa/resources/resource.py", line 278, in open
    self.session, status = self._resource_manager.open_bare_resource(
  File "/home/j/.local/lib/python3.10/site-packages/pyvisa/highlevel.py", line 3209, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "/home/j/.local/lib/python3.10/site-packages/pyvisa_py/highlevel.py", line 167, in open
    sess = cls(session, resource_name, parsed, open_timeout)
  File "/home/j/.local/lib/python3.10/site-packages/pyvisa_py/sessions.py", line 325, in __init__
    self.after_parsing()
  File "/home/j/.local/lib/python3.10/site-packages/pyvisa_py/usb.py", line 81, in after_parsing
    self.interface = self._intf_cls(
  File "/home/j/.local/lib/python3.10/site-packages/pyvisa_py/protocols/usbtmc.py", line 292, in __init__
    self.usb_dev.reset()
  File "/home/j/.local/lib/python3.10/site-packages/usb/core.py", line 961, in reset
    self._ctx.backend.reset_device(self._ctx.handle)
  File "/home/j/.local/lib/python3.10/site-packages/usb/backend/libusb1.py", line 911, in reset_device
    _check(self.lib.libusb_reset_device(dev_handle.handle))
  File "/home/j/.local/lib/python3.10/site-packages/usb/backend/libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 2] Entity not found
```

There's probably a bug in the Siglent firmware because when I connect the device I see the following output from `dmesg`:

```
[17783.146501] usb 1-4: new high-speed USB device number 26 using xhci_hcd
[17784.466518] usb 1-4: device descriptor read/64, error -71
[17784.761197] usb 1-4: config 1 interface 0 altsetting 0 bulk endpoint 0x81 has invalid maxpacket 64
[17784.761212] usb 1-4: config 1 interface 0 altsetting 0 bulk endpoint 0x1 has invalid maxpacket 64
[17784.802463] usb 1-4: New USB device found, idVendor=f4ec, idProduct=1103, bcdDevice= 0.20
[17784.802478] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[17784.802483] usb 1-4: Product: SDG1032X
[17784.802488] usb 1-4: Manufacturer: Siglent
[17784.802492] usb 1-4: SerialNumber: SDG1XDCX6R2773
```

(Note the errors in lines 2,3,4)

However it would be nice to have USB communication working with pyvisa-py (it works with NI-VISA under Windows and python-usbtmc on Linux).

<details>
<summary><b>Output of pyvisa-info</b></summary>

```
Machine Details:
   Platform ID:    Linux-5.19.0-76051900-generic-x86_64-with-glibc2.35
   Processor:      x86_64
Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.10.4
   Compiler:       GCC 11.2.0
   Bits:           64bit
   Build:          Jun 29 2022 12:14:53 (#main)
   Unicode:        UCS4

PyVISA Version: 1.12.0

Backends:
   ivi:
      Version: 1.12.0 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.5.3
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named 'serial'
      USB INSTR: Available via PyUSB (1.2.1). Backend: libusb1
      USB RAW: Available via PyUSB (1.2.1). Backend: libusb1
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
         No module named 'gpib'
```

</details>

MatthieuDartiailh, 2022-09-14:
There has been numerous report of issues with the USB part of pyvisa-py. A reasonable option to clean the code of spurious resets that may be the root cause of this has been discussed in https://github.com/pyvisa/pyvisa-py/issues/256#issuecomment-1102934510 but I had no time to follow up on it and others did not report anything. Could you have a look ?

knuesel, 2022-09-16:
Thanks for the pointer. Indeed I can get `*IDN?` to work by commenting some of these lines. Here are some observations:

* If both of these blocks are present I get a timeout error and the device freezes:

   https://github.com/pyvisa/pyvisa-py/blob/76bda99a304c03c708364ec5a321bc20f3477e18/pyvisa_py/protocols/usbtmc.py#L215-L223

   <details><summary>Traceback</summary>

   ```
   Traceback (most recent call last):
     File "/home/j/lab/./awg.py", line 14, in <module>
       dev = rm.open_resource(addr)
     File "/home/j/.local/lib/python3.10/site-packages/pyvisa/highlevel.py", line 3284, in open_resource
       res.open(access_mode, open_timeout)
     File "/home/j/.local/lib/python3.10/site-packages/pyvisa/resources/resource.py", line 278, in open
       self.session, status = self._resource_manager.open_bare_resource(
     File "/home/j/.local/lib/python3.10/site-packages/pyvisa/highlevel.py", line 3209, in open_bare_resource
       return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
     File "/home/j/.local/lib/python3.10/site-packages/pyvisa_py/highlevel.py", line 167, in open
       sess = cls(session, resource_name, parsed, open_timeout)
     File "/home/j/.local/lib/python3.10/site-packages/pyvisa_py/sessions.py", line 325, in __init__
       self.after_parsing()
     File "/home/j/.local/lib/python3.10/site-packages/pyvisa_py/usb.py", line 81, in after_parsing
       self.interface = self._intf_cls(
     File "/home/j/.local/lib/python3.10/site-packages/pyvisa_py/protocols/usbtmc.py", line 307, in __init__
       self._enable_remote_control()
     File "/home/j/.local/lib/python3.10/site-packages/pyvisa_py/protocols/usbtmc.py", line 313, in _enable_remote_control
       self.usb_dev.ctrl_transfer(
     File "/home/j/.local/lib/python3.10/site-packages/usb/core.py", line 1082, in ctrl_transfer
       ret = self._ctx.backend.ctrl_transfer(
     File "/home/j/.local/lib/python3.10/site-packages/usb/backend/libusb1.py", line 893, in ctrl_transfer
       ret = _check(self.lib.libusb_control_transfer(
     File "/home/j/.local/lib/python3.10/site-packages/usb/backend/libusb1.py", line 602, in _check
       raise USBTimeoutError(_strerror(ret), ret, _libusb_errno[ret])
   usb.core.USBTimeoutError: [Errno 110] Operation timed out
   ```

   </details>

* Same thing with the following two lines (even if the previous two blocks are commented):

   https://github.com/pyvisa/pyvisa-py/blob/76bda99a304c03c708364ec5a321bc20f3477e18/pyvisa_py/protocols/usbtmc.py#L292-L293

   (same traceback as above)

* Same thing if I keep the two `set_configuration` calls (even if the `sest_interface_altsetting` (221) and `reset` (292) are commented).

* Finally if I keep the `set_interface_altsetting` (221) and the second `set_configuration` (216) I get a "Resource busy" error but the device doesn't freeze.

   <details><summary>Traceback</summary>

   ```
   Traceback (most recent call last):
     File "/home/j/lab/./awg.py", line 14, in <module>
       dev = rm.open_resource(addr)
     File "/home/j/.local/lib/python3.10/site-packages/pyvisa/highlevel.py", line 3284, in open_resource
       res.open(access_mode, open_timeout)
     File "/home/j/.local/lib/python3.10/site-packages/pyvisa/resources/resource.py", line 278, in open
       self.session, status = self._resource_manager.open_bare_resource(
     File "/home/j/.local/lib/python3.10/site-packages/pyvisa/highlevel.py", line 3209, in open_bare_resource
       return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
     File "/home/j/.local/lib/python3.10/site-packages/pyvisa_py/highlevel.py", line 167, in open
       sess = cls(session, resource_name, parsed, open_timeout)
     File "/home/j/.local/lib/python3.10/site-packages/pyvisa_py/sessions.py", line 325, in __init__
       self.after_parsing()
     File "/home/j/.local/lib/python3.10/site-packages/pyvisa_py/usb.py", line 81, in after_parsing
       self.interface = self._intf_cls(
     File "/home/j/.local/lib/python3.10/site-packages/pyvisa_py/protocols/usbtmc.py", line 295, in __init__
       self.usb_dev.set_configuration()
     File "/home/j/.local/lib/python3.10/site-packages/usb/core.py", line 915, in set_configuration
       self._ctx.managed_set_configuration(self, configuration)
     File "/home/j/.local/lib/python3.10/site-packages/usb/core.py", line 113, in wrapper
       return f(self, *args, **kwargs)
     File "/home/j/.local/lib/python3.10/site-packages/usb/core.py", line 159, in managed_set_configuration
       self.backend.set_configuration(self.handle, cfg.bConfigurationValue)
     File "/home/j/.local/lib/python3.10/site-packages/usb/backend/libusb1.py", line 812, in set_configuration
       _check(self.lib.libusb_set_configuration(dev_handle.handle, config_value))
     File "/home/j/.local/lib/python3.10/site-packages/usb/backend/libusb1.py", line 604, in _check
       raise USBError(_strerror(ret), ret, _libusb_errno[ret])
   usb.core.USBError: [Errno 16] Resource busy
   ```

   </details>

The following combinations seem to work fine (at least for multiple `*IDN?` queries):

1. Commenting all four calls.
1. Keeping one of the four calls and commenting the other three.
1. Keeping the first `set_configuration` (216) and the `reset`.
1. Keeping the `set_interface_altsetting` and the `reset`.

It would be interesting to know if one of options 2-4 also works for @mwcmwc12 and @pfjarschel...

MatthieuDartiailh, 2022-09-16:
Thanks for looking into it. I will try to see if I can make a PR making the inheritance less buggy and we can try to see if that works for you.

knuesel, 2023-01-06:
I cannot reproduce with the 0.6.0 release. Thanks for fixing this!

MatthieuDartiailh, 2023-01-06:
Happy this old pain looks gone. I hope it is gone for good !