pyvisa/pyvisa-py #337: list_resources returning empty list with TCPIP instrument available

Drayrs, 2022-09-29:
Creating a ResourceManager instance and using the list_resources() method is not returning the expected TCPIP instrument.
An empty list is returned instead, which was suspicious considering I have instruments available on my subnet.
Interestingly, I recreated the TCPIPInstrSession.list_resources method, and was able to return the expected instrument.

Attempting:
```
import pyvisa
import common, rpc, sessions, vxi11, ipaddress, socket, time, psutil

pyvisa.log_to_screen()

print('Intended usage of rm:')
rm = pyvisa.ResourceManager('@py')
ins = rm.list_resources()
print(ins)

print('\nMirroring rpc/vxi11 script from TCPIPInstrSession.list_resources:')
rpc_vx_test()

print('\nsessions.Session._session_classes')
print(sessions.Session._session_classes)
print(repr(pyvisa.constants.InterfaceType.tcpip))
```

Returns:
```
2022-09-29 12:48:23,698 - pyvisa - DEBUG - SerialSession was not imported No module named 'serial'.
2022-09-29 12:48:23,699 - pyvisa - DEBUG - USBSession and USBRawSession were not imported No module named 'usb'.
2022-09-29 12:48:23,703 - pyvisa - DEBUG - TCPIPSession was correctly imported.
2022-09-29 12:48:23,709 - pyvisa - DEBUG - GPIBSession was not imported No module named 'gpib'.
2022-09-29 12:48:23,709 - pyvisa - DEBUG - Created library wrapper for py
2022-09-29 12:48:23,709 - pyvisa - DEBUG - Created ResourceManager with session 9359651
Intended usage of rm:
()

Mirroring rpc/vxi11 script from TCPIPInstrSession.list_resources:
['TCPIP::192.168.228.43::INSTR']
rname.filter(all_res, "?*::INSTR")
('TCPIP::192.168.228.43::INSTR',)

sessions.Session._session_classes
{}
<InterfaceType.tcpip: 6>
2022-09-29 12:48:24,741 - pyvisa - DEBUG - Closing ResourceManager (session: 9359651)
2022-09-29 12:48:24,741 - pyvisa - DEBUG - Closing ResourceManager (session: 9359651)
```

**Output of `pyvisa-info`**
```
Machine Details:
   Platform ID:    Windows-10-10.0.19044-SP0
   Processor:      Intel64 Family 6 Model 158 Stepping 13, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Users\xxxxxxxxxl\AppData\Local\pypoetry\Cache\virtualenvs\lattice-rnD1xMsC-py3.10\Scripts\python.exe
   Version:        3.10.0
   Compiler:       MSC v.1929 64 bit (AMD64)
   Bits:           64bit
   Build:          Oct  4 2021 19:00:18 (#tags/v3.10.0:b494f59)
   Unicode:        UCS4

PyVISA Version: 1.12.0

Backends:
   ivi:
      Version: 1.12.0 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: Keysight Technologies
         Impl. Version: 1378253280
         Spec. Version: 7340288
      #2: C:\WINDOWS\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: Keysight Technologies
         Impl. Version: 1378253280
         Spec. Version: 7340288
   py:
      Version: 0.5.3
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named 'serial'
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      TCPIP INSTR: Available
      TCPIP SOCKET: Available
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
         No module named 'gpib'
```

MatthieuDartiailh, 2022-09-29:
The PR that added this capability has been merged recently and is not part of the 5.3 release. You can try installing directly from github while waiting for the new release.

Drayrs, 2022-09-29:
Thanks! I am having a hard time installing from github, but that could just be a problem on my end. I will learn to check the release next time.

closed 2022-09-29 by Drayrs

Drayrs, 2022-09-29:
Can confirm list_resources is functioning on intended now that I have the correct PR installed:

Calling `pyvisa.ResourceManager('@py').list_resources()` returns:
```
2022-09-29 13:49:08,816 - pyvisa - DEBUG - SerialSession was not imported No module named 'serial'.
2022-09-29 13:49:08,817 - pyvisa - DEBUG - USBSession and USBRawSession were not imported No module named 'usb'.
2022-09-29 13:49:08,836 - pyvisa - DEBUG - TCPIPSession was correctly imported.
2022-09-29 13:49:08,841 - pyvisa - DEBUG - GPIBSession was not imported No module named 'gpib'.
2022-09-29 13:49:08,842 - pyvisa - DEBUG - Created library wrapper for py
2022-09-29 13:49:08,842 - pyvisa - DEBUG - Created ResourceManager with session 8399426
('TCPIP::192.168.228.43::INSTR',)
2022-09-29 13:49:09,864 - pyvisa - DEBUG - Closing ResourceManager (session: 8399426)
2022-09-29 13:49:09,864 - pyvisa - DEBUG - Closing ResourceManager (session: 8399426)
```

Specifying a branch when installing from github fixed the issues I was having there:
`poetry add git+https://github.com/pyvisa/pyvisa-py.git#main`