pyvisa/pyvisa-py #338: Attribute error setting keepalive on macosx

bobmcnamara, 2022-10-12:
The keepalive code for TCPIPInstrVxi11 and TCPIPSocketSession is broken on OSX.  Specifically, socket.TCP_KEEPIDLE doesn't exist on OSX.

Here is the code I tried to run:

```
import pyvisa
from pyvisa.constants import ResourceAttribute

rm = pyvisa.ResourceManager("@py")
scope = rm.open_resource("TCPIP::192.168.1.9::INSTR")
scope.set_visa_attribute(ResourceAttribute.tcpip_keepalive, True)
```

Here is the error:
```
AttributeError: module 'socket' has no attribute 'TCP_KEEPIDLE'
```

See https://stackoverflow.com/questions/12248132/how-to-change-tcp-keepalive-timer-using-python-script for more details.

abunimeh, 2022-10-25:
on macOS it is `TCP_KEEPALIVE`

bobmcnamara, 2022-10-26:
Here is the offending code:

```
            if attribute_state is True:
                self.interface.sock.setsockopt(
                    socket.SOL_SOCKET, socket.SO_KEEPALIVE, 1
                )
                self.interface.sock.setsockopt(
                    socket.IPPROTO_TCP, socket.TCP_KEEPIDLE, 60
                )
                self.interface.sock.setsockopt(
                    socket.IPPROTO_TCP, socket.TCP_KEEPINTVL, 60
                )
                self.interface.sock.setsockopt(
                    socket.IPPROTO_TCP, socket.TCP_KEEPCNT, 5
                )
                self.keepalive = True
```
The problem is that on macOS, the socket module has no attribute TCP_KEEPIDLE.