pyvisa/pyvisa-py #340: Large messages not sent correct via TCP/IP

wintermuteger, 2022-11-18:
Hi pyvisa team,

I am trying to upload a larger file via Visa to an instrument. I want to use a tcp/ip based connection. Unfortunately it failed, whenever the size of the message exceeded 1024 bytes.

After quite some debugging the root cause seems to be in the referenced line of code:

https://github.com/pyvisa/pyvisa-py/blob/3f63d8ac3140e2d6a8f4cf0f68e4a82ab469963b/pyvisa_py/tcpip.py#L541

chunk_size is set to 1024 and is used to feed the while loop. Selection of data is using self.max_recv_size, which defaults to 4096.

https://github.com/pyvisa/pyvisa-py/blob/3f63d8ac3140e2d6a8f4cf0f68e4a82ab469963b/pyvisa_py/tcpip.py#L552

Therefore in my example of about 1.5kb the full block is sent via TCP/IP, but the EOI indicator is not set, as chunk_size was smaller than 1.5kb and the flag selection assumed that a second chunk would follow. 

My proposal would be to replace self.max_recv_size in line 552 by chunk_size so the inherent logic of the code is maintained. Afterwards the chunk_size could be kept at the hard coded 1024 or be linked to self.max_recv_size unifying it with the receive direction block size.

MatthieuDartiailh, 2022-11-18:
Could you make a PR with your proposed changes ?

I will have to dive into the VXI-11 spec to figure out if there is a link between the send and receive sides.

MatthieuDartiailh, 2022-11-18:
Actually the vxi-11 specification (http://www.vxibus.org/specifications.html) explicitly says that ``maxRecvSize the size of the largest data parameter the network instrument server can accept in a device_write RPC. This value SHALL be at least 1024.`` so please just get rid of chunk_size.

That makes me doubt the logic in read but it works so let's just not touch it right now.

MatthieuDartiailh, 2022-12-22:
Closed by #344

closed 2022-12-22 by MatthieuDartiailh

wintermuteger, 2023-01-17:
Thanks for following up, and sorry that I didn't manage to provide a pull request.