pyvisa/pyvisa-py #347: [COM] Communication issue with Kikusui PLZ334WL, Agilent 34410A using USB

michaelknapman, 2023-01-12:
0.5.3 list_resources('@py') lists no resources whereas NI list_resources() lists resources.
NI-VISA 2022Q3 backend has successful IDN query for both devices, but pyvisa-py fails.
I am not using a VM.

Here is my python code. In the code below, I successfully connect to my oscilloscope, but fail to connect to both other devices. 

```
import pyvisa
pyvisa.log_to_screen() #enables verbose console logging in red text

rm = pyvisa.ResourceManager('@py') # @py or blank
print(rm.list_resources())

scope = rm.open_resource("TCPIP0::10.6.141.33::inst0::INSTR")
print(scope.query('*IDN?'))

try:
    load1 = rm.open_resource("USB0::0x0B3E::0x1020::CR001341::INSTR")
    print(load1.query('*IDN?'))
except:
    print("connecting to kikusui PLZ334WL didnt work")
    
try:
    multi_input = rm.open_resource("USB0::0x0957::0x0618::TW46340011::INSTR")
    print(multi_input.query('*IDN?'))
except:
    print("connecting to Agilent 34410A didn't work")
```
Here is the output to the command prompt caused by pyvisa.log_to_screen().

```
2023-01-12 16:45:16,056 - pyvisa - DEBUG - SerialSession was correctly imported.
2023-01-12 16:45:16,086 - pyvisa - DEBUG - USBSession and USBRawSession were correctly imported.
2023-01-12 16:45:16,092 - pyvisa - DEBUG - TCPIPSession was correctly imported.
2023-01-12 16:45:16,094 - pyvisa - DEBUG - GPIBSession was not imported No module named 'gpib'.
2023-01-12 16:45:16,094 - pyvisa - DEBUG - Created library wrapper for py
2023-01-12 16:45:16,094 - pyvisa - DEBUG - Created ResourceManager with session 2236029
()
2023-01-12 16:45:16,114 - pyvisa - DEBUG - TCPIP0::10.6.141.33::inst0::INSTR - opening ...
2023-01-12 16:45:16,115 - pyvisa - DEBUG - RawTCPClient: connecting to socket at (10.6.141.33, 111)
2023-01-12 16:45:16,163 - pyvisa - DEBUG - Make call <PortMapperVersion.get_port: 3>, (395183, 1, 6, 0), <bound method PortMapperPacker.pack_mapping of <pyvisa_py.protocols.rpc.PortMapperPacker object at 0x000001E5F5AF9750>>, <bound method Unpacker.unpack_uint of <pyvisa_py.protocols.rpc.PortMapperUnpacker object at 0x000001E5F5AF97D0>>
2023-01-12 16:45:16,164 - pyvisa - DEBUG - Sending record through <socket.socket fd=876, family=2, type=1, proto=0, laddr=('10.4.32.185', 56932), raddr=('10.6.141.33', 111)>: b'\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x02\x00\x01\x86\xa0\x00\x00\x00\x02\x00\x00\x00\x03\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x06\x07\xaf\x00\x00\x00\x01\x00\x00\x00\x06\x00\x00\x00\x00'
2023-01-12 16:45:16,164 - pyvisa - DEBUG - RawTCPClient: procedure type 3
2023-01-12 16:45:16,164 - pyvisa - DEBUG - Start receiving at least 1 packages
2023-01-12 16:45:16,195 - pyvisa - DEBUG - received b'\x80\x00\x00\x1c'
2023-01-12 16:45:16,195 - pyvisa - DEBUG - received b'\x00\x00\x00\x01\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02\xce'
2023-01-12 16:45:16,195 - pyvisa - DEBUG - Received record through <socket.socket fd=876, family=2, type=1, proto=0, laddr=('10.4.32.185', 56932), raddr=('10.6.141.33', 111)>: bytearray(b'\x00\x00\x00\x01\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02\xce')
2023-01-12 16:45:16,196 - pyvisa - DEBUG - RawTCPClient: closing socket
2023-01-12 16:45:16,196 - pyvisa - DEBUG - RawTCPClient: connecting to socket at (10.6.141.33, 718)
2023-01-12 16:45:16,214 - pyvisa - DEBUG - Make call 10, (933470925, 0, 10000, 'inst0'), <bound method Vxi11Packer.pack_create_link_parms of <pyvisa_py.protocols.vxi11.Vxi11Packer object at 0x000001E5F5AF9650>>, <bound method Vxi11Unpacker.unpack_create_link_resp of <pyvisa_py.protocols.vxi11.Vxi11Unpacker object at 0x000001E5F5AF96D0>>
2023-01-12 16:45:16,215 - pyvisa - DEBUG - Sending record through <socket.socket fd=1200, family=2, type=1, proto=0, laddr=('10.4.32.185', 56933), raddr=('10.6.141.33', 718)>: b"\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x02\x00\x06\x07\xaf\x00\x00\x00\x01\x00\x00\x00\n\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x007\xa3\xa2\xcd\x00\x00\x00\x00\x00\x00'\x10\x00\x00\x00\x05inst0\x00\x00\x00"
2023-01-12 16:45:16,215 - pyvisa - DEBUG - RawTCPClient: procedure type 10
2023-01-12 16:45:16,226 - pyvisa - DEBUG - received b'\x80\x00\x00('
2023-01-12 16:45:16,226 - pyvisa - DEBUG - received b'\x00\x00\x00\x01\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xba\xbe\x00\x0c\x00\x00\x02\xd1\x00\x00\x10\x00'
2023-01-12 16:45:16,226 - pyvisa - DEBUG - Received record through <socket.socket fd=1200, family=2, type=1, proto=0, laddr=('10.4.32.185', 56933), raddr=('10.6.141.33', 718)>: bytearray(b'\x00\x00\x00\x01\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xba\xbe\x00\x0c\x00\x00\x02\xd1\x00\x00\x10\x00')
2023-01-12 16:45:16,226 - pyvisa - DEBUG - TCPIP0::10.6.141.33::inst0::INSTR - is open with session 9222430
2023-01-12 16:45:16,227 - pyvisa - DEBUG - Make call 11, (-1161953268, 2000, 10000, 8, b'*IDN?\r\n'), <bound method Vxi11Packer.pack_device_write_parms of <pyvisa_py.protocols.vxi11.Vxi11Packer object at 0x000001E5F5AF9650>>, <bound method Vxi11Unpacker.unpack_device_write_resp of <pyvisa_py.protocols.vxi11.Vxi11Unpacker object at 0x000001E5F5AF96D0>>
2023-01-12 16:45:16,227 - pyvisa - DEBUG - Sending record through <socket.socket fd=1200, family=2, type=1, proto=0, laddr=('10.4.32.185', 56933), raddr=('10.6.141.33', 718)>: b"\x00\x00\x00\x02\x00\x00\x00\x00\x00\x00\x00\x02\x00\x06\x07\xaf\x00\x00\x00\x01\x00\x00\x00\x0b\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xba\xbe\x00\x0c\x00\x00\x07\xd0\x00\x00'\x10\x00\x00\x00\x08\x00\x00\x00\x07*IDN?\r\n\x00"
2023-01-12 16:45:16,227 - pyvisa - DEBUG - RawTCPClient: procedure type 11
2023-01-12 16:45:16,243 - pyvisa - DEBUG - received b'\x80\x00\x00 '
2023-01-12 16:45:16,243 - pyvisa - DEBUG - received b'\x00\x00\x00\x02\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x07'
2023-01-12 16:45:16,245 - pyvisa - DEBUG - Received record through <socket.socket fd=1200, family=2, type=1, proto=0, laddr=('10.4.32.185', 56933), raddr=('10.6.141.33', 718)>: bytearray(b'\x00\x00\x00\x02\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x07')
2023-01-12 16:45:16,245 - pyvisa - DEBUG - TCPIP0::10.6.141.33::inst0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2023-01-12 16:45:16,245 - pyvisa - DEBUG - Make call 12, (-1161953268, 4096, 2000, 10000, 0, 0), <bound method Vxi11Packer.pack_device_read_parms of <pyvisa_py.protocols.vxi11.Vxi11Packer object at 0x000001E5F5AF9650>>, <bound method Vxi11Unpacker.unpack_device_read_resp of <pyvisa_py.protocols.vxi11.Vxi11Unpacker object at 0x000001E5F5AF96D0>>
2023-01-12 16:45:16,246 - pyvisa - DEBUG - Sending record through <socket.socket fd=1200, family=2, type=1, proto=0, laddr=('10.4.32.185', 56933), raddr=('10.6.141.33', 718)>: b"\x00\x00\x00\x03\x00\x00\x00\x00\x00\x00\x00\x02\x00\x06\x07\xaf\x00\x00\x00\x01\x00\x00\x00\x0c\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xba\xbe\x00\x0c\x00\x00\x10\x00\x00\x00\x07\xd0\x00\x00'\x10\x00\x00\x00\x00\x00\x00\x00\x00"
2023-01-12 16:45:16,246 - pyvisa - DEBUG - RawTCPClient: procedure type 12
2023-01-12 16:45:16,264 - pyvisa - DEBUG - received b'\x80\x00\x00T'
2023-01-12 16:45:16,264 - pyvisa - DEBUG - received b'\x00\x00\x00\x03\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x04\x00\x00\x00.TEKTRONIX,DPO4104,C020899,CF:91.1CT FV:v2.68 \n\x00\x00'
2023-01-12 16:45:16,264 - pyvisa - DEBUG - Received record through <socket.socket fd=1200, family=2, type=1, proto=0, laddr=('10.4.32.185', 56933), raddr=('10.6.141.33', 718)>: bytearray(b'\x00\x00\x00\x03\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x04\x00\x00\x00.TEKTRONIX,DPO4104,C020899,CF:91.1CT FV:v2.68 \n\x00\x00')
TEKTRONIX,DPO4104,C020899,CF:91.1CT FV:v2.68
[... 44 lines cut ...]
   ivi:
      Version: 1.12.0 (bundled with PyVISA)
      #1: C:\windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 23069952
         Spec. Version: 5244928
      #2: C:\windows\system32\visa64.dll:
         found by: auto
         bitness: 64
[... cut ...]
         Spec. Version: 5244928
   py:
      Version: 0.5.3
      ASRL INSTR: Available via PySerial (3.5)
      USB INSTR: Available via PyUSB (1.2.1). Backend: N/A
      USB RAW: Available via PyUSB (1.2.1). Backend: N/A
      TCPIP INSTR: Available
      TCPIP SOCKET: Available
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
         No module named 'gpib'
```

MatthieuDartiailh, 2023-01-13:
Pyvisa-py 0.6.0 introduces a couple fixes for USB devices. Could you try to give it a spin ?

michaelknapman, 2023-01-13:
Kikusui PLZ334WL and Agilent 34410A still don't connect with pyvisa-py 0.60.

Code:

```
import pyvisa
pyvisa.log_to_screen() #enables verbose console logging in red text

rm = pyvisa.ResourceManager('@py') # @py or blank
print(rm.list_resources())

scope = rm.open_resource("TCPIP0::10.6.141.33::inst0::INSTR")
print(scope.query('*IDN?'))

try:
    load1 = rm.open_resource("USB0::0x0B3E::0x1020::CR001341::INSTR")
    print(load1.query('*IDN?'))
except:
    print("connecting to kikusui PLZ334WL didnt work")
    
try:
    multi_input = rm.open_resource("USB0::0x0957::0x0618::TW46340011::INSTR")
    print(multi_input.query('*IDN?'))
except:
    print("connecting to Agilent 34410A didn't work")

```

Logging:

```

2023-01-13 10:23:31,471 - pyvisa - DEBUG - SerialSession was correctly imported.
2023-01-13 10:23:31,499 - pyvisa - DEBUG - USBSession and USBRawSession were correctly imported.
2023-01-13 10:23:31,521 - pyvisa - DEBUG - TCPIPSession was not imported vicp.
2023-01-13 10:23:31,523 - pyvisa - DEBUG - GPIBSession was not imported No module named 'gpib'.
2023-01-13 10:23:31,523 - pyvisa - DEBUG - Created library wrapper for py
2023-01-13 10:23:31,523 - pyvisa - DEBUG - Created ResourceManager with session 7887554
('ASRL5::INSTR',)
2023-01-13 10:23:31,543 - pyvisa - DEBUG - TCPIP0::10.6.141.33::inst0::INSTR - opening ...
2023-01-13 10:23:31,544 - pyvisa - DEBUG - RawTCPClient: connecting to socket at (10.6.141.33, 111)
2023-01-13 10:23:31,551 - pyvisa - DEBUG - Make call <PortMapperVersion.get_port: 3>, (395183, 1, 6, 0), <bound method PortMapperPacker.pack_mapping of <pyvisa_py.protocols.rpc.PortMapperPacker object at 0x00000147D6F3F510>>, <bound method Unpacker.unpack_uint of <pyvisa_py.protocols.rpc.PortMapperUnpacker object at 0x00000147D6F3F4D0>>
2023-01-13 10:23:31,551 - pyvisa - DEBUG - Sending record through <socket.socket fd=928, family=2, type=1, proto=0, laddr=('10.4.32.185', 52879), raddr=('10.6.141.33', 111)>: b'\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x02\x00\x01\x86\xa0\x00\x00\x00\x02\x00\x00\x00\x03\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x06\x07\xaf\x00\x00\x00\x01\x00\x00\x00\x06\x00\x00\x00\x00'
2023-01-13 10:23:31,551 - pyvisa - DEBUG - RawTCPClient: procedure type 3
2023-01-13 10:23:31,551 - pyvisa - DEBUG - Start receiving at least 1 packages
2023-01-13 10:23:31,556 - pyvisa - DEBUG - received b'\x80\x00\x00\x1c'
2023-01-13 10:23:31,557 - pyvisa - DEBUG - received b'\x00\x00\x00\x01\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02\xce'
2023-01-13 10:23:31,557 - pyvisa - DEBUG - Received record through <socket.socket fd=928, family=2, type=1, proto=0, laddr=('10.4.32.185', 52879), raddr=('10.6.141.33', 111)>: bytearray(b'\x00\x00\x00\x01\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02\xce')
2023-01-13 10:23:31,557 - pyvisa - DEBUG - RawTCPClient: closing socket
2023-01-13 10:23:31,557 - pyvisa - DEBUG - RawTCPClient: connecting to socket at (10.6.141.33, 718)
2023-01-13 10:23:31,563 - pyvisa - DEBUG - Make call 10, (841882712, 0, 10000, 'inst0'), <bound method Vxi11Packer.pack_create_link_parms of <pyvisa_py.protocols.vxi11.Vxi11Packer object at 0x00000147D6F5D3D0>>, <bound method Vxi11Unpacker.unpack_create_link_resp of <pyvisa_py.protocols.vxi11.Vxi11Unpacker object at 0x00000147D6F5D410>>
2023-01-13 10:23:31,563 - pyvisa - DEBUG - Sending record through <socket.socket fd=928, family=2, type=1, proto=0, laddr=('10.4.32.185', 52880), raddr=('10.6.141.33', 718)>: b"\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x02\x00\x06\x07\xaf\x00\x00\x00\x01\x00\x00\x00\n\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x002.\x1cX\x00\x00\x00\x00\x00\x00'\x10\x00\x00\x00\x05inst0\x00\x00\x00"
2023-01-13 10:23:31,564 - pyvisa - DEBUG - RawTCPClient: procedure type 10
2023-01-13 10:23:31,569 - pyvisa - DEBUG - received b'\x80\x00\x00('
2023-01-13 10:23:31,569 - pyvisa - DEBUG - received b'\x00\x00\x00\x01\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xba\xbe\x00\x03\x00\x00\x02\xd1\x00\x00\x10\x00'
2023-01-13 10:23:31,569 - pyvisa - DEBUG - Received record through <socket.socket fd=928, family=2, type=1, proto=0, laddr=('10.4.32.185', 52880), raddr=('10.6.141.33', 718)>: bytearray(b'\x00\x00\x00\x01\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xba\xbe\x00\x03\x00\x00\x02\xd1\x00\x00\x10\x00')
2023-01-13 10:23:31,570 - pyvisa - DEBUG - TCPIP0::10.6.141.33::inst0::INSTR - is open with session 4154323
2023-01-13 10:23:31,570 - pyvisa - DEBUG - Make call 11, (-1161953277, 2000, 10000, 8, b'*IDN?\r\n'), <bound method Vxi11Packer.pack_device_write_parms of <pyvisa_py.protocols.vxi11.Vxi11Packer object at 0x00000147D6F5D3D0>>, <bound method Vxi11Unpacker.unpack_device_write_resp of <pyvisa_py.protocols.vxi11.Vxi11Unpacker object at 0x00000147D6F5D410>>
2023-01-13 10:23:31,570 - pyvisa - DEBUG - Sending record through <socket.socket fd=928, family=2, type=1, proto=0, laddr=('10.4.32.185', 52880), raddr=('10.6.141.33', 718)>: b"\x00\x00\x00\x02\x00\x00\x00\x00\x00\x00\x00\x02\x00\x06\x07\xaf\x00\x00\x00\x01\x00\x00\x00\x0b\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xba\xbe\x00\x03\x00\x00\x07\xd0\x00\x00'\x10\x00\x00\x00\x08\x00\x00\x00\x07*IDN?\r\n\x00"
2023-01-13 10:23:31,571 - pyvisa - DEBUG - RawTCPClient: procedure type 11
[... cut ...]
         Impl. Version: 23069952
         Spec. Version: 5244928
   py:
      Version: 0.6.0
      ASRL INSTR: Available via PySerial (3.5)
      USB INSTR: Available via PyUSB (1.2.1). Backend: N/A
      USB RAW: Available via PyUSB (1.2.1). Backend: N/A
      TCPIP INSTR: Available
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
         No module named 'gpib'
```

MatthieuDartiailh, 2023-01-13:
I missed that, but since you are on Windows, what is your interest in using pyvisa-py over the standard IVI backend that will use NI-VISA ?

Setting up USB on Windows when another player has already installed drivers can be challenging, so I do not recommend it if you can avoid it. Also from pyvisa info reports, it appears you are missing libusb.

michaelknapman, 2023-01-13:
My desire to use pyvisa-py is simply because of convenience. NI-VISA initial installation is error prone and takes about 1 hour including forced system restarts whereas pyvisa-py is an easy pip install. 
My end goal is to put pyvisa-py on a bunch of machines without ever touching NI-VISA. I believe I don't need anything else in the NI suite. 

Should I attempt to install libusb?

MatthieuDartiailh, 2023-01-13:
You can try. Alternatively you could try the VISA implementation of other manufacturers. Some users told me the Rohde&Schwarz one was less bloated.

michaelknapman, 2023-01-13:
Despite running both:
pip install libusb
pip install libusb-package

Pyvisa-info remains the same, saying Backend: N/A.
How do I get pyvisa to recognize this backend?

michaelknapman, 2023-01-16:
I got Pyvisa to recognize libusb1 backend by using this thread https://stackoverflow.com/a/68064311
I'll test with my Kikusui and Agilent equipment later this week.

MatthieuDartiailh, 2023-01-16:
If this is not already in pyvisa-py doc feel fre to make a PR to add it if you manage to get things to work. (You may have to uninstall the USB driver NI installed first).

MatthieuDartiailh, 2023-01-18:
The issue you encountered with zeroconf is a bug and I will look into it ASAP. Regarding your direct issue did you look at the documentation https://pyvisa.readthedocs.io/projects/pyvisa-py/en/latest/installation.html#usb-resources-usb-instr-raw and in particular did you went through the dance of uninstalling the drivers Windows installed (through NI) ? 

I do not remember the specifics but we cannot use those drivers and need a generic driver which makes using USB on windows quite cumbersome (USB never being easy). Hope this helps.

michaelknapman, 2023-01-19:
At the link specified it says:
On Windows, you may have to uninstall the USBTMC specific driver installed by Windows and re-install a generic driver.

I uninstalled the driver for Kikusui PLZ334WL and now the device is in Other devices in device manager. Windows doesn't know how to automatically supply drivers to it, and I am not sure what a generic driver for this is.
[image]

Now not even NI-VISA works with the PLZ334WL.

I can reinstall drivers manually and successfully by navigating to C:\Program Files\IVI Foundation\VISA\IVI USB Staging\b841aa80\Windows
I think I got that driver from either Keysight at the following link, or from Kikusui's support page, not sure.
https://edadocs.software.keysight.com/connect/usb-instrument-under-other-devices-in-device-manager-471980441.html
This driver still causes the blank instrument list problem.

MatthieuDartiailh, 2023-01-19:
Can you give the procedure described here a shot ? https://learn.microsoft.com/en-us/windows-hardware/drivers/usbcon/winusb-installation?source=recommendations

I honestly do not remember when that part of the doc was written or by whom. If I get a chance I will try to test at work.

michaelknapman, 2023-05-17:
Hi, I fixed the problem yesterday with help from issue #364 on a fresh windows system.

It appears that the minimum number of steps required to get TCPIP and USB discovery on a fresh windows system is as follows:

- Install python 3.11.3, add to path
- Run in command line: pip install pyvisa pyvisa-py psutil zeroconf pyusb libusb 
- Install USB drivers with Zadig for all three devices: PLZ334WL, Agilent 34410A
- Add the libusb x64 to path: C:\Users\mknapman\AppData\Local\Programs\Python\Python311\Lib\site-packages\libusb\_platform\_windows\x64 
- Add the libusb x86 to path: C:\Users\mknapman\AppData\Local\Programs\Python\Python311\Lib\site-packages\libusb\_platform\_windows\x86
- Restart the computer

Now typing into the shell these three lines returns both the TCPIP resources and USB resources discovered on my system.
import pyvisa
rm = pyvisa.ResourceManager('@py')
print(rm.list_resources())

I can see that fix #365 added the proper documentation to https://pyvisa.readthedocs.io/projects/pyvisa-py/en/latest/installation.html#usb-resources-usb-instr-raw

I successfully migrated all of my code using the '@py' with no problems. 

It appears that the step I missed previously was using Zadig. Note that I am using the latest pyvisa-py, I am not sure if the latest 0.70 version helped versus the older 0.53.

This issue can be closed now.

closed 2023-05-18 by MatthieuDartiailh