pyvisa/pyvisa-py #355: open_resource() fails with python 3.7 and pyvisa-py==0.6.2

sbor23, 2023-02-16:
The latest version (0.6.2) is broken with python 3.7. Simply opening a `TCPIP` connection throws an error.

To Reproduce
--------------
Steps to reproduce the behavior:
1. Make sure you have python 3.7 installed
2. Make sure you have pyvisa-py 0.6.2 installed
3. Open a connection via TCPIP socket:
  ```python3
     pyvisa.ResourceManager("@py").open_resource("TCPIP::10.0.0.101::1080::SOCKET")
  ```

This results in the following stacktrace:
```
  [first part of the stacktrace is omitted]
    self.device = self.rm.open_resource(self.VisaString)
  File "C:\Users\labor72\Desktop\test\venv\lib\site-packages\pyvisa\highlevel.py", line 3284, in open_resource
    res.open(access_mode, open_timeout)
  File "C:\Users\labor72\Desktop\test\venv\lib\site-packages\pyvisa\resources\resource.py", line 279, in open
    self._resource_name, access_mode, open_timeout
  File "C:\Users\labor72\Desktop\test\venv\lib\site-packages\pyvisa\highlevel.py", line 3209, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "C:\Users\labor72\Desktop\test\venv\lib\site-packages\pyvisa_py\highlevel.py", line 164, in open
    parsed.interface_type_const, parsed.resource_class
  File "C:\Users\labor72\Desktop\test\venv\lib\site-packages\pyvisa_py\sessions.py", line 198, in get_session_class
    "No class registered for %s, %s" % (interface_type, resource_class)
ValueError: No class registered for InterfaceType.tcpip, SOCKET
```

**Output of `pyvisa-info`**
```
Machine Details:
   Platform ID:    Windows-10-10.0.19041-SP0
   Processor:      Intel64 Family 6 Model 165 Stepping 5, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Users\labor72\Desktop\test\venv\Scripts\python.exe
   Version:        3.7.9
   Compiler:       MSC v.1900 64 bit (AMD64)
   Bits:           64bit
   Build:          Aug 17 2020 18:58:18 (#tags/v3.7.9:13c94747c7)
   Unicode:        UCS4

PyVISA Version: 1.12.0

Backends:
   ivi:
      Version: 1.12.0 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 23069952
         Spec. Version: 5244928
      #2: C:\WINDOWS\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 23069952
         Spec. Version: 5244928
   py:
      Version: 0.6.2
      ASRL INSTR: Available via PySerial (3.5)
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      TCPIP INSTR: Available
         Resource discovery:
         - VXI-11: partial (psutil not installed)
         - hislip: disabled (zeroconf not installed)
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
         No module named 'gpib'
```

MatthieuDartiailh, 2023-02-16:
Can you run with logging to screen enabled ? Somebody reported a similar issue before but for INSTR and it was fixed https://github.com/pyvisa/pyvisa-py/pull/344.

MatthieuDartiailh, 2023-02-16:
Okay I finally got what you are dealing with. And I did mess up. Basically since you run pyvisa 1.12 (as is allowed) you do not get the vicp constant and as a consequence the module fail to load the SOCKET resource but manage to get the VXI-11 and HiSLIP. I will work on a fix.

MatthieuDartiailh, 2023-02-17:
0.6.3 which fixes the issue is out

closed 2023-02-17 by MatthieuDartiailh