pyvisa/pyvisa-py #363: pyvisa list_resources fails with unhandled usb.USBError

eliotb, 2023-03-29:
```
>>> rm = pyvisa.ResourceManager("@py")
>>> rl = rm.list_resources()

USBError: [Errno 32] Pipe error
```

I have 2 USB instruments (Keysight) attached, plus a bunch of other USB devices.

In these two places below, (NotImplementedError, ValueError) are handled (and the device is skipped), but not `usb.USBError`

https://github.com/pyvisa/pyvisa-py/blob/5765485764dcf951d9f16743e6db27885347d1e1/pyvisa_py/usb.py#L261
https://github.com/pyvisa/pyvisa-py/blob/5765485764dcf951d9f16743e6db27885347d1e1/pyvisa_py/usb.py#L318

If I add the usb.USBError to the exception list, I can correctly enumerate my VISA instruments.
Enabling logging to screen, I get
`2023-03-29 17:13:50,044 - pyvisa - WARNING - Found a device whose serial number cannot be read. The partial VISA resource name is: USB0::3285::7::???::0::RAW`

Question: Is this an acceptable solution?

MatthieuDartiailh, 2023-03-29:
Ideally I would like to understand why a device identified as a USBTMC device manage to fail to provide a serial number. 

Do you need to fix both USBTMC and USBRAW or is fixing USB raw devices sufficient ? USBRAW are not officially part of VISA and are simply a NI extension so I would it less surprising if things were not that well defined in this case.

eliotb, 2023-03-29:
Fixing usb raw is sufficient for me, thanks.

MatthieuDartiailh, 2023-03-29:
In that case I would be open to merge such a change. Can you make a PR ?

MatthieuDartiailh, 2023-04-17:
@eliotb do you need help opening the PR (I see you committed some changes to your branch).

eliotb, 2023-04-19:
No thanks. I'll do it. At the time of our conversation above, I was away from the place where I could test my changes so I didn't make the PR before testing again.

Ryan-Draves, 2023-04-27:
I have also been seeing `[Errno 32] Pipe error` as of version `0.6.2`. It won't reproduce consistently across machines & instruments, but the setups it does occur on will happen every time.

The steps to reproduce I've been using are:
```
import pyvisa

rm = pyvisa.ResourceManager()
inst = rm.open_resource('some_usb_address')
inst.query('*IDN?')  # make sure communication works
inst.close()  # clean up resource

rm_2 = pyvisa.ResourceManager()
inst_2 = rm.open_resource('some_usb_address')
inst_2.query('*IDN?')  # USBError: [Errno 32] Pipe error
```

I think #352 broke something more than what #370 will handle. Let me know if I should submit a separate issue for this case.

PR #370 (2023-04-27): Ewb/usberr
Fixes #363 

Handle a new exception type that I observed in my setup.
I have tested this change with my hardware and confirm it works (for me).
closed 2023-04-27 by MatthieuDartiailh