pyvisa/pyvisa-py #364: [COM] Not Detecting USB Devices

mccormackt, 2023-04-03:
I'm looking to use pyvisa with the pyvisa-py backend but it's not listing USB devices like the NI VISA drivers are.
I already had the NI VISA drivers installed on my machine but want to use pyvisa-py in the application where computers may or may not already have the NI VISA drivers installed as well.

I have also copied libusb-1.0.dll to System32 like i've found in answers to questions online, as well as in the script folder.

Instrument details
------------------
* Model: Rigol DS1104Z
* Communication: USB
* Link to the documentation (if available): https://beyondmeasure.rigoltech.com/acton/attachment/1579/f-0504/1/-/-/-/-/MSO1000Z_Datasheet.pdf

Output of `pyvisa-info`
-----------------------

Machine Details:
   Platform ID:    Windows-10-10.0.19045-SP0
   Processor:      Intel64 Family 6 Model 158 Stepping 10, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\CompanyName\PythonEnv\TestExec\Scripts\python.exe
   Version:        3.10.4
   Compiler:       MSC v.1929 64 bit (AMD64)
   Bits:           64bit
   Build:          Sep 19 2022 14:44:45 (#main)
   Unicode:        UCS4

PyVISA Version: 1.13.0

Backends:
   ivi:
      Version: 1.13.0 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 23069952
         Spec. Version: 5244928
      #2: C:\WINDOWS\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 23069952
         Spec. Version: 5244928
   py:
      Version: 0.6.3
      ASRL INSTR: Available via PySerial (3.5)
      USB INSTR: Available via PyUSB (1.2.1). Backend: libusb1
      USB RAW: Available via PyUSB (1.2.1). Backend: libusb1
      TCPIP INSTR: Available 
         Resource discovery:
         - VXI-11: ok
         - hislip: ok
      VICP INSTR:
         Please install PyVICP to use this resource type.
      TCPIP SOCKET: Available 
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
         No module named 'gpib'

Here's my program and the output
------------------
import pyvisa

rm = pyvisa.ResourceManager()
rmPy = pyvisa.ResourceManager('@py')
pyvisa.util.get_debug_info()

print(rm.list_resources('?*'))
print(rmPy.list_resources('?*'))

('USB0::0x1AB1::0x04CE::DS1ZB153900305::INSTR', 'ASRL10::INSTR')
()

Here's my requirements.txt in case it's helpful
------------------
altgraph==0.17.3
appdirs==1.4.4
async-timeout==4.0.2
click==8.1.2
colorama==0.4.4
dataclasses==0.6
docutils==0.18.1
et-xmlfile==1.1.0
Flask==2.1.1
formulas==1.2.2
future==0.18.2
graphviz==0.20
ifaddr==0.2.0
importlib-metadata==6.1.0
importlib-resources==5.12.0
itsdangerous==2.1.2
Jinja2==3.1.1
libusb1==3.0.0
MarkupSafe==2.1.1
modbus-tk==1.1.2
numpy==1.22.3
numpy-financial==1.0.0
openpyxl==3.0.9
packaging==21.3
pefile==2022.5.30
pkg-about==1.0.8
psutil==5.9.4
psycopg2==2.9.5
pycparser==2.21
pycryptodome==3.14.1
Pygments==2.12.0
pyinstaller==5.7.0
pyinstaller-hooks-contrib==2022.14
pymodbus==2.5.3
pyModbusTCP==0.1.10
pyparsing==3.0.9
pyserial==3.5
python-can==4.0.0
python-dateutil==2.8.2
pyusb==1.2.1
PyVISA==1.13.0
PyVISA-py==0.6.3
pywin32==304
pywin32-ctypes==0.2.0
regex==2022.4.24
schedula==1.2.15
six==1.16.0
tomli==2.0.1
typing_extensions==4.2.0
uptime==3.0.1
Werkzeug==2.1.1
windows-curses==2.3.0
wrapt==1.14.1
zeroconf==0.47.4
zipp==3.15.0

arr-ee, 2023-04-03:
I'll try to reproduce this in a bit, but in the meantime: have you looked at https://github.com/libusb/libusb/wiki/Windows#user-content-How_to_use_libusb_on_Windows ?

mccormackt, 2023-04-03:
> I'll try to reproduce this in a bit, but in the meantime: have you looked at https://github.com/libusb/libusb/wiki/Windows#user-content-How_to_use_libusb_on_Windows ?

Replacing the driver worked. Thanks!

MatthieuDartiailh, 2023-04-03:
@arr-ee @mccormackt would one of you be motivated to make a PR to add to the docs ? This kind of issue pops often enough that it warrants a dedicated entry in the documentation.

arr-ee, 2023-04-03:
yeah I'll update docs later today