pyvisa/pyvisa-py #366: Problem with _read() in pyvisa_py/sessions.py?

spukhafte-sys, 2023-04-15:
I reported a problem with pyvisa's read_raw() [(original issue with details here)](https://github.com/pyvisa/pyvisa/issues/741), but after some investigation I suspect the problem is actually in the _read() function in pyvisa_py/sessions.py. I get the documented behavour from pyvisa's read_raw() if this code at line 820:
```
if finish_time and time.time() > finish_time:
    return bytes(out), StatusCode.error_timeout
```
is changed to this:
```
if finish_time and time.time() > finish_time:
    if termination_char_en:
        return bytes(out), StatusCode.error_timeout
    else:
        return bytes(out), StatusCode.success
```
However I'm not familiar enough with the code to be certain I didn't break something else. Recommendations?

spukhafte-sys, 2023-04-17:
It appears read_raw() and read_bytes() aren't exercised by the test suite. So I wouldn't be expected to provide test code if I submit a pull request?

MatthieuDartiailh, 2023-04-17:
Both methods require an actual device to be tested which we do not have available as part of CI hence the lack of tests.

spukhafte-sys, 2023-04-17:
> Both methods require an actual device to be tested which we do not have available as part of CI hence the lack of tests.

We'd have to write a mock instrument for the test framework?

MatthieuDartiailh, 2023-04-18:
Then the issue is at which level do we mock the device. To really cover the code we would need to mock pyserial. It can be done but this is not an endeavor I ever had the bandwidth to tackle (in particular because we would need similar shims for other resource types).