pyvisa/pyvisa-py #367: Communication different Siglent devices with LAN

sshsurabhi, 2023-04-18:
Hello,

I am trying to program different Siglent devices such as E'Load, Power Supply, Multimeter, and as well as digital Oscilloscope.

Initially, I installed pyvisa, NI-Instruments libraries, and everything then I tried connecting USBTC cables to communicate with each other. Initially, I tried small commands to send and receive information. Everything working fine.

But, Now I don't want to use USBTC cables. I want to use only an Ethernet connection i.e I want to connect with only a LAN cable to every device and try to receive information from them with programming.

```
import socket

# Set the IP address and port number of the SPD3303 Programmable DC Power Supply
IP_ADDRESS = "192.168.222.205"
PORT_NUMBER = 5025

# Create a socket object and connect it to the SPD3303 Programmable DC Power Supply
s = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
s.connect((IP_ADDRESS, PORT_NUMBER))

# Send a command to the SDM3055 digital multimeter
command = "*IDN?\n"
s.send(command.encode())

# Receive the response from the SPD3303 Programmable DC Power Supply and print it
response = s.recv(4096).decode()
print(response)

# s.send(b'CONF:VOLT:DC AUTO,DEF\r\n')
# s.send(b'INIT\r\n')
# res = s.recv(1024).decode().strip()
# print(res)

# Close the connection to the SPD3303 Programmable DC Power Supply
s.close()
```

As I have shown in the example code (Which I took from the Siglent Website documents), I want to create a TCP connection and then I want to give some commands to the devices and receive them as well without using the pyvisa library.

Is it possible??

I could able to get the device name from the given code but not anything else. Does anybody do the way I am trying? No other commands are working. Please help me in doing this.

arr-ee, 2023-04-18:
pyvisa supports ethernet connections, and there should be no issues communicating with your instruments that way.

> I want to create a TCP connection and then I want to give some commands to the devices and receive them as well without using the pyvisa library.

If you explicitly want to avoid using pyvisa I’d suggest asking this question to Siglent’s support team or Stack Overflow.

sshsurabhi, 2023-04-19:
As you said, as well as per the working instructions of the device, It's working well with 'pyvisa' library, which needs a NI-instruments '.dll' file running in the backend. Without NI-Instruments installed, the 'pyvisa' library also not works as far as my understanding. So, I am just trying to communicate it with a pure TCP connection, which was also mentioned online.

[https://www.siglenteu.com/application-note/programming-example-using-python-to-configure-a-basic-waveform-with-an-sdg-x-series-generator-via-open-sockets-lan/?pdf=6875](url)

Now a small script is running. Yesterday it did not work.

MatthieuDartiailh, 2023-04-19:
PyVISA requires a backend to be available. The default one uses ctypes to bind a third party VISA implementation (by NI, Keysight, R&S, ...).

Pyvisa-py is a pure Python backends and typically supports TCPIP based resources. So while you can use a bare socket, you can also use the pyvisa-py backend to avoid relying on a third party VISA implementation.

closed 2023-04-27 by MatthieuDartiailh