pyvisa/pyvisa-py #369: list_resources() raises usb.core.USBError when Labjack T7 is connected to USB hub

wallacbe, 2023-04-21:
Bus 001 Device 021: ID 0cd5:0007 LabJack Corporation LabJack T7  

File "/venv/lib/python3.9/site-packages/pyvisa/highlevel.py", line 3129, in list_resources
    return self.visalib.list_resources(self.session, query)
  File "/venv/lib/python3.9/site-packages/pyvisa_py/highlevel.py", line 486, in list_resources
    resources += st.list_resources()
  File "/venv/lib/python3.9/site-packages/pyvisa_py/usb.py", line 317, in list_resources
    serial = dev.serial_number
  File "/venv/lib/python3.9/site-packages/usb/core.py", line 864, in serial_number
    self._serial_number = util.get_string(self, self.iSerialNumber)
  File "/venv/lib/python3.9/site-packages/usb/util.py", line 317, in get_string
    buf = get_descriptor(
  File "/venv/lib/python3.9/site-packages/usb/control.py", line 172, in get_descriptor
    desc = dev.ctrl_transfer(
  File "/venv/lib/python3.9/site-packages/usb/core.py", line 1082, in ctrl_transfer
    ret = self._ctx.backend.ctrl_transfer(
  File "/venv/lib/python3.9/site-packages/usb/backend/libusb1.py", line 893, in ctrl_transfer
    ret = _check(self.lib.libusb_control_transfer(
  File "/venv/lib/python3.9/site-packages/usb/backend/libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 32] Pipe error

Solution:
Add usb.core.USBError to try/catch block
https://github.com/pyvisa/pyvisa-py/blob/main/pyvisa_py/usb.py#L318

MatthieuDartiailh, 2023-04-22:
Duplicate of #363 whose author will open a PR.

closed 2023-04-27 by MatthieuDartiailh

wallacbe, 2023-04-27:
Is there a process to request pypi release?

MatthieuDartiailh, 2023-04-27:
I will try to do one by the end of the May. Feel free to ping me if you do not see anything coming.