pyvisa/pyvisa-py #372: How to disable discovery of VXI-11, HiSLIP and VICP devices

canol, 2023-05-11:
Hello, we have been using PyVISA-Py backend for a long time. Relatively recently, we noticed that it gives these warnings when we call `ResourceManager.list_resources()` method:

```
C:\my_project\venv\lib\site-packages\pyvisa_py\tcpip.py:404: UserWarning: TCPIP:instr resource discovery is limited to the default interface.Install psutil: pip install psutil if you want to scan all interfaces.
  warnings.warn(
C:\my_project\venv\lib\site-packages\pyvisa_py\tcpip.py:121: UserWarning: TCPIP::hislip resource discovery requires the zeroconf package to be installed... try 'pip install zeroconf'
  warnings.warn(
```

We can install these dependencies as well, but we don't have any issues finding our current set of devices, and hence don't want to introduce any behavior change in device discovery. Is there a way to not get these warnings other than installing these additional dependencies?

MatthieuDartiailh, 2023-05-11:
Like any other warning you can filter them out. Since TCPIP discovery is not always desirable we could consider adding a way to disable it but it is not part of the standard VISA interface.

canol, 2023-05-11:
Okay, thank you very much.

closed 2023-05-11 by canol