pyvisa/pyvisa-py #384: PyUSB does not seem to be properly installed one one computer

calugaruful-dotcom, 2023-07-13:
I have a python script I am running in pycharm and recently I am getting this error: **ValueError: PyUSB does not seem to be properly installed.**

I have ran this script on this computer before, and it worked. I usually install, libusb, pyvisa, pyvisa-py, pyusb packages and that seems to take care of the backend. For example, if I copy this same script and install the same packages on a different computer, it works and I get no error.

On the non working computer when I run pyvisa-info I can also see the usb backend is not configured:
`PyVISA Version: 1.13.0

Backends:
   ivi:
      Version: 1.13.0 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.7.0
      TCPIP INSTR: Available
         Resource discovery:
         - VXI-11: ok
         - hislip: ok
      TCPIP SOCKET: Available
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named 'serial'
      USB INSTR:
         PyUSB does not seem to be properly installed.
         Please refer to PyUSB documentation and
         install a suitable backend like
         libusb 0.1, libusb 1.0, libusbx,
         libusb-win32 or OpenUSB.
         No backend available
      USB RAW:
         PyUSB does not seem to be properly installed.
         Please refer to PyUSB documentation and
         install a suitable backend like
         libusb 0.1, libusb 1.0, libusbx,
         libusb-win32 or OpenUSB.
         No backend available
      VICP INSTR:
         Please install PyVICP to use this resource type.
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'
      GPIB INTFC:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'`

calugaruful-dotcom, 2023-07-13:
This is the error output I am getting from `usb.core.find()`

2023-07-13 11:23:59,582 ERROR:usb.backend.libusb1:Error loading libusb 1.0 backend
2023-07-13 11:23:59,584 ERROR:usb.libloader:'Libusb 1' could not be found
2023-07-13 11:23:59,584 ERROR:usb.backend.libusb1:Error loading libusb 1.0 backend
2023-07-13 11:23:59,584 ERROR:usb.libloader:'OpenUSB library' could not be found
2023-07-13 11:23:59,584 ERROR:usb.backend.openusb:Error loading OpenUSB backend
2023-07-13 11:23:59,585 ERROR:usb.libloader:'Libusb 0' could not be found
2023-07-13 11:23:59,585 ERROR:usb.backend.libusb0:Error loading libusb 0.1 backend

Where is usb automatically looking for the libusb files? I understood it was system32 folder, and I put the .dlls there but that didn't work.

arr-ee, 2023-07-13:
`usb` is `pyusb`, there is no pyvisa code there. This should help you find the answer or ask the question in the relevant project (where more people might have relevant knowledge).