pyvisa/pyvisa-py #385: [COM] Communication issue loop

LongnoseRob, 2023-07-17:
For tests (in pymeasure) I would like to create a Visa-resource pointing to a loop as supported by PySerial.
Is there away to create a VISA resource string point to a `serial.serial_for_url("loop://", timeout=0.2)` object?

Or would there be another way to create a rm-object for a loop device?

------------------
* Link to the documentation (if available): https://pyserial.readthedocs.io/en/latest/url_handlers.html#loop

Output of `pyvisa-info`
-----------------------

```
Machine Details:
   Platform ID:    Linux-6.4.3-arch1-2-x86_64-with-glibc2.37
   Processor:      

Python:
   Implementation: CPython
   Executable:     /usr/bin/python
   Version:        3.11.3
   Compiler:       GCC 13.1.1 20230429
   Architecture:   ('x86', 64)
   Build:          Jun  5 2023 09:32:32 (#main)
   Unicode:        UCS4

PyVISA Version: 1.13.1.dev29+g55f01db

Backends:
   ivi:
      Version: 1.13.1.dev29+g55f01db (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.7.0
      ASRL INSTR: Available via PySerial (3.5)
      USB INSTR: Available via PyUSB (1.2.1). Backend: libusb1
      USB RAW: Available via PyUSB (1.2.1). Backend: libusb1
      TCPIP INSTR: Available 
         Resource discovery:
         - VXI-11: ok
         - hislip: ok
      TCPIP SOCKET: Available 
      GPIB INSTR: Available via Linux GPIB (b'4.3.5')
      GPIB INTFC: Available via Linux GPIB (b'4.3.5')
      VICP INSTR:
         Please install PyVICP to use this resource type.
```

LongnoseRob, 2023-07-17:
Thank you for the feedback and the proposal,
let me see if I can get it working and I will open a PR

You basically mean to modify `SerialSession.after_parsing()` to take `serial.searial_for_url` arguments?

MatthieuDartiailh, 2023-07-17:
Basically you can try replacing cls on line 87 by serial.serial_from_url.

LongnoseRob, 2023-07-17:
Alright let me give it a try over the next days and report back..

PR #386 (2023-07-29): replace the call to serial.Serial with serial.serial_for_url
to support loop:// and other url-types supported in pyserial (>3.0)

tested with the follwing code on python 3.11.3, pyserial 3.5:
```
import pyvisa
rm = pyvisa.ResourceManager("@py")
abc = rm.open_resource("ASRLloop://::INSTR")
abc.timeout = 3000
abc.read_termination = "\r"
abc.write_termination = "\r\n"
abc.write("Test1234567890")
print(abc.read())
```

- [ ] Closes #385
- [x] Executed ``black . && isort -c . && flake8`` with no errors
- [x] The change is fully covered by automated unit tests
- [x] Documented in docs/ as appropriat [... cut]
closed 2023-07-29 by MatthieuDartiailh