pyvisa/pyvisa-py #390: [COM] Communication issue with PL303QMD-P using TCPIP(RAW)

lambcutlet, 2023-08-19:
Hi, 
I'm attempting to link a few bits of kit together using pyvisa.
I'm stumped at my PSU, When performing list resources its listed as 'TCPIP::192.168.1.124::INSTR'
But the manual says when using Ethernet, use TCPIP address, port & RAW protocol.
As i understand pyvisa this should be in the format of 'TCPIP::192.168.1.124::9221::SOCKET'. 

**Using 'TCPIP::192.168.1.124::INSTR' I get this error message:**
------------------
  ('ASRL/dev/ttyUSB1::INSTR', 'ASRL/dev/ttyUSB0::INSTR', 'ASRL/dev/ttyACM1::INSTR', 'TCPIP::192.168.1.124::INSTR')
  ASRL/dev/ttyUSB1::INSTR  :device not responding
  ASRL/dev/ttyUSB0::INSTR  :device not responding
  ASRL/dev/ttyACM1::INSTR  :device not responding
  TCPIP::192.168.1.124::INSTR  :device not responding
  quitting!
  Exception ignored in atexit callback: <function ResourceManager.__new__.<locals>.call_close at 0x7fc3f2559d00>
  Traceback (most recent call last):
    File "/home/kp/.local/lib/python3.11/site-packages/pyvisa/highlevel.py", line 3014, in call_close
      meth()
    File "/home/kp/.local/lib/python3.11/site-packages/pyvisa/highlevel.py", line 3073, in close
      resource.close()
    File "/home/kp/.local/lib/python3.11/site-packages/pyvisa/resources/resource.py", line 319, in close
      self.visalib.close(self.session)
    File "/home/kp/.local/lib/python3.11/site-packages/pyvisa_py/highlevel.py", line 439, in close
      return self.handle_return_value(session, sess.close())
                                               ^^^^^^^^^^^^
    File "/home/kp/.local/lib/python3.11/site-packages/pyvisa_py/tcpip.py", line 481, in close
      self.interface.destroy_link(self.link)
    File "/home/kp/.local/lib/python3.11/site-packages/pyvisa_py/protocols/vxi11.py", line 344, in destroy_link
      return self.make_call(
             ^^^^^^^^^^^^^^^
    File "/home/kp/.local/lib/python3.11/site-packages/pyvisa_py/protocols/rpc.py", line 478, in make_call
      return super(RawTCPClient, self).make_call(proc, args, pack_func, unpack_func)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "/home/kp/.local/lib/python3.11/site-packages/pyvisa_py/protocols/rpc.py", line 236, in make_call
      result = unpack_func()
               ^^^^^^^^^^^^^
    File "/home/kp/.local/lib/python3.11/site-packages/pyvisa_py/protocols/vxi11.py", line 167, in unpack_device_error
      return self.unpack_int()
             ^^^^^^^^^^^^^^^^^
    File "/usr/lib64/python3.11/xdrlib.py", line 171, in unpack_int
      raise EOFError
  EOFError: 
  Exception ignored in: <function Resource.__del__ at 0x7fc3f255ad40>
  Traceback (most recent call last):
    File "/home/kp/.local/lib/python3.11/site-packages/pyvisa/resources/resource.py", line 180, in __del__
    File "/home/kp/.local/lib/python3.11/site-packages/pyvisa/resources/resource.py", line 319, in close
    File "/home/kp/.local/lib/python3.11/site-packages/pyvisa_py/highlevel.py", line 439, in close
    File "/home/kp/.local/lib/python3.11/site-packages/pyvisa_py/tcpip.py", line 481, in close
    File "/home/kp/.local/lib/python3.11/site-packages/pyvisa_py/protocols/vxi11.py", line 344, in destroy_link
    File "/home/kp/.local/lib/python3.11/site-packages/pyvisa_py/protocols/rpc.py", line 478, in make_call
    File "/home/kp/.local/lib/python3.11/site-packages/pyvisa_py/protocols/rpc.py", line 236, in make_call
    File "/home/kp/.local/lib/python3.11/site-packages/pyvisa_py/protocols/vxi11.py", line 167, in unpack_device_error
    File "/usr/lib64/python3.11/xdrlib.py", line 171, in unpack_int
  EOFError:

**Using 'TCPIP::192.168.1.124::9221::SOCKET'**
------------------
It just times out. I've tried TCPIP0 & TCPIP1 as well all result in the same failure.
I have also tried playing around with terminations but they make no difference.
I have also attempted 'socket' as I see a few other people have shown this method as functional and in my case it does work.
Can you provide an insight to get this PSU talking to pyvisa?

Here's my code :
------------------
import pyvisa
rm = pyvisa.ResourceManager('@py')
resources = rm.list_resources()
print(resources)
psu_tti = False
psu_tti_ID = 'THURLBY THANDAR, PL303QMD-P,  371832, 3.01-4.06\n'
for devices in resources:
    try:
        #if devices == 'TCPIP::192.168.1.124::INSTR':
            #devices = 'TCPIP::192.168.1.124::9221::SOCKET'
        dev = rm.open_resource(devices)
        dev.query_delay = 0.1
        dev.timeout = 5000
        response = dev.query("*IDN?")
        time.sleep(0.1)
        print(response)
        # TTI PL303QMD-P
        if response:
            psu_tti = dev
            print("   ", psu_tti_ID, "   connected to:\n   ", devices, "\n")
    except:
        print(devices, " :device not responding")
        pass
#print(psu_tti, " :dmm status check")
#if psu_tti: print("psu_tti: ", psu_tti.query("v1?"))
print("quitting!")
quit()

Instrument details
------------------
* Model: THURLBY THANDAR, PL303QMD-P
* Communication: TCPIP port 9221 RAW
* Link to the documentation (if available): Page 26, https://resources.aimtti.com/manuals/New_PL+PL-P_Series_Instruction_Manual-Iss18.pdf

Output of `pyvisa-info`
-----------------------

`pyvisa-info
/home/kp/.local/lib/python3.11/site-packages/gpib_ctypes/gpib/gpib.py:54: UserWarning: GPIB library not found. Please manually load it using _load_lib(filename). All GPIB functions will raise OSError until the library is manually loaded.
  warnings.warn(message)
Machine Details:
   Platform ID:    Linux-6.4.10-200.fc38.x86_64-x86_64-with-glibc2.37
   Processor:      

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.11.4
   Compiler:       GCC 13.1.1 20230511 (Red Hat 13.1.1-2)
   Bits:           64bit
   Build:          Jun  7 2023 00:00:00 (#main)
   Unicode:        UCS4

PyVISA Version: 1.13.0

Backends:
   ivi:
      Version: 1.13.0 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.6.3
      ASRL INSTR: Available via PySerial (3.5)
      USB INSTR: Available via PyUSB (1.2.1). Backend: libusb1
      USB RAW: Available via PyUSB (1.2.1). Backend: libusb1
      TCPIP INSTR: Available 
         Resource discovery:
         - VXI-11: ok
         - hislip: ok
      VICP INSTR:
         Please install PyVICP to use this resource type.
      TCPIP SOCKET: Available 
      GPIB INSTR:
         gpib_ctypes is installed but could not locate the gpib library.
         Please manually load it using:
           gpib_ctypes.gpib.gpib._load_lib(filename)
         before importing pyvisa.`

Sorry long message and rubbish formatting.

lambcutlet, 2023-08-20:
Hi, 
Made a little progress and managed to get a response before an EOF error

## Code:
```
from datetime import datetime
import pyvisa
rm = pyvisa.ResourceManager('@py')
resources = rm.list_resources()

def debug(*args):
    print(datetime.now(), *args)

devices = 'TCPIP0::192.168.1.124::9221::INSTR' #)#, , , 'TCPIP0::192.168.1.124::9221::SOCKET'

try:
    dev = rm.open_resource(devices)
    dev.query_delay = 0.1
    dev.timeout = 1000
    dev.read_termination = None
    dev.write_termination = '\n'
    response = dev.query("*IDN?")
    print("response: ", response)
except Exception as ex:
    debug(ex)
except:
    print(devices, " :device not responding")
    pass

print("quitting!\n")
quit()
```

## Error message:
response:  THURLBY THANDAR, PL303QMD-P,  371832, 3.01-4.06
quitting!

Exception ignored in atexit callback: <function ResourceManager.__new__.<locals>.call_close at 0x7f35f8a85d00>
Traceback (most recent call last):
  File "/home/kp/.local/lib/python3.11/site-packages/pyvisa/highlevel.py", line 3014, in call_close
    meth()
  File "/home/kp/.local/lib/python3.11/site-packages/pyvisa/highlevel.py", line 3073, in close
    resource.close()
  File "/home/kp/.local/lib/python3.11/site-packages/pyvisa/resources/resource.py", line 319, in close
    self.visalib.close(self.session)
  File "/home/kp/.local/lib/python3.11/site-packages/pyvisa_py/highlevel.py", line 439, in close
    return self.handle_return_value(session, sess.close())
                                             ^^^^^^^^^^^^
  File "/home/kp/.local/lib/python3.11/site-packages/pyvisa_py/tcpip.py", line 481, in close
    self.interface.destroy_link(self.link)
  File "/home/kp/.local/lib/python3.11/site-packages/pyvisa_py/protocols/vxi11.py", line 344, in destroy_link
    return self.make_call(
           ^^^^^^^^^^^^^^^
  File "/home/kp/.local/lib/python3.11/site-packages/pyvisa_py/protocols/rpc.py", line 478, in make_call
    return super(RawTCPClient, self).make_call(proc, args, pack_func, unpack_func)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kp/.local/lib/python3.11/site-packages/pyvisa_py/protocols/rpc.py", line 236, in make_call
    result = unpack_func()
             ^^^^^^^^^^^^^
  File "/home/kp/.local/lib/python3.11/site-packages/pyvisa_py/protocols/vxi11.py", line 167, in unpack_device_error
    return self.unpack_int()
           ^^^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.11/xdrlib.py", line 171, in unpack_int
    raise EOFError
EOFError: 

Exception ignored in: <function Resource.__del__ at 0x7f35f8a86d40>
Traceback (most recent call last):
  File "/home/kp/.local/lib/python3.11/site-packages/pyvisa/resources/resource.py", line 180, in __del__
  File "/home/kp/.local/lib/python3.11/site-packages/pyvisa/resources/resource.py", line 319, in close
  File "/home/kp/.local/lib/python3.11/site-packages/pyvisa_py/highlevel.py", line 439, in close
  File "/home/kp/.local/lib/python3.11/site-packages/pyvisa_py/tcpip.py", line 481, in close
  File "/home/kp/.local/lib/python3.11/site-packages/pyvisa_py/protocols/vxi11.py", line 344, in destroy_link
  File "/home/kp/.local/lib/python3.11/site-packages/pyvisa_py/protocols/rpc.py", line 478, in make_call
  File "/home/kp/.local/lib/python3.11/site-packages/pyvisa_py/protocols/rpc.py", line 236, in make_call
  File "/home/kp/.local/lib/python3.11/site-packages/pyvisa_py/protocols/vxi11.py", line 167, in unpack_device_error
  File "/usr/lib64/python3.11/xdrlib.py", line 171, in unpack_int
EOFError:

lambcutlet, 2023-08-20:
More info using 
```
pyvisa.log_to_screen()
```
Turns out  this request 'TCPIP0::192.168.1.124::9221::INSTR'  is detected as RAW and is redirected to port 111. It only ever receives *IDN? info regardless of which ever command is sent. Obviously not the correct way forward.

Using 'TCPIP::192.168.1.124::9221::SOCKET' I now see this response below. Where there appears to be data in the buffer but because there isn't a terminator it times out, my guess. Need to find away of writing and reading.
I saw an 'issue' earlier today but i can't find it now.  About a read() command needing modified adding an if-else statement to prevent crashes. Could this be the issue here?

2023-08-20 20:26:06,910 - pyvisa - DEBUG - TCPIP0::192.168.1.124::9221::SOCKET - opening ...
2023-08-20 20:26:06,912 - pyvisa - DEBUG - TCPIP0::192.168.1.124::9221::SOCKET - is open with session 6286428
2023-08-20 20:26:07,012 - pyvisa - DEBUG - TCPIP0::192.168.1.124::9221::SOCKET - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2023-08-20 20:26:08,021 - pyvisa - DEBUG - TCPIP0::192.168.1.124::9221::SOCKET - exception while reading: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
Buffer content: bytearray(b'')
2023-08-20 20:26:08.021145 VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
2023-08-20 20:26:08,021 - pyvisa - DEBUG - Closing ResourceManager (session: 9578851)
2023-08-20 20:26:08,021 - pyvisa - DEBUG - TCPIP0::192.168.1.124::9221::SOCKET - closing
2023-08-20 20:26:08,021 - pyvisa - DEBUG - TCPIP0::192.168.1.124::9221::SOCKET - is closed

lambcutlet, 2023-08-20:
Got it.
Address = 'TCPIP0::192.168.1.124::9221::SOCKET'
read_termination = '\r\n'

Read out each byte from the buffer and noticed '\r\n' at the end.

closed 2023-08-20 by lambcutlet

MatthieuDartiailh, 2023-08-20:
Happy you figured out a solution.