pyvisa/pyvisa-py #391: pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

jorsalas123, 2023-08-21:
Instrument details
------------------
* Model: Keysight E3634A
* Communication: Serial - USB

Output of `pyvisa-info`
-----------------------
root@PCBINT:/usr/local/lib/python2.7/dist-packages# pyvisa-info
Machine Details:
   Platform ID:    Linux-5.15.0-79-generic-x86_64-with-Ubuntu-20.04-focal
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python2
   Version:        2.7.18
   Compiler:       GCC 9.4.0
   Bits:           64bit
   Build:          Jul  1 2022 12:27:04 (#default)
   Unicode:        UCS4

PyVISA Version: 1.10.1

Backends:
   ni:
      Version: 1.10.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.4.1
      ASRL INSTR: Available via PySerial (3.5)
      TCPIP INSTR: Available
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named usb
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named usb
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
         No module named gpib
      TCPIP SOCKET: Available

jorsalas123, 2023-08-21:
Im tried to connect with the PSU E3634A but not receive any answer

root@PCBINT:/usr/local/lib/python2.7/dist-packages# sudo python
Python 2.7.18 (default, Jul  1 2022, 12:27:04)
[GCC 9.4.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import visa; rm = visa.ResourceManager("@py"); inst = rm.open_resource('ASRL/dev/ttyUSB0::INSTR');
>>> inst.query("*IDN?")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "pyvisa/resources/messagebased.py", line 613, in query
    return self.read()
  File "pyvisa/resources/messagebased.py", line 427, in read
    message = self._read_raw().decode(enco)
  File "pyvisa/resources/messagebased.py", line 400, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "pyvisa-py/highlevel.py", line 403, in read
    raise errors.VisaIOError(ret[1])
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
>>>

arr-ee, 2023-08-21:
First off, you’re using a very old version of pyvisa on an old and unsupported version of python.

Next, have you checked if the instrument shows up when you list instruments? You’re running from root so device permissions shouldn’t be a problem.

arr-ee, 2023-08-21:
I’d suggest switching to a modern version of python & pyvisa, and trying again with `pyvisa.log_to_console()` for additional information.

jorsalas123, 2023-08-21:
Gimme some minutes to install python3 and add the pkgs

jorsalas123, 2023-08-21:
Now i tried with python3

root@PCBINT:~# python3
Python 3.8.10 (default, May 26 2023, 14:05:08)
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyvisa as visa
>>> rm = visa.ResourceManager("@py")
/usr/local/lib/python3.8/dist-packages/gpib_ctypes/gpib/gpib.py:54: UserWarning: GPIB library not found. Please manually load it using _load_lib(filename). All GPIB functions will raise OSError until the library is manually loaded.
  warnings.warn(message)
>>> rm.list_resources()
('ASRL/dev/ttyUSB0::INSTR',)
>>> inst = rm.open_resource('ASRL/dev/ttyUSB0::INSTR')
>>> inst.query("*IDN?")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.8/dist-packages/pyvisa/resources/messagebased.py", line 648, in query
    return self.read()
  File "/usr/local/lib/python3.8/dist-packages/pyvisa/resources/messagebased.py", line 486, in read
    message = self._read_raw().decode(enco)
  File "/usr/local/lib/python3.8/dist-packages/pyvisa/resources/messagebased.py", line 442, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "/usr/local/lib/python3.8/dist-packages/pyvisa_py/highlevel.py", line 519, in read
    return data, self.handle_return_value(session, status_code)
  File "/usr/local/lib/python3.8/dist-packages/pyvisa/highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
>>>

MatthieuDartiailh, 2023-08-22:
First, I suggest you confirm the serial configuration you use (baud rate, parity and data bits) from the manual the defaults should match pyvisa defaults but it is better to check. Second, the manual state on page 131:

> It is very important that you send the SYST:REM command to place the power
supply in the remote mode. Sending or receiving data over the RS-232 interface
when not configured for remote operation can cause unpredictable results

Does sending this command first or manually setting the device in remote mode helps ?