pyvisa/pyvisa-py #395: MacOS 12.7 w/ NI 488.2 Driver and GPIB?

pac1085, 2023-10-05:
Hi,
I'm attempting to use pyvisa-py with gpib on my macOS 12.7 machine (NI GPIB-USB-HS). 
I've installed NI 488.2 driver, and confirmed it's working thru the NI GPIB explorer, troubleshooter, etc.

I see the documentation for pyvisa-py it says "On MacOSX, you should install the NI-488 library from National instrument."

When I run pyvisa-info this is what it says.  How do I get it to detect the NI 488.2 driver that's installed on the system?  Thanks!

➜  bin ./pyvisa-info
Machine Details:
   Platform ID:    macOS-12.7-x86_64-i386-64bit
   Processor:      i386

Python:
   Implementation: CPython
   Executable:     /Library/Developer/CommandLineTools/usr/bin/python3
   Version:        3.9.6
   Compiler:       Clang 14.0.0 (clang-1400.0.29.202)
   Bits:           64bit
   Build:          Oct 18 2022 12:41:40 (#default)
   Unicode:        UCS4

PyVISA Version: 1.13.0

Backends:
   ivi:
      Version: 1.13.0 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.7.0
      TCPIP INSTR: Available 
         Resource discovery:
         - VXI-11: partial (psutil not installed)
         - hislip: disabled (zeroconf not installed)
      TCPIP SOCKET: Available 
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named 'serial'
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      VICP INSTR:
         Please install PyVICP to use this resource type.
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'
      GPIB INTFC:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'

MatthieuDartiailh, 2023-10-05:
As hinted by the message you get you need to install gpib-ctypes. It does not mention MacOs but it should.

pac1085, 2023-10-05:
Thanks for the reply.  I did notice that and installed gpib-ctypes.  Unfortunately it states that it cannot find the library, and when I try to load it manually with _load_lib, it throws an error.  Perhaps I need to open an issue with the gpib-ctypes GitHub, however I figured I'd start here since the docs for pyvisa-py specifically mentioned MacOs with NI 488, and I can't find any mention of it on the gpib-ctypes project page; they only mention Windows and Linux.

Python 3.9.6 (default, Oct 18 2022, 12:41:40) 
[Clang 14.0.0 (clang-1400.0.29.202)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import gpib_ctypes
/Users/philip/Library/Python/3.9/lib/python/site-packages/gpib_ctypes/gpib/gpib.py:54: UserWarning: GPIB library not found. Please manually load it using _load_lib(filename). All GPIB functions will raise OSError until the library is manually loaded.
  warnings.warn(message)
>>> import pyvisa

>>> gpib_ctypes.gpib.gpib._load_lib('/Library/Frameworks/NI4882.framework/NI4882')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/philipchapman/Library/Python/3.9/lib/python/site-packages/gpib_ctypes/gpib/gpib.py", line 91, in _load_lib
    libfunction = _lib[name]
  File "/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/ctypes/__init__.py", line 384, in __getitem__
    func = self._FuncPtr((name_or_ordinal, self))
AttributeError: dlsym(0x2170abf40, ibsre): symbol not found
>>>

berzoll, 2024-02-09:
hey there. I am having exactly the same issue. 

Did you figure out a solution to this?
The resources are listed correctly, but when I try loading some Keithley modules I get this error:

ValueError: gpib_ctypes is installed but could not locate the gpib library.
Please manually load it using:
  gpib_ctypes.gpib.gpib._load_lib(filename)
before importing pyvisa.

I cannot figure out any path that seems to be working as a valid library for gpib_ctypes.