pyvisa/pyvisa-py #398: [COM] Communication issue with Siglent SPS5051X using Ubuntu VM

ragrylesvani, 2023-10-24:
In the past I was able to get python to derive a lot of information about the connected device (such as Manufacturer ID), but now pyvis-py fails to identify anything about the device other than the bare basics.

[image]

PyUSB seems to be able to derive this information, and so is the Linux OS. I am wondering if there is an issue with how the Resource Class in instantiated, or how/why there is a failure to propagate this infromation up  into pyvisa-py?

Instrument details
------------------
* Model: Siglent SPS5051X
* Communication: USB
* https://int.siglent.com/u_file/download/22_06_28/SPS5000X_Usermanual_E02B.pdf

Output of `pyvisa-info`
-----------------------
Machine Details:
   Platform ID:    Linux-5.4.0-135-generic-x86_64-with-glibc2.27
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     ...bin/python
   Version:        3.8.0
   Compiler:       GCC 8.4.0
   Bits:           64bit
   Build:          Dec  9 2021 17:53:27 (#default)
   Unicode:        UCS4

PyVISA Version: 1.13.0

Backends:
   ivi:
      Version: 1.13.0 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.7.0
      USB INSTR: Available via PyUSB (1.2.1). Backend: libusb1
      USB RAW: Available via PyUSB (1.2.1). Backend: libusb1
      TCPIP INSTR: Available 
         Resource discovery:
         - VXI-11: partial (psutil not installed)
         - hislip: disabled (zeroconf not installed)
      TCPIP SOCKET: Available 
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named 'serial'
      VICP INSTR:
         Please install PyVICP to use this resource type.
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'
      GPIB INTFC:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'

MatthieuDartiailh, 2023-10-25:
I need to double-check the code, but right now, I suspect that the issue comes from the resource name looking weird (::0 before the INSTR). Typically the device manufacturer and device ID are not read from the bus since they are part of the resource name, but since here the name looks odd it may not have been parsed properly.

How did you obtain that resource name ?

ragrylesvani, 2023-10-26:
It seems the change is successful in populating the aforementioned fields.

[image]

Thank you for working on this!

MatthieuDartiailh, 2023-10-26:
I will merge and try to release soon

PR #399 (2023-10-26): Usb attr fixes
Closes #398
closed 2023-10-26 by MatthieuDartiailh