pyvisa/pyvisa-py #400: GPIB and SRQ

sjlukacs, 2023-10-30:
hello one and all,

i put this down for a while but now i'm back on it.  i'm using a raspberry pi 4b with raspbian buster 32-bit with linux-gpib and python3.7 on it, below.  pyvisa and pyvisa-py work great until we have to wait_for_srq() in order for the controller, i.e., the raspberry, to sit and wait for the device to call back saying it is ready to send its collected acquired data back to the controller for storage and/or processing.  i think i ran into these problems (below) some years ago, the dreaded "NotImplementedError" and i'm still having the same problems even though i've upgraded both pyvisa and pyvisa-py.  but after much toil, i was able to get a form of wait_for_srq to work under pure "import gpib".  not "import pyvisa or pyvisa-py or even Gpib.  so here are my questions:

am i missing any developments on pyvisa-py?

should i contribute to this project for the GPIB wait_for_ srq() to add that functionality?

is pyvisa-py had any movement forward to where we do not have to include the gpib kernel compiling or install the NI-GPIB drives so that in the future we need only install pyvisa-py and pyvisa via pip in order to access and use our gpib devices?

as referenced above:::

$ my_script_returned.py
pyvisa version: 1.12.0
KEITHLEY INSTRUMENTS INC.,MODEL 2000,0687508,A07  /A02
wait_for_timeout: 15000
Traceback (most recent call last):
  File "./prac0_gpib_waiton_2_pyvisa_keithleyDMM2000.py", line 30, in <module>
    keithley.wait_for_srq(wait_for_timeout)
  File "/usr/local/lib/python3.7/dist-packages/pyvisa/resources/gpib.py", line 66, in wait_for_srq
    constants.EventType.service_request, constants.EventMechanism.queue
  File "/usr/local/lib/python3.7/dist-packages/pyvisa/resources/resource.py", line 511, in enable_event
    self.visalib.enable_event(self.session, event_type, mechanism, context)
  File "/usr/local/lib/python3.7/dist-packages/pyvisa/highlevel.py", line 909, in enable_event
    raise NotImplementedError
NotImplementedError

and...
$ pyvisa-info 
Machine Details:
   Platform ID:    Linux-5.10.17-v7l+-armv7l-with-debian-10.9
   Processor:      

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.7.3
   Compiler:       GCC 8.3.0
   Bits:           32bit
   Build:          Jan 22 2021 20:04:44 (#default)
   Unicode:        UCS4

PyVISA Version: 1.12.0

Backends:
   ivi:
      Version: 1.12.0 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.6.3
      ASRL INSTR: Available via PySerial (3.4)
      USB INSTR: Available via PyUSB (1.2.1). Backend: libusb1
      USB RAW: Available via PyUSB (1.2.1). Backend: libusb1
      TCPIP INSTR: Available 
         Resource discovery:
         - VXI-11: ok
         - hislip: disabled (zeroconf not installed)
      TCPIP SOCKET: Available 
      GPIB INSTR: Available via Linux GPIB (b'4.3.4')
      GPIB INTFC: Available via Linux GPIB (b'4.3.4')

MatthieuDartiailh, 2023-10-31:
Hi,

PyVISA-py does not support so far any event mechanism, which is why you are reaching the NotImplementedError. If you are able to contribute some work in that direction it would be nice.

GPIB communication requires a driver for the GPIB board/adapter that cannot be implemented in pure Python (at least that's my understanding), so you will always need to install it separately using either linux-gpib or a VISA implementation.

dirkjankrijnders, 2023-11-13:
Hi,

I'm running into the same issue and I can contribute with coding and testing. I have a linux-gpib setup with several older IEEE 488.1 devices that I can test on. I read the log of issue #142 and the accompanying PR, is that still a valid starting point for implementing this?

hinxx, 2024-09-12:
@MatthieuDartiailh ,  @dirkjankrijnders I'm interested in seeing events supported for USBTMC devices. Would GPIB code be of use as a starting point?