pyvisa/pyvisa-py #403: M2 Mac - list_resources() not returning available TCPIP instrument

dstrande, 2023-11-27:
On an M2 Mac, `rm.list_resources()` isn't returning an available TCPIP instrument.

```
rm = pyvisa.ResourceManager("@py")
print(rm.list_resources())

inst = rm.open_resource("TCPIP::169.254.4.179::INSTR")
print(inst.query("*IDN?"))
```

Returns:

```
()
KEITHLEY INSTRUMENTS,MODEL 2450,04505775,1.7.12b
```

Showing the instrument will connect but fails to show for list_resources.

I tried doing a bit of debugging but not totally sure what I was doing:

`resp = pmap.recv_port((vxi11.DEVICE_CORE_PROG, vxi11.DEVICE_CORE_VERS, rpc.IPPROTO_TCP, 0))`

seems to fail to return any response on Mac but succeeds on Windows. 

To Reproduce
--------------
Steps to reproduce the behavior:

1. Connect instrument withFind IP address of your instrument
2. Use code above with a ARM Mac (?) replacing IP address with your instrument's

**Output of `pyvisa-info`**
```
Machine Details:
   Platform ID:    macOS-13.5-arm64-arm-64bit
   Processor:      arm

Python:
   Implementation: CPython
   Executable:     /***
   Version:        3.11.6
   Compiler:       Clang 15.0.7 
   Architecture:   ('arm', 64)
   Build:          Oct  3 2023 10:37:07 (#main)
   Unicode:        UCS4

PyVISA Version: 1.14.1

Backends:
   ivi:
      Version: 1.14.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.7.1
      TCPIP INSTR: Available 
         Resource discovery:
         - VXI-11: ok
         - hislip: ok
      TCPIP SOCKET: Available 
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named 'serial'
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      VICP INSTR:
         Please install PyVICP to use this resource type.
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'
      GPIB INTFC:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'
```

MatthieuDartiailh, 2023-11-28:
Could instrument a bit the code so that we get a better idea of what is going on ?
I think it may be helpful to:
- print/log the list of discovery addresses over which we broadcast
- check the pmap_list to see if broadcast fails on any address
- increase the sleep duration

MatthieuDartiailh, 2023-11-29:
Reading my previous message I realize I was a vague. The function to instrument is the following: https://github.com/pyvisa/pyvisa-py/blob/main/pyvisa_py/tcpip.py#L394

dstrande, 2023-11-29:
It seems like the send succeeds but doesn't receive a response.

I pulled out this part of the code and ran it:

```python
import socket
import ipaddress
from pyvisa_py.protocols import vxi11, rpc
import time
import psutil

broadcast_addr = []
for interface, snics in psutil.net_if_addrs().items():
    for snic in snics:
        if snic.family is socket.AF_INET:
            addr = snic.address
            mask = snic.netmask
            network = ipaddress.IPv4Network(addr + "/" + mask, strict=False)
            broadcast_addr.append(str(network.broadcast_address))

print("addresses: ", broadcast_addr)
pmap_list = [rpc.BroadcastUDPPortMapperClient(ip) for ip in broadcast_addr]
for pmap in list(pmap_list):
    pmap.set_timeout(0)
    pmap.send_port((vxi11.DEVICE_CORE_PROG, vxi11.DEVICE_CORE_VERS, rpc.IPPROTO_TCP, 0))

# Timeout for responses
print("pmap send: ", pmap_list)
time.sleep(5)

all_res = []
for pmap in pmap_list:
    resp = pmap.recv_port(
        (vxi11.DEVICE_CORE_PROG, vxi11.DEVICE_CORE_VERS, rpc.IPPROTO_TCP, 0)
    )
    print("response: ", resp)

    res = [r[1][0] for r in resp if r[0] > 0]
    res = sorted(res, key=lambda ip: tuple(int(part) for part in ip.split(".")))
    # TODO: Detect GPIB over TCPIP
    res = ["TCPIP::{}::INSTR".format(host) for host in res]
    all_res.extend(res)
```

Which returns: 

```
addresses:  ['127.255.255.255', '10.0.0.255', '169.254.255.255']
pmap send:  [<pyvisa_py.protocols.rpc.BroadcastUDPPortMapperClient object at 0x10c099650>, <pyvisa_py.protocols.rpc.BroadcastUDPPortMapperClient object at 0x105d77510>, <pyvisa_py.protocols.rpc.BroadcastUDPPortMapperClient object at 0x106a6c090>]
response:  []
response:  []
response:  []
```

dstrande, 2023-11-29:
Also here's the output on Windows:

```
addresses:  ['169.254.255.255', '10.0.0.255', '169.254.255.255', '127.255.255.255']
pmap send:  [<pyvisa_py.protocols.rpc.BroadcastUDPPortMapperClient object at 0x000001A8746D7650>, <pyvisa_py.protocols.rpc.BroadcastUDPPortMapperClient object at 0x000001A8748FC710>, <pyvisa_py.protocols.rpc.BroadcastUDPPortMapperClient object at 0x000001A874695F50>, <pyvisa_py.protocols.rpc.BroadcastUDPPortMapperClient object at 0x000001A8746A0BD0>]
response:  [(621, ('169.254.145.144', 111))]
response:  []
response:  [(621, ('169.254.145.144', 111))]
response:  []
```

MatthieuDartiailh, 2023-11-29:
Thanks ! Having the M2/Windows comparison is nice.

So we know that we are indeed sending requests and that we would be able to reach the instrument since we address the right subnet. So we need to understand why we do not get any response on Mac. I am not familiar with MacOS firewall settings but it may worth checking that it is not filtering either the query or responses.

dstrande, 2023-11-29:
I'm also not too familiar. Seems like my firewall is disabled though. If I ping the instrument I get a response as well:

```
PING 169.254.145.144 (169.254.145.144): 56 data bytes
64 bytes from 169.254.145.144: icmp_seq=0 ttl=64 time=0.567 ms
64 bytes from 169.254.145.144: icmp_seq=1 ttl=64 time=0.586 ms
64 bytes from 169.254.145.144: icmp_seq=2 ttl=64 time=0.589 ms
64 bytes from 169.254.145.144: icmp_seq=3 ttl=64 time=0.631 ms
```

[image]

MatthieuDartiailh, 2023-11-29:
Since I do not have a Mac on hand this is getting hard. Could you try to install wireshark and look at the traffic on the subnet 169.254 ?

dstrande, 2023-11-29:
Not fully sure what I'm looking at but the packets in blue occur regularly and the packet in yellow is from the script above.

[image]

dstrande, 2023-11-30:
Unfortunately I don't have access to linux. I can get a usb boot if you think it'll make a difference. 

On Windows it's different, unsurprisingly I guess. The script starts with No. 43.

[image]

dstrande, 2023-11-30:
Also I've confirmed the 169.254 subnet on mac:

```
% ipconfig getifaddr en6
169.254.185.47
```

dstrande, 2023-11-30:
Maybe this isn't surprising, but I changed the test script above to include the devices address from the start ( line 8 `broadcast_addr = ["169.254.145.145"]`). And this is what is returned:

```
addresses:  ['169.254.145.145', '127.255.255.255', '10.0.0.255', '169.254.255.255']
pmap send:  [<pyvisa_py.protocols.rpc.BroadcastUDPPortMapperClient object at 0x10124fe10>, <pyvisa_py.protocols.rpc.BroadcastUDPPortMapperClient object at 0x100daf590>, <pyvisa_py.protocols.rpc.BroadcastUDPPortMapperClient object at 0x1026c9650>, <pyvisa_py.protocols.rpc.BroadcastUDPPortMapperClient object at 0x100c960d0>]
b'\x00\x00\x00\x01\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02Y' ('169.254.145.145', 111)
response:  [(601, ('169.254.145.145', 111))]
response:  []
response:  []
response:  []
```

CompThing, 2023-11-30:
Hi Dallas,
have you tried putting the computers on a network with some kind of
router present?
Using APIPA addresses 169.254.x.x might behave differently to using a
private subnet with addresses like 10.0.x.x or 192.168.x.x.
A network broadcast to the APIPA broadcast address 169.254.255.255
might be treated as a special case by a firewall,

Michael Wilkinson

MatthieuDartiailh, 2023-12-01:
@CompThing may well be onto something. It is at the very least worth checking and documenting if it turns out it is the root cause.

dstrande, 2023-12-01:
Yes you're right. If I plug something into my network instead of directly into the Mac, it shows up for list_resources:

`('ASRL/dev/cu.Bluetooth-Incoming-Port::INSTR', 'TCPIP::10.0.0.219::INSTR')`

CompThing, 2023-12-01:
> No unfortunately. I did try that originally and it still doesn't
> work.

You would also need to give the test instrument a static IP address
within the same subnet for it to work

dstrande, 2023-12-01:
Yeah 169.254.100.101 / 255.255.0.0 and 169.254.145.45 / 255.255.0.0 on the instrument. No gateway needed correct?

[image]

CompThing, 2023-12-01:
Hi Dallas,
if you make the addresses:
192.168.1.100/255.255.255.0
and
192.168.1.101/255.255.255.0
they will both have static IP addresses on the same private subnet. No
gateway needed, but if the config requires one, you could set it as
192.168.1.1

169.255.x.x addresses should not be assigned statically and are not a
normal subnet. They belong to a special set of addresses for APIPA
which
is a very old Windows way of auto-negotiating addresses without a
proper network. It does not play well with modern networking and should
be avoided.

Hope that helps,
Michael Wilkinson

MatthieuDartiailh, 2023-12-01:
I think it is not so much a question of mask than it is a question of the ip address prefix. As @CompThing pointed out the 169.255 prefix is special and this appears to be the root issue of your problem. 

Could you make a PR adding this piece of information to the documentation ?