pyvisa/pyvisa-py #410: Reading (query) from TCPIP device always gets Timeout error

schlatterbeck, 2024-02-02:
Opening a TCPIP resource works partially: Some commands can be sent (and have an effect on a Rigol DS2072 oscilloscope) but trying to read anything (e.g. with 'query') produce a timeout. Running a tcpdump in the background sees packets going to and from the device. Even when I specify a *long* timeout (30s) as second argument to query, I'm still getting a timeout. The name `oszi` is the DNS name of my oscilloscope.

To Reproduce
--------------
```
>>> import pyvisa
>>> rm = pyvisa.ResourceManager('@py')
>>> inst = rm.open_resource('TCPIP::oszi')
>>> # This seems to work:
>>> inst.write (':CHAN1:SCALE 3.0')
18
>>> # But this doesn't (no effect):
>>> inst.write (':STOP')
7
>>> # And query produces a timeout
>>> inst.query('*IDN?')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/ralf/.virtualenvs/visa/lib/python3.11/site-packages/pyvisa/resources/messagebased.py", line 647, in query
    return self.read()
           ^^^^^^^^^^^
  File "/home/ralf/.virtualenvs/visa/lib/python3.11/site-packages/pyvisa/resources/messagebased.py", line 485, in read
    message = self._read_raw().decode(enco)
              ^^^^^^^^^^^^^^^^
  File "/home/ralf/.virtualenvs/visa/lib/python3.11/site-packages/pyvisa/resources/messagebased.py", line 441, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ralf/.virtualenvs/visa/lib/python3.11/site-packages/pyvisa_py/highlevel.py", line 519, in read
    return data, self.handle_return_value(session, status_code)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ralf/.virtualenvs/visa/lib/python3.11/site-packages/pyvisa/highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
```

**Output of `pyvisa-info`**

```
Machine Details:
   Platform ID:    Linux-6.1.0-13-amd64-x86_64-with-glibc2.36
   Processor:      

Python:
   Implementation: CPython
   Executable:     /home/ralf/.virtualenvs/visa/bin/python3
   Version:        3.11.2
   Compiler:       GCC 12.2.0
   Architecture:   ('x86', 64)
   Build:          Mar 13 2023 12:18:29 (#main)
   Unicode:        UCS4

PyVISA Version: 1.14.1

Backends:
   ivi:
      Version: 1.14.1 (bundled with PyVISA)
      #1: /usr/local/lib/libvisa.so.0.0.0:
         found by: auto
         architecture:
            ('x86', 64)
         Could not get more info:
            VI_ERROR_NSUP_ATTR (-1073807331): The specified attribute is not defined or supported by the referenced object.
   py:
      Version: 0.7.1
      TCPIP INSTR: Available 
         Resource discovery:
         - VXI-11: partial (psutil not installed)
         - hislip: disabled (zeroconf not installed)
      TCPIP SOCKET: Available 
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named 'serial'
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      VICP INSTR:
         Please install PyVICP to use this resource type.
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'
      GPIB INTFC:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'
```

schlatterbeck, 2024-02-02:
I've traced this with tcpdump. The working version (ds2000-shell, a c++ app based on librevisa) does not append '\r\n' to the command sent while pyvisa-py apparently automagically appends '\r\n' to my command. So far I see no further difference in the packets, it seems with carriage-return and newline appended my oscilloscope simply refuses to provide any output.

Note that the timeout error is produced on the oscilloscope side -- the answer returned by the read command contains the error.

Is there a way to turn off the auto-append of '\r\n' in pyvisa-py?

schlatterbeck, 2024-02-02:
Hypothesis confirmed, my device doesn't like the `write_termination` with '\r\n':
```
>>> import pyvisa
>>> rm = pyvisa.ResourceManager('@py')
>>> inst = rm.open_resource('TCPIP::oszi')
>>> inst.write_termination = ''
>>> inst.query('*IDN?')
'RIGOL TECHNOLOGIES,DS2...,........,........\n'
```
[Serial number in device output censored]
Now the question is: Is this a bug or a feature? Of pyvisa or of my device?
Maybe the `write_termination` should be set to the empty string for TCPIP?

MatthieuDartiailh, 2024-02-02:
Changing the default is likely to cause more churn than it is worth. I advise to always set explicitly the desired read and write termination.

schlatterbeck, 2024-02-02:
Thanks for the quick feedback.
I guess this should be documented.
So far the documentation provided on readthedocs is very brief, I'd put a ref to the write_termination there on the first page (index.rst) with the example mentioning TCP?