pyvisa/pyvisa-py #422: [COM] Performance issue - slow transfer - USB

0xDEADBEEF-ASAP, 2024-03-28:
Hi! SDS1202XE , USB, pyvisa, pyvisa-py, arch Linux (os not virtualized), python running in venv.

750.1 KiB / 1 min transfer rate.

Can it be improved? What can I change or check to make it faster?

[image]
[image]
[image]

MatthieuDartiailh, 2024-04-01:
Can you post the output of ptvisa-info and try to downgrade to 0.7.1 ?
 
0.7.2 introduce a fix that may be responsible for the slow transfer you observe.

0xDEADBEEF-ASAP, 2024-04-03:
Hi! pyvisa-py 0.7.1 it download screenshot super fast but first screenshot is incomplete 200kb , next screenshots are 60kb - full screenshot should be ~750kb).

Pyvisa-info:

Machine Details:
   Platform ID:    Linux-6.8.2-arch2-1-x86_64-with-glibc2.39
   Processor:      

Python:
   Implementation: CPython
   Executable:     /home/arch/test2/.venv/bin/python
   Version:        3.11.8
   Compiler:       GCC 13.2.1 20230801
   Architecture:   ('x86', 64)
   Build:          Feb 12 2024 14:50:05 (#main)
   Unicode:        UCS4

PyVISA Version: 1.14.1

Backends:
   ivi:
      Version: 1.14.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.7.1
      USB INSTR: Available via PyUSB (1.2.1). Backend: libusb1
      USB RAW: Available via PyUSB (1.2.1). Backend: libusb1
      TCPIP INSTR: Available 
         Resource discovery:
         - VXI-11: partial (psutil not installed)
         - hislip: disabled (zeroconf not installed)
      TCPIP SOCKET: Available 
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named 'serial'
      VICP INSTR:
         Please install PyVICP to use this resource type.
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'
      GPIB INTFC:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'

And log file:

(.venv) [arch@archlinux test2]$ sudo python ./3-sds-pyvisa.py
2024-04-03 06:45:14,175 - pyvisa - DEBUG - SerialSession was not imported No module named 'serial'.
2024-04-03 06:45:14,223 - pyvisa - DEBUG - USBSession and USBRawSession were correctly imported.
2024-04-03 06:45:14,235 - pyvisa - DEBUG - TCPIPSession was correctly imported.
2024-04-03 06:45:14,237 - pyvisa - DEBUG - GPIBSession was not imported No module named 'gpib'.
2024-04-03 06:45:14,238 - pyvisa - DEBUG - Created library wrapper for py
2024-04-03 06:45:14,239 - pyvisa - DEBUG - Created ResourceManager with session 3509306
2024-04-03 06:45:14,240 - pyvisa - DEBUG - USB0::62701::60986::SDS1EBAQ1R3930::0::INSTR - opening ...
2024-04-03 06:45:14,297 - pyvisa - DEBUG - USB0::62701::60986::SDS1EBAQ1R3930::0::INSTR - is open with session 6351064
2024-04-03 06:45:14,298 - pyvisa - DEBUG - USB0::62701::60986::SDS1EBAQ1R3930::0::INSTR - reading 1048576 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
/home/arch/test2/.venv/lib/python3.11/site-packages/pyvisa_py/protocols/usbtmc.py:115: UserWarning: Unexpected MsgID format. Consider updating the device's firmware. See https://github.com/pyvisa/pyvisa-py/issues/20
  warnings.warn(
2024-04-03 06:45:14,366 - pyvisa - DEBUG - USB0::62701::60986::SDS1EBAQ1R3930::0::INSTR - reading 1048576 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2024-04-03 06:45:14,376 - pyvisa - DEBUG - USB0::62701::60986::SDS1EBAQ1R3930::0::INSTR - closing
2024-04-03 06:45:14,377 - pyvisa - DEBUG - USB0::62701::60986::SDS1EBAQ1R3930::0::INSTR - is closed
2024-04-03 06:45:14,377 - pyvisa - DEBUG - Closing ResourceManager (session: 3509306)
(.venv) [arch@archlinux test2]$ sudo python ./3-sds-pyvisa.py
2024-04-03 06:45:33,084 - pyvisa - DEBUG - SerialSession was not imported No module named 'serial'.
2024-04-03 06:45:33,130 - pyvisa - DEBUG - USBSession and USBRawSession were correctly imported.
2024-04-03 06:45:33,142 - pyvisa - DEBUG - TCPIPSession was correctly imported.
2024-04-03 06:45:33,145 - pyvisa - DEBUG - GPIBSession was not imported No module named 'gpib'.
2024-04-03 06:45:33,146 - pyvisa - DEBUG - Created library wrapper for py
2024-04-03 06:45:33,146 - pyvisa - DEBUG - Created ResourceManager with session 2279201
2024-04-03 06:45:33,147 - pyvisa - DEBUG - USB0::62701::60986::SDS1EBAQ1R3930::0::INSTR - opening ...
2024-04-03 06:45:33,169 - pyvisa - DEBUG - USB0::62701::60986::SDS1EBAQ1R3930::0::INSTR - is open with session 7938331
2024-04-03 06:45:33,171 - pyvisa - DEBUG - USB0::62701::60986::SDS1EBAQ1R3930::0::INSTR - reading 1048576 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
/home/arch/test2/.venv/lib/python3.11/site-packages/pyvisa_py/protocols/usbtmc.py:115: UserWarning: Unexpected MsgID format. Consider updating the device's firmware. See https://github.com/pyvisa/pyvisa-py/issues/20
  warnings.warn(
2024-04-03 06:45:33,185 - pyvisa - DEBUG - USB0::62701::60986::SDS1EBAQ1R3930::0::INSTR - reading 1048576 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2024-04-03 06:45:33,195 - pyvisa - DEBUG - USB0::62701::60986::SDS1EBAQ1R3930::0::INSTR - closing
2024-04-03 06:45:33,196 - pyvisa - DEBUG - USB0::62701::60986::SDS1EBAQ1R3930::0::INSTR - is closed
2024-04-03 06:45:33,197 - pyvisa - DEBUG - Closing ResourceManager (session: 2279201)
(.venv) [arch@archlinux test2]$

0xDEADBEEF-ASAP, 2024-04-18:
Updated BIOS to latest - but there is the same issue. Maybe I have to change some BIOS settings?

Jimmyvandenbergh, 2024-12-07:
@0xDEADBEEF-ASAP would you be able to test the PR #470. This will read the data again with the large chunks as the 0.7.1 version of pyvisa-py and should solve the slow read speed.