pyvisa/pyvisa-py #428: Calling open_resource with a hislip address when the instrument is not available causes hang (works ok using Keysight backend)

DareDevilDenis, 2024-05-21:
The code below attempts to connect to a non-existent hislip instrument "hislip999" on the local machine. The call to 'open_resource' hangs.

To Reproduce
--------------
```python
import pyvisa

rm = pyvisa.ResourceManager("@py")   # If I remove "@py" then it works i.e. the connection attempt immediately fails
my_instrument = rm.open_resource("TCPIP0::127.0.0.1::hislip999::INSTR")
```

If I remove `"@py"` so that it uses my installation of Keysight IVI backend then it works as expected - we get immediate exception `pyvisa.errors.VisaIOError: VI_ERROR_RSRC_NFOUND (-1073807343): Insufficient location information or the requested device or resource is not present in the system`

**Output of `pyvisa-info`**
```
Machine Details:
   Platform ID:    Windows-10-10.0.19045-SP0
   Processor:      Intel64 Family 6 Model 141 Stepping 1, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Program Files\Python312\python.exe
   Version:        3.12.0
   Compiler:       MSC v.1935 64 bit (AMD64)
   Architecture:   AMD64
   Build:          Oct  2 2023 13:03:39 (#tags/v3.12.0:0fb18b0)
   Unicode:        UCS4

PyVISA Version: 1.14.1

Backends:
   ivi:
      Version: 1.14.1 (bundled with PyVISA)
      #1: C:\Windows\system32\visa32.dll:
         found by: auto
         architecture:
            ('x86', 64)
         Vendor: Keysight Technologies
         Impl. Version: 1377199637
         Spec. Version: 5244928
      #2: C:\Windows\system32\visa64.dll:
         found by: auto
         architecture:
            ('x86', 64)
         Vendor: Keysight Technologies
         Impl. Version: 1377199637
         Spec. Version: 5244928
   py:
      Version: 0.7.2
      TCPIP INSTR: Available
         Resource discovery:
         - VXI-11: partial (psutil not installed)
         - hislip: disabled (zeroconf not installed)
      TCPIP SOCKET: Available
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named 'serial'
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      VICP INSTR:
         Please install PyVICP to use this resource type.
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'
      GPIB INTFC:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'
```

MatthieuDartiailh, 2024-05-27:
Indeed hslip currently ignore the open_timeout. Please have a look at #430

DareDevilDenis, 2024-06-06:
Hi @MatthieuDartiailh
I don't have enough knowledge of pyvisa-py to be able to review your PR however I tried your fix by installing it as follows:

`pip install git+https://github.com/pyvisa/pyvisa-py@hislip-connect-timeout`

I repeated the above test, trying to connect to a non-existent hislip instrument ("hislip999") and it threw the following exception:

```
Traceback (most recent call last):
  File "C:\temp\test_timeout.py", line 4, in <module>
    my_instrument = rm.open_resource("TCPIP0::127.0.0.1::hislip999::INSTR")
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python312\Lib\site-packages\pyvisa\highlevel.py", line 3292, in open_resource
    res.open(access_mode, open_timeout)
  File "C:\Program Files\Python312\Lib\site-packages\pyvisa\resources\resource.py", line 281, in open
    self.session, status = self._resource_manager.open_bare_resource(
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python312\Lib\site-packages\pyvisa\highlevel.py", line 3217, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python312\Lib\site-packages\pyvisa_py\highlevel.py", line 168, in open
    sess = cls(session, resource_name, parsed, open_timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python312\Lib\site-packages\pyvisa_py\tcpip.py", line 86, in __new__
    return newcls(resource_manager_session, resource_name, parsed, open_timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python312\Lib\site-packages\pyvisa_py\sessions.py", line 322, in __init__
    self.after_parsing()
  File "C:\Program Files\Python312\Lib\site-packages\pyvisa_py\tcpip.py", line 138, in after_parsing
    self.interface = hislip.Instrument(
                     ^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python312\Lib\site-packages\pyvisa_py\protocols\hislip.py", line 386, in __init__
    self._sync.connect((ip_addr, port))
BlockingIOError: [WinError 10035] A non-blocking socket operation could not be completed immediately
```

I initially thought that this was the expected result however when I repeated the test with an available instrument it threw a similar exception (not exactly the same exception traceback):

```
Traceback (most recent call last):
  File "C:\temp\valid_address.py", line 4, in <module>
    my_instrument = rm.open_resource("TCPIP0::127.0.0.1::hislip2::INSTR")
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python312\Lib\site-packages\pyvisa\highlevel.py", line 3292, in open_resource
    res.open(access_mode, open_timeout)
  File "C:\Program Files\Python312\Lib\site-packages\pyvisa\resources\resource.py", line 281, in open
    self.session, status = self._resource_manager.open_bare_resource(
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python312\Lib\site-packages\pyvisa\highlevel.py", line 3217, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python312\Lib\site-packages\pyvisa_py\highlevel.py", line 168, in open
    sess = cls(session, resource_name, parsed, open_timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python312\Lib\site-packages\pyvisa_py\tcpip.py", line 86, in __new__
    return newcls(resource_manager_session, resource_name, parsed, open_timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python312\Lib\site-packages\pyvisa_py\sessions.py", line 322, in __init__
    self.after_parsing()
  File "C:\Program Files\Python312\Lib\site-packages\pyvisa_py\tcpip.py", line 138, in after_parsing
    self.interface = hislip.Instrument(
                     ^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python312\Lib\site-packages\pyvisa_py\protocols\hislip.py", line 386, in __init__
    self._sync.connect((ip_addr, port))
BlockingIOError: [WinError 10035] A non-blocking socket operation could not be completed immediately
```

MatthieuDartiailh, 2024-06-17:
Sorry I have been quite busy lately with little time for OSS. I will try to look into it this week.

MatthieuDartiailh, 2024-09-17:
I finally managed to get back to this. Could you give this another spin ?

MatthieuDartiailh, 2024-09-17:
ACtually I broke some things sorry. I will let you when you can test.

MatthieuDartiailh, 2024-09-18:
Ready for some tests

DareDevilDenis, 2024-10-01:
Hi @MatthieuDartiailh 
It didn't work for me.

When using a non-existent hislip address "hislip999" I got:

```
>py test_timeout.py
Exception ignored in atexit callback: <function ResourceManager.__new__.<locals>.call_close at 0x000002821D639DA0>
Traceback (most recent call last):
  File "C:\Program Files\Python312\Lib\site-packages\pyvisa\highlevel.py", line 3014, in call_close
    meth()
  File "C:\Program Files\Python312\Lib\site-packages\pyvisa\highlevel.py", line 3081, in close
    resource.close()
  File "C:\Program Files\Python312\Lib\site-packages\pyvisa\resources\resource.py", line 318, in close
    self.visalib.close(self.session)
  File "C:\Program Files\Python312\Lib\site-packages\pyvisa_py\highlevel.py", line 447, in close
    return self.handle_return_value(session, StatusCode.error_invalid_object)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python312\Lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_INV_OBJECT (-1073807346): The given session or object reference is invalid.
Exception ignored in: <function Resource.__del__ at 0x000002821D63A480>
Traceback (most recent call last):
  File "C:\Program Files\Python312\Lib\site-packages\pyvisa\resources\resource.py", line 179, in __del__
  File "C:\Program Files\Python312\Lib\site-packages\pyvisa\resources\resource.py", line 318, in close
  File "C:\Program Files\Python312\Lib\site-packages\pyvisa_py\highlevel.py", line 447, in close
  File "C:\Program Files\Python312\Lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
pyvisa.errors.VisaIOError: VI_ERROR_INV_OBJECT (-1073807346): The given session or object reference is invalid.
```
I'm not sure if this is the exception you are expecting or not.

When using a valid hislip address the call to open_resource doesn't throw an exception however if I then call:
`my_instrument.read_termination = '\n'`

It throws exception:
```
Traceback (most recent call last):
  File "C:\Program Files\Python312\Lib\site-packages\pyvisa_py\highlevel.py", line 664, in set_attribute
    self.sessions[session].set_attribute(attribute, attribute_state),
    ~~~~~~~~~~~~~^^^^^^^^^
KeyError: 0

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "valid_address_hccu.py", line 6, in <module>
    my_instrument.read_termination = '\n'
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python312\Lib\site-packages\pyvisa\resources\messagebased.py", line 108, in read_termination
    self.set_visa_attribute(
  File "C:\Program Files\Python312\Lib\site-packages\pyvisa\resources\resource.py", line 377, in set_visa_attribute
    return self.visalib.set_attribute(self.session, name, state)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python312\Lib\site-packages\pyvisa_py\highlevel.py", line 667, in set_attribute
    return self.handle_return_value(session, StatusCode.error_invalid_object)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python312\Lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_INV_OBJECT (-1073807346): The given session or object reference is invalid.
Exception ignored in atexit callback: <function ResourceManager.__new__.<locals>.call_close at 0x000002545AE09DA0>
Traceback (most recent call last):
  File "C:\Program Files\Python312\Lib\site-packages\pyvisa\highlevel.py", line 3014, in call_close
    meth()
  File "C:\Program Files\Python312\Lib\site-packages\pyvisa\highlevel.py", line 3081, in close
    resource.close()
  File "C:\Program Files\Python312\Lib\site-packages\pyvisa\resources\resource.py", line 318, in close
    self.visalib.close(self.session)
  File "C:\Program Files\Python312\Lib\site-packages\pyvisa_py\highlevel.py", line 447, in close
    return self.handle_return_value(session, StatusCode.error_invalid_object)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python312\Lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_INV_OBJECT (-1073807346): The given session or object reference is invalid.
Exception ignored in: <function Resource.__del__ at 0x000002545AE0A480>
Traceback (most recent call last):
  File "C:\Program Files\Python312\Lib\site-packages\pyvisa\resources\resource.py", line 179, in __del__
  File "C:\Program Files\Python312\Lib\site-packages\pyvisa\resources\resource.py", line 318, in close
  File "C:\Program Files\Python312\Lib\site-packages\pyvisa_py\highlevel.py", line 447, in close
  File "C:\Program Files\Python312\Lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
pyvisa.errors.VisaIOError: VI_ERROR_INV_OBJECT (-1073807346): The given session or object reference is invalid.
```

MatthieuDartiailh, 2024-10-01:
In both case the exception originate from closing the resource but not when opening it. Could you share your test script ?

MatthieuDartiailh, 2024-10-01:
This is the traceback I get trying to connect to a random hislip resource:
```
Traceback (most recent call last):
  File "C:\Users\DartiailhMatthieu\Documents\Coding\Projects\pyvisa-py\pyvisa_py\tcpip.py", line 478, in after_parsing
    self.interface = Vxi11CoreClient(host_address, port, self.open_timeout)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\DartiailhMatthieu\Documents\Coding\Projects\pyvisa-py\pyvisa_py\tcpip.py", line 382, in __init__
    rpc.TCPClient.__init__(self, host, prog, vers, open_timeout)
  File "C:\Users\DartiailhMatthieu\Documents\Coding\Projects\pyvisa-py\pyvisa_py\protocols\rpc.py", line 823, in __init__
    pmap = TCPPortMapperClient(host, open_timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\DartiailhMatthieu\Documents\Coding\Projects\pyvisa-py\pyvisa_py\protocols\rpc.py", line 803, in __init__
    RawTCPClient.__init__(self, host, PMAP_PROG, PMAP_VERS, PMAP_PORT, open_timeout)
  File "C:\Users\DartiailhMatthieu\Documents\Coding\Projects\pyvisa-py\pyvisa_py\protocols\rpc.py", line 454, in __init__
    self.connect(1e-3 * open_timeout)
  File "C:\Users\DartiailhMatthieu\Documents\Coding\Projects\pyvisa-py\pyvisa_py\protocols\rpc.py", line 488, in connect
    raise RPCError("can't connect to server")
pyvisa_py.protocols.rpc.RPCError: can't connect to server

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\DartiailhMatthieu\Documents\Coding\Projects\pyvisa\pyvisa\highlevel.py", line 3293, in open_resource
    res.open(access_mode, open_timeout)
  File "C:\Users\DartiailhMatthieu\Documents\Coding\Projects\pyvisa\pyvisa\resources\resource.py", line 282, in open
    self.session, status = self._resource_manager.open_bare_resource(
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\DartiailhMatthieu\Documents\Coding\Projects\pyvisa\pyvisa\highlevel.py", line 3218, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\DartiailhMatthieu\Documents\Coding\Projects\pyvisa-py\pyvisa_py\highlevel.py", line 168, in open
    sess = cls(session, resource_name, parsed, open_timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\DartiailhMatthieu\Documents\Coding\Projects\pyvisa-py\pyvisa_py\tcpip.py", line 86, in __new__
    return newcls(resource_manager_session, resource_name, parsed, open_timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\DartiailhMatthieu\Documents\Coding\Projects\pyvisa-py\pyvisa_py\sessions.py", line 322, in __init__
    self.after_parsing()
  File "C:\Users\DartiailhMatthieu\Documents\Coding\Projects\pyvisa-py\pyvisa_py\tcpip.py", line 480, in after_parsing
    raise errors.VisaIOError(constants.VI_ERROR_RSRC_NFOUND)
pyvisa.errors.VisaIOError: VI_ERROR_RSRC_NFOUND (-1073807343): Insufficient location information or the requested device or resource is not present in the system.
```
For comparison I get the following when using the IVI backend:
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\DartiailhMatthieu\Documents\Coding\Projects\pyvisa\pyvisa\highlevel.py", line 3293, in open_resource
    res.open(access_mode, open_timeout)
  File "C:\Users\DartiailhMatthieu\Documents\Coding\Projects\pyvisa\pyvisa\resources\resource.py", line 282, in open
    self.session, status = self._resource_manager.open_bare_resource(
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\DartiailhMatthieu\Documents\Coding\Projects\pyvisa\pyvisa\highlevel.py", line 3218, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\DartiailhMatthieu\Documents\Coding\Projects\pyvisa\pyvisa\ctwrapper\functions.py", line 1853, in open
    ret = library.viOpen(
          ^^^^^^^^^^^^^^^
  File "C:\Users\DartiailhMatthieu\Documents\Coding\Projects\pyvisa\pyvisa\ctwrapper\highlevel.py", line 227, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\DartiailhMatthieu\Documents\Coding\Projects\pyvisa\pyvisa\highlevel.py", line 252, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_RSRC_NFOUND (-1073807343): Insufficient location information or the requested device or resource is not present in the system.
```
The errors are identical in term of error code. I chose to provide more in depth view in the case of pyvisa-py since I consider it potentially useful when debugging.

Finallt note that I did not exactly follow the spec since from the spec the open_timeout argument only apply to how long to wait to acquire a lock and is not related to opening of the underlying socket.

DareDevilDenis, 2024-10-02:
Here's my script with an invalid hislip address:
```python
import pyvisa

rm = pyvisa.ResourceManager("@py")
my_instrument = rm.open_resource("TCPIP0::127.0.0.1::hislip999::INSTR")
```

Here's my script with corrrect hislip address:
```python
import pyvisa

rm = pyvisa.ResourceManager("@py")
my_instrument = rm.open_resource("TCPIP0::127.0.0.1::hislipHCCU::INSTR")
print("Setting termination")
my_instrument.read_termination = '\n'
print("Calling *IDN?")
result = my_instrument.query("*IDN?")
print("Result:", result)
```

If I remove "@py" from the second script it runs fine, printing the following:
```
Setting termination
Calling *IDN?
Result: Keysight Technologies,hislipHCCU,0,2.32.0.0
```

pip list shows:
```
PyVISA                        1.14.1
PyVISA-py                     0.7.3.dev33+g79a39c5
```

MatthieuDartiailh, 2024-10-02:
I only tested a case in which I was connecting to an invalid IP address. Here it seems only the hslip "id" is wrong in the bad case. 

Out of curiosity what is your setup to have a Keysight hislip instrument available on the loop back ?

DareDevilDenis, 2024-10-03:
My setup is using Keysight software instrument "Hardware Configuration Control Utility" (HCCU): https://www.keysight.com/gb/en/assets/9018-04860/user-manuals/9018-04860.pdf

This is running my local PC, however the same issue happens if I run it on a remote PC.

MatthieuDartiailh, 2024-10-28:
Hi @DareDevilDenis 
I took another shot at your problem. I tested locally both the case of a bad IP and of only a bad hislip name. In my case the bad name was leading to a deadlock. After fixing that I could reproduce your issue and realised I was not correctly handling a status code. This is now fixed, could you give the patch another try ?

MatthieuDartiailh, 2024-12-17:
Yes the difference is expected because we provide more information details into what happens internally in pyvisa-py. 

I will merge when I get a chance and I am planning a release by the end of the year (hopefully).

MatthieuDartiailh, 2024-12-29:
Closed by #430

closed 2024-12-29 by MatthieuDartiailh