pyvisa/pyvisa-py #432: Raspberry Pi issues with write_raw()

Matthew-Boyd, 2024-05-31:
Writing a waveform to a UNI-T UTG932E function generator [documentation](https://storage.googleapis.com/uni-tdocuments/UTG900E_Programming_Manual.pdf) using write_raw() over USB (via pyusb) works on Windows but not from a RPi 2. There are no errors given, the name in the command is properly shown on the device (also using write_raw()) but the data is not set. Very small waveforms with only 30 points fail too. The hashes in the following code are equal on both Windows and the RPi. There are no other issues controlling the device from the RPi. Different `write_termination` values have been tried.

What could be the issue, and how can this be debugged?

```
import pyvisa
import hashlib

ADDR = 'USB0::0x6656::0x0834::AWG1222520093::INSTR'
filePath = "ThirtyPoints.bsv"
fileName = "ThirtyPoints"

rm = pyvisa.ResourceManager()
device = rm.open_resource(ADDR)

with open(filePath, mode="rb") as fh:
    # Form and hash the command
    fileNameCommand = "WARB1:Carrier {}".format(fileName)
    hash = hashlib.sha256(fileNameCommand.encode()).hexdigest()

    # Read and hash the data
    arbdata = fh.read()
    hash = hashlib.sha256(arbdata).hexdigest()

    # Write the command
    device.write_raw(fileNameCommand.encode())

    # Write the data
    device.write_raw(arbdata)
```

Output of `pyvisa-info`:

```
Machine Details:
   Platform ID:    Linux-6.1.21-v7+-armv7l-with-glibc2.31
   Processor:      

Python:
   Implementation: CPython
   Executable:     /home/mboyd/Documents/Ultrasonic/bin/python
   Version:        3.9.2
   Compiler:       GCC 10.2.1 20210110
   Architecture:   armv7l
   Build:          Mar 12 2021 04:06:34 (#default)
   Unicode:        UCS4

PyVISA Version: 1.14.1

Backends:
   ivi:
      Version: 1.14.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.7.2
      USB INSTR: Available via PyUSB (1.2.1). Backend: libusb1
      USB RAW: Available via PyUSB (1.2.1). Backend: libusb1
```

MatthieuDartiailh, 2024-05-31:
Is the behavior the same if you proceed to a single write ? 

Since you use write_raw the write_termination will not matter but you will get different end line on different OS. Could you try stripping the end of the file and adding manually `\r\n` in an OS independent manner ?

Matthew-Boyd, 2024-05-31:
I can't get it to work with a single write, not even updating the name.
```device.write_raw(fileNameCommand.encode() + arbdata)```

I've tried various line terminations as well:
```device.write_raw(fileNameCommand.encode() + "\n".encode() + arbdata + "\n".encode())```

Keeping the `write_raw` commands separate at least updates the name.

In my above example, what unspecified characters are being written after each `write_raw`, and what changes when running on Linux (RPi)? It sounds like there is no `write_termination` character being added with `write_raw`, but is the OS still adding something after each `write_raw`?

I've tried adding a `\r\n` to the end of the file (it was lacking this on the last line) but that had no effect.

Matthew-Boyd, 2024-06-03:
Using Wireshark with usbmon (Linux) and USBPcap (Windows), it can be seen that write_raw is sending all the bytes at once on Windows (using IVI backend), but it is being chunked by pyvisa-py on Linux. I've chunked the data in the Python code to match the chunk sizes (76 bytes total, but 64 bytes is payload), but that hasn't worked. Can I send the bytes all together?

I'm also wondering if the device setup is correct. Sending `fileNameCommand` has a larger header and gets a different response than the corresponding behavior using IVI on Windows.

filters for Wireshark files:
`usb.device_address==20` for Windows
`usb.device_address==28` for Pi

[image]
[captures.zip](https://github.com/user-attachments/files/15527370/captures.zip)

2xB, 2025-02-22:
@Matthew-Boyd Looking at your Wireshark observations, I believe this to be the issue I've fixed in #490 . Could you try if that fixes this issue as well?

MatthieuDartiailh, 2026-01-02:
Running Ni on windows is fine. I am just questioning the fact you send raw data without header or termination because it is very unusual.