pyvisa/pyvisa-py #443: [COM] Communication issue with [keysight 53230A ] using USB

ahmedyahia1060, 2024-07-30:
Instrument details
------------------
* Model: keysight 53230A
* Communication: usb
* Link to the documentation (if available):https://www.keysight.com/gb/en/library/manuals/programming-and-syntax-guide/53220a53230a-frequency-universalcountertimer-programmers-reference-1971001.html

Output of `pyvisa-info`
-----------------------

USB0::2391::6407::MY59370120::0::INSTR
Traceback (most recent call last):
  File "/home/user/Desktop/Build_1_Test/counter3.py", line 38, in <module>
    instrument = connect_device(resource_name)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/Desktop/Build_1_Test/counter3.py", line 13, in connect_device
    instrument = rm.open_resource(resource_name)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/dist-packages/pyvisa/highlevel.py", line 3292, in open_resource
    res.open(access_mode, open_timeout)
  File "/usr/local/lib/python3.12/dist-packages/pyvisa/resources/resource.py", line 281, in open
    self.session, status = self._resource_manager.open_bare_resource(
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/dist-packages/pyvisa/highlevel.py", line 3217, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/pyvisa_py/highlevel.py", line 167, in open
    sess = cls(session, resource_name, parsed, open_timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/pyvisa_py/sessions.py", line 323, in __init__
    self.after_parsing()
  File "/usr/lib/python3/dist-packages/pyvisa_py/usb.py", line 81, in after_parsing
    self.interface = self._intf_cls(
                     ^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/pyvisa_py/protocols/usbtmc.py", line 287, in __init__
    super(USBTMC, self).__init__(vendor, product, serial_number, **kwargs)
  File "/usr/lib/python3/dist-packages/pyvisa_py/protocols/usbtmc.py", line 199, in __init__
    raise ValueError("No device found.")
ValueError: No device found.

[CODE]:
import pyvisa

def list_devices():
    """List all connected VISA devices."""
    rm = pyvisa.ResourceManager('@py')
    devices = rm.list_resources()
    for device in devices:
        print(device)

def connect_device(resource_name):
    """Connect to the USBTMC device."""
    rm = pyvisa.ResourceManager('@py')
    instrument = rm.open_resource(resource_name)
    return instrument

def query_device(instrument, command):
    """Send a query to the device and get the response."""
    response = instrument.query(command)
    return response

def write_device(instrument, command):
    """Send a command to the device."""
    instrument.write(command)

def read_device(instrument):
    """Read the response from the device."""
    response = instrument.read()
    return response

if __name__ == "__main__":
    # List all connected VISA devices
    list_devices()
    
    # Replace 'USB0::0xXXXX::0xYYYY::INSTR' with your device's resource string
    resource_name = 'USB0::0x0957::0x1907::MY1234567::INSTR'
    
    # Connect to the device
    instrument = connect_device(resource_name)
    
    # Query the device (example: ask for identification)
    idn_response = query_device(instrument, '*IDN?')
    print(f"Device IDN: {idn_response}")
    
    # Send a command to the device (example: reset the device)
    write_device(instrument, '*RST')
    print("Device reset command sent.")
    
    # Close the connection
    instrument.close()

ahmedyahia1060, 2024-07-30:
It shows time out error : An unexpected error occurred with device USB0::2391::6407::MY59370120::0::INSTR: [Errno 110] Operation timed out

arr-ee, 2024-07-30:
Could you please check device permissions? See https://github.com/pyvisa/pyvisa/issues/758#issuecomment-2041328096, specifically “My solution” section.

ahmedyahia1060, 2024-07-30:
[image]
[image]
I have already tried this issues before but the output is still the same , I am using Ubuntu so i downloaded keysight communication expert and tried to communicate with it and it worked but when i try with pyvisa it never work with keysight appliction again

ahmedyahia1060, 2024-07-30:
I run with pyvisa debug mode get the following :
#####################################################
DEBUG:pyvisa:SerialSession was correctly imported.
DEBUG:pyvisa:USBSession and USBRawSession were correctly imported.
DEBUG:pyvisa:TCPIPSession was correctly imported.
DEBUG:pyvisa:GPIBSession was not imported No module named 'gpib'.
DEBUG:pyvisa:Created library wrapper for py
DEBUG:pyvisa:Created ResourceManager with session 8724403
('ASRL/dev/ttyS0::INSTR', 'ASRL/dev/ttyS1::INSTR', 'ASRL/dev/ttyS2::INSTR', 'ASRL/dev/ttyS3::INSTR', 'ASRL/dev/ttyS5::INSTR', 'ASRL/dev/ttyS6::INSTR', 'ASRL/dev/ttyS7::INSTR', 'ASRL/dev/ttyS8::INSTR', 'ASRL/dev/ttyS9::INSTR', 'ASRL/dev/ttyS10::INSTR', 'ASRL/dev/ttyS11::INSTR', 'ASRL/dev/ttyS12::INSTR', 'ASRL/dev/ttyS13::INSTR', 'ASRL/dev/ttyS14::INSTR', 'ASRL/dev/ttyS15::INSTR', 'ASRL/dev/ttyS16::INSTR', 'ASRL/dev/ttyS17::INSTR', 'ASRL/dev/ttyS18::INSTR', 'ASRL/dev/ttyS19::INSTR', 'ASRL/dev/ttyS20::INSTR', 'ASRL/dev/ttyS21::INSTR', 'ASRL/dev/ttyS22::INSTR', 'ASRL/dev/ttyS23::INSTR', 'ASRL/dev/ttyS24::INSTR', 'ASRL/dev/ttyS25::INSTR', 'ASRL/dev/ttyS26::INSTR', 'ASRL/dev/ttyS27::INSTR', 'ASRL/dev/ttyS28::INSTR', 'ASRL/dev/ttyS29::INSTR', 'ASRL/dev/ttyS30::INSTR', 'ASRL/dev/ttyS31::INSTR', 'ASRL/dev/ttyS4::INSTR', 'ASRL/dev/ttyUSB0::INSTR', 'USB0::11975::25344::800070020756710009::0::INSTR', 'USB0::2391::6407::MY59370120::0::INSTR')
DEBUG:pyvisa:ASRL/dev/ttyUSB0::INSTR - opening ...
DEBUG:pyvisa:ASRL/dev/ttyUSB0::INSTR - is open with session 5625419
DEBUG:pyvisa:Serial.write b'*IDN?\r\n'
DEBUG:pyvisa:ASRL/dev/ttyUSB0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
DEBUG:pyvisa:ASRL/dev/ttyUSB0::INSTR - exception while reading: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
Buffer content: bytearray(b'')
ERROR:root:Error with device ASRL/dev/ttyUSB0::INSTR: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
DEBUG:pyvisa:USB0::11975::25344::800070020756710009::0::INSTR - opening ...
DEBUG:pyvisa:USB0::11975::25344::800070020756710009::0::INSTR - is open with session 1082917
DEBUG:pyvisa:ASRL/dev/ttyUSB0::INSTR - closing
DEBUG:pyvisa:ASRL/dev/ttyUSB0::INSTR - is closed
DEBUG:pyvisa:USB0::11975::25344::800070020756710009::0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
ERROR:pyvisa:Unknown attribute 1073676342 (0x3fff0036 - VI_ATTR_SUPPRESS_END_EN)
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pyvisa_py/sessions.py", line 669, in get_attribute
    return self._get_attribute(attribute)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/pyvisa_py/usb.py", line 207, in _get_attribute
    raise UnknownAttribute(attribute)
pyvisa_py.sessions.UnknownAttribute: Unknown attribute 1073676342 (0x3fff0036 - VI_ATTR_SUPPRESS_END_EN)
DEBUG:pyvisa:USB0::2391::6407::MY59370120::0::INSTR - opening ...
DEBUG:pyvisa:USB0::2391::6407::MY59370120::0::INSTR - is open with session 4817422
DEBUG:pyvisa:USB0::11975::25344::800070020756710009::0::INSTR - closing
DEBUG:pyvisa:USB0::11975::25344::800070020756710009::0::INSTR - is closed
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pyvisa_py/protocols/usbtmc.py", line 256, in write
    return self.usb_send_ep.write(data)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/usb/core.py", line 408, in write
    return self.device.write(self, data, timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/usb/core.py", line 989, in write
    return fn(
           ^^^
  File "/usr/lib/python3/dist-packages/usb/backend/libusb1.py", line 837, in bulk_write
    return self.__write(self.lib.libusb_bulk_transfer,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/usb/backend/libusb1.py", line 938, in __write
    _check(retval)
  File "/usr/lib/python3/dist-packages/usb/backend/libusb1.py", line 602, in _check
    raise USBTimeoutError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBTimeoutError: [Errno 110] Operation timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/user/Desktop/Build_1_Test/temp_mini.py", line 23, in <module>
    idn = inst.query("*IDN?")
          ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/dist-packages/pyvisa/resources/messagebased.py", line 641, in query
    self.write(message)
  File "/usr/local/lib/python3.12/dist-packages/pyvisa/resources/messagebased.py", line 196, in write
    count = self.write_raw(message.encode(enco))
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/dist-packages/pyvisa/resources/messagebased.py", line 156, in write_raw
    return self.visalib.write(self.session, message)[0]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/pyvisa_py/highlevel.py", line 543, in write
    written, status_code = self.sessions[session].write(data)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/pyvisa_py/usb.py", line 179, in write
[... 23 lines cut ...]
devices = rm.list_resources()
print(devices)

# Dictionary to store identified devices
identified_devices = {}

# Loop through each device and try to identify it
for device in devices:
    try:
[... cut ...]
            identified_devices[device] = idn
        else:
            identified_devices[device] = "Non-GPIB/USB device, identification not attempted"
    except pyvisa.errors.VisaIOError as e:
        logging.error(f"Error with device {device}: {e}")
        identified_devices[device] = "Error during identification"

# Print the identified devices
for device, idn in identified_devices.items():
    print(f"{device}: {idn}")
#####################################################################
couldn't identify the problem yet

arr-ee, 2024-07-30:
Interesting — thank you for debug logging, it is clear you've spent time digging into this. It appears that both your USB devices encounter issues, but different ones. 

`USB0::11975::25344::800070020756710009::0::INSTR`, which appears to be an ITECH something, is hitting precisely https://github.com/pyvisa/pyvisa-py/issues/293 — an issue that's been fixed three years ago.

Keysight device appears to be timeouting on write, which is also unusual.

Could you please post the following:

1. output of `pyvisa-info` — I am specifically interested in versions;
2. debug output of cut down test script that only accesses the keysight instrument from clean state, i.e. reconnect the usb cable. I doubt the issue is in one session poisoning others, but I'd rather exclude it completely.

Thanks!

ahmedyahia1060, 2024-07-30:
the first device is working and i cancommunicate , the problem is with the agilent one 

I am using the latest build from development but befor that i was using latest offical version from pyvisa they both produce same error 
for the ouput pyvisa-info 
pyvisa-info
Machine Details:
   Platform ID:    Linux-6.8.0-39-generic-x86_64-with-glibc2.39
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /home/user/Desktop/Build_1_Test/myenv/bin/python3
   Version:        3.12.3
   Compiler:       GCC 13.2.0
   Architecture:   ('x86', 64)
   Build:          Apr 10 2024 05:33:47 (#main)
   Unicode:        UCS4

PyVISA Version: 1.14.2.dev64+g6fb569d

Backends:
   ivi:
      Version: 1.14.2.dev64+g6fb569d (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.7.2
      ASRL INSTR: Available via PySerial (3.5)
      TCPIP INSTR: Available 
         Resource discovery:
         - VXI-11: partial (psutil not installed)
         - hislip: disabled (zeroconf not installed)
      TCPIP SOCKET: Available 
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      VICP INSTR:
         Please install PyVICP to use this resource type.
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'
      GPIB INTFC:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'

ahmedyahia1060, 2024-07-30:
the second point I cant understand well , I am out of the lap now and I will try it when somone go there , 
I have tried to unplug and plug usb then try but to be honest there were always another device connected 
Thanks

ahmedyahia1060, 2024-07-31:
I reconnected the agilent device , after restarting laptop, power cycle the counter . I got the same ouput the error in the write command 
DEBUG:pyvisa:SerialSession was correctly imported.
DEBUG:pyvisa:USBSession and USBRawSession were correctly imported.
DEBUG:pyvisa:TCPIPSession was correctly imported.
DEBUG:pyvisa:GPIBSession was not imported No module named 'gpib'.
DEBUG:pyvisa:Created library wrapper for py
DEBUG:pyvisa:Created ResourceManager with session 3974861
('ASRL/dev/ttyS0::INSTR', 'ASRL/dev/ttyS1::INSTR', 'ASRL/dev/ttyS2::INSTR', 'ASRL/dev/ttyS3::INSTR', 'ASRL/dev/ttyS5::INSTR', 'ASRL/dev/ttyS6::INSTR', 'ASRL/dev/ttyS7::INSTR', 'ASRL/dev/ttyS8::INSTR', 'ASRL/dev/ttyS9::INSTR', 'ASRL/dev/ttyS10::INSTR', 'ASRL/dev/ttyS11::INSTR', 'ASRL/dev/ttyS12::INSTR', 'ASRL/dev/ttyS13::INSTR', 'ASRL/dev/ttyS14::INSTR', 'ASRL/dev/ttyS15::INSTR', 'ASRL/dev/ttyS16::INSTR', 'ASRL/dev/ttyS17::INSTR', 'ASRL/dev/ttyS18::INSTR', 'ASRL/dev/ttyS19::INSTR', 'ASRL/dev/ttyS20::INSTR', 'ASRL/dev/ttyS21::INSTR', 'ASRL/dev/ttyS22::INSTR', 'ASRL/dev/ttyS23::INSTR', 'ASRL/dev/ttyS24::INSTR', 'ASRL/dev/ttyS25::INSTR', 'ASRL/dev/ttyS26::INSTR', 'ASRL/dev/ttyS27::INSTR', 'ASRL/dev/ttyS28::INSTR', 'ASRL/dev/ttyS29::INSTR', 'ASRL/dev/ttyS30::INSTR', 'ASRL/dev/ttyS31::INSTR', 'ASRL/dev/ttyS4::INSTR', 'USB0::2391::6407::MY59370120::0::INSTR')
DEBUG:pyvisa:USB0::2391::6407::MY59370120::0::INSTR - opening ...
DEBUG:pyvisa:USB0::2391::6407::MY59370120::0::INSTR - is open with session 6392229
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pyvisa_py/protocols/usbtmc.py", line 256, in write
    return self.usb_send_ep.write(data)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/usb/core.py", line 408, in write
    return self.device.write(self, data, timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/usb/core.py", line 989, in write
    return fn(
           ^^^
  File "/usr/lib/python3/dist-packages/usb/backend/libusb1.py", line 837, in bulk_write
    return self.__write(self.lib.libusb_bulk_transfer,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/usb/backend/libusb1.py", line 938, in __write
    _check(retval)
  File "/usr/lib/python3/dist-packages/usb/backend/libusb1.py", line 602, in _check
    raise USBTimeoutError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBTimeoutError: [Errno 110] Operation timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/user/Desktop/Build_1_Test/temp_mini.py", line 23, in <module>
    idn = inst.query("*IDN?")
          ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/dist-packages/pyvisa/resources/messagebased.py", line 641, in query
    self.write(message)
  File "/usr/local/lib/python3.12/dist-packages/pyvisa/resources/messagebased.py", line 196, in write
    count = self.write_raw(message.encode(enco))
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/dist-packages/pyvisa/resources/messagebased.py", line 156, in write_raw
    return self.visalib.write(self.session, message)[0]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/pyvisa_py/highlevel.py", line 543, in write
    written, status_code = self.sessions[session].write(data)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/pyvisa_py/usb.py", line 179, in write
    count = self.interface.write(data)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/pyvisa_py/protocols/usbtmc.py", line 436, in write
    bytes_sent += raw_write(data)
                  ^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/pyvisa_py/protocols/usbtmc.py", line 258, in write
    raise ValueError(str(e))
ValueError: [Errno 110] Operation timed out
DEBUG:pyvisa:Closing ResourceManager (session: 3974861)
DEBUG:pyvisa:USB0::2391::6407::MY59370120::0::INSTR - closing
DEBUG:pyvisa:USB0::2391::6407::MY59370120::0::INSTR - is closed

ahmedyahia1060, 2024-07-31:
After using the pyvisa code I cant back control on device using keysight connection expert 
[image]

ahmedyahia1060, 2024-07-31:
I tried this code :import pyvisa

def initialize_resource_manager():
    try:
        rm = pyvisa.ResourceManager()
        return rm
    except Exception as e:
        print(f"Failed to initialize resource manager: {e}")
        exit(1)

def identify_device(rm, visa_resource):
    try:
        inst = rm.open_resource(visa_resource)
        inst.timeout = 5000  # Set a longer timeout (in milliseconds) if needed
        idn = inst.query("*IDN?")
        inst.close()
        return idn
    except pyvisa.errors.VisaIOError as e:
        print(f"VISA IO Error with device {visa_resource}: {e}")
        return None
    except Exception as e:
        print(f"Unexpected error with device {visa_resource}: {e}")
        return None

def measure_frequency(rm, visa_resource):
    try:
        inst = rm.open_resource(visa_resource)
        inst.timeout = 5000  # Set a longer timeout (in milliseconds) if needed

        # Setup the device for frequency measurement
        inst.write("CONF:FREQ")
        
        # Initiate measurement
        frequency = inst.query("READ?")
        
        inst.close()
        return frequency
    except pyvisa.errors.VisaIOError as e:
        print(f"VISA IO Error with device {visa_resource}: {e}")
        return None
    except Exception as e:
        print(f"Unexpected error with device {visa_resource}: {e}")
        return None

def main():
    rm = initialize_resource_manager()

    # Define the specific VISA resource
    visa_resource = "USB0::0x0957::0x1907::MY59370120::INSTR"
    
    # Identify the device
    idn = identify_device(rm, visa_resource)
    if idn:
        print(f"Identification response for {visa_resource}: {idn}")
    
    # Measure frequency
    frequency = measure_frequency(rm, visa_resource)
    if frequency:
        print(f"Measured frequency for {visa_resource}: {frequency} Hz")

if __name__ == "__main__":
    main()
###################################################################
it worked on windows completely fine but on linux it doesnt work , same error appear is there anythin you have in mind ?

arr-ee, 2024-08-05:
So,

> the second point I cant understand well , I am out of the lap now and I will try it when somone go there, I have tried to unplug and plug usb then try but to be honest there were always another device connected

I was asking you to change your test code to only query device in question — you have done so in the last snippet you posted, thanks!

> After using the pyvisa code I cant back control on device using keysight connection expert

This is to be expected — `pyvisa-py` detaches kernel USBTMC driver from the device since it implements USBTMC in userspace => after attempting to talk to the device via `pyvisa-py` (not via pyvisa + vendor VISA driver) you need to re-attach the device to re-enable kernel USBTMC driver.

> https://github.com/pyvisa/pyvisa-py/issues/443#issuecomment-2259895806
> File "/usr/lib/python3/dist-packages/pyvisa_py/protocols/usbtmc.py", line 436, in write
bytes_sent += raw_write(data)
> ^^^^^^^^^^^^^^^
> File "/usr/lib/python3/dist-packages/pyvisa_py/protocols/usbtmc.py", line 258, in write
raise ValueError(str(e))
> ValueError: [Errno 110] Operation timed out

This does not make sense to me. Line 258 is not in `write` in pyvisa-py 0.7.2 (latest release) nor in `main`: https://github.com/pyvisa/pyvisa-py/blob/32ca1d5848e7a1a0a91f4c5ca907c81f7e9b5c6c/pyvisa_py/protocols/usbtmc.py#L258

Another thing that does not make sense to me is that we're getting a timeout error from the usb library here, yet from your previous message showing `pyvisa-info` output:

> USB INSTR:
> Please install PyUSB to use this resource type.
> No module named 'usb'

Given that same code works on windows but not on linux, and given the oddness I've outlined above, I suggest we try to reset (figuratively speaking) and try the following:

1. Add `pyvisa.log_to_screen()` to the beginning of the script you last posted.
2. Please post output of `pyvisa-info` and from running the script you posted from both machines.

I understand that it might feel like we're walking in circles, but what I am seeing so far does not make sense.