pyvisa/pyvisa-py #461: [COM] Communication issue with SR770 using GPIB

HaydenS3, 2024-11-03:
#### Instrument details
* Model: SR770
* Communication: GPIB
* Link to the documentation (if available): https://www.thinksrs.com/downloads/pdfs/manuals/SR770m.pdf

#### The Error
I'm trying to communicate with an SR770 over GPIB using PyVISA-py. I've successfully communicated with the device on a Windows 10 pc, but attempting to do so on a Windows 11 pc results in the error below. To my knowledge, the only effective difference between the systems is the Windows version.
 
```
ValueError: gpib_ctypes is installed but could not locate the gpib library.
Please manually load it using:
  gpib_ctypes.gpib.gpib._load_lib(filename)
```

#### Debugging
1. Tried manually loading the library using `gpib._load_lib('C:\\Windows\\SysWOW64\\gpib-32.dll')`
2. Confirmed that NI 488.2 2024 Q3 is installed correctly as I'm able to communicate with the instrument through NI Max.
3. Copied `gpib-32.dll` into System32.

Not sure if this is the right place to post this, but any advice or guidance would be greatly appreciated!

MatthieuDartiailh, 2024-11-05:
To be honest on Windows with an installed implementation of VISA you have no reason to go through pyvisa-py. Just use the default backend (IVI) which will use NI VISA in your case.

Apart from that, are you using a 32 bits Python ? If not you should directly the library from System32 gpib-32.dll. 
You mention that manually loading fails. Could you detail how it fails exactly ?

HaydenS3, 2024-11-12:
I switched to the default backend and all is working. Thank you for the suggestion! I did need to point `pyvisa.ResourceManager` directly to `visa64.dll` for some reason.

To answer your question, I'm using 64 bit Python and the manually loading failed in the sense that the ValueError above was thrown. The `_load_lib` function never threw its own error.

This issue can be closed if there aren't any further questions.

MatthieuDartiailh, 2024-11-12:
Happy you got things to work.

closed 2024-11-12 by MatthieuDartiailh

mattjaffe42, 2024-12-12:
@HaydenS3 could you elaborate on how you got this to work? I'm seeing something really similar and can't quite figure it out. Thanks in advance.

mattjaffe42, 2024-12-12:
posting on this thread is magical: i uninstalled and re-installed NI-VISA and the NI 488.2 driver, and then it worked.

It's strange because `pyvisa-info` still seems to think it doesn't know about GPIB, and using `gpib._load_lib()` seems to fail (gives `False`), but `pyvisa` communication seems to just work if I open an instrument and send commands.

i'm typing this on a different computer so am not posting messages or anything, but could later if that would be of interest.