pyvisa/pyvisa-py #463: Rigol DS1102Z-E USBTMC read regression

ius, 2024-11-06:
Recent changes to the USBTMC read implementation break support for Rigol DS1102Z-E (and possibly other Rigol devices).

PR #449 changed the USBTMC response read implementation to assume a _transfer_ is complete once a short packet (i.e. `<= usb_recv_ep.wMaxPacketSize`) is received based on the following excerpt from the USBTMC spec (§3.3):

> The device must always terminate a Bulk-IN transfer by sending a short packet. The short packet
may be zero-length or non zero-length.

Unfortunately, it looks like Rigol does not adhere to this part of the spec - no such packets are sent by my Rigol DS1102Z-E. Which also implies pyvisa never manages to complete a _transfer_ if the `transfer` is exactly `usb_recv_ep.wMaxPacketSize` bytes.

Trivial reproducer:
```
#!/usr/bin/env python3
import pyvisa

rm = pyvisa.ResourceManager('@py')
scope = rm.open_resource(rm.list_resources()[0])

scope.write('WAV:SOUR CH1')
scope.write('WAV:STAR 1')
scope.write('WAV:STOP 128')
data = scope.query_binary_values('WAV:DATA?', datatype='B', expect_termination=False)
print('received:', len(data))
```

Pyivsa should hang, not even responding to `SIGINT`.

(As it happens the `*IDN?` response for my device is also exactly `usb_recv_ep.wMaxPacketSize` (64) bytes, but that might not be the case for other models)

Note: looks to me @arr-ee might have been experiencing a similar issue on his Siglent device when testing the PR: https://github.com/pyvisa/pyvisa-py/pull/449#issuecomment-2364103093 - alas, my device is a Rigol and it's on the latest firmware available (00.06.03.SP2).

ius, 2024-11-06:
Forgot to mention the original PR author for notification purposes: @hcoffin-vecatom

hcoffin-vecatom, 2024-11-06:
Hmm that is unfortunate. Do you happen to know if the usb_recv_ep.wMaxPacketSize long transfer ends in a newline character? If it is, I think the only ways to fix this issue are either to remove the changes from pr #449 or to implement some kind of special logic for devices that don't adhere to the standard (e.g., check for termination characters at the usb read level), which is annoying at best.

ius, 2024-11-06:
If I request `>=wMaxPacketSize` bytes (using the reproducer, with #449 reverted) only the last _transaction_ (and thus also the full _transfer_) ends on a newline (not sure whether that's part of the protocol of the `WAV:DATA` response?).

The old read behavior (which relied on multiple requests, checking for the EOM bit to be set in the transaction) works for my Rigol, but if I understand correctly based on reading #449 it seems like at least some Siglent devices fail to set it?

Then I don't really know how to fix this, other than to consider introducing a Siglent (no EOM case) and/or Rigol (no short packet as last transaction) quirk.

hcoffin-vecatom, 2024-11-06:
As far as I can tell from the examples in the DS1000Z-E-ProgrammingGuide (see screenshot at bottom of this reply) \n (0x0A) is the termination character used by the device. I think it is unlikely tofix things, but if you haven't already, try setting the termination character for pyvisa using
```
scope.read_termination = '\n'
```
and use ```expect_termination=True``` instead of False. If the termination character isn't set properly then I think that could also be causing your program to hang since ```Session._read()``` (pyvisa-py/sessions.py) will return ```constants.StatusCode.success_max_count_read``` and ```MessageBasedResource._read_raw``` (pyvisa/resources/messagebased.py) would loop forever.

I also want to make sure, when using the old read behavior, are you getting the correct data back, or is some part of it being misinterpreted as a header for the Bulk In message? If you are getting the correct data with the old read behavior then your scope is behaving in a quirky way that is different from the quirkiness of either mine or @arr-ee 's instruments.

[image]

Jimmyvandenbergh, 2024-12-07:
I think maybe this issue is caused by the USBTMC buffer that is requested at the first read, that equals the exact amount of ‘wMaxPacketSize’ as during testing with a Rigol I have also seen unexpected behavior. (Asking for 52bytes, excluding USBTMC header would result in hang when new bytes are asked, as there where non left to read anymore). Now we ask for the amount of specified bytes by user / default pyvisa buffer size. 

can you checkout the or #470 as this would probably solve your issue.

ngkz-ierae, 2025-08-25:
#470 doesnt fix this issue.
everything >=0.7.2 including latest main (533914998f9e4ade925fd0f803b641776f43f691) doesnt work.
i had to downgrade to pyvisa-py==0.7.1 after wasting hours.
 im using MSO1104Z and PyVISA 1.15.0