pyvisa/pyvisa-py #474: unable to list_resources() on Mac: TypeError

mattjaffe42, 2024-12-06:
I have a Siglent SDS1104X-E that I am able to connect to and read data from using a more manual approach from [here](https://github.com/FPiorski/SDS1104X-E_socket_waveform/blob/master/main.py). However, I am unable to connect using pyvisa with the pyvisa-py backend:

```
import pyvisa
rm = pyvisa.ResourceManager('@py')
rm.list_resources()
```

I following error:
```
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
Cell In[60], line 3
      1 import pyvisa
      2 rm = pyvisa.ResourceManager('@py')
----> 3 rm.list_resources()

File [~/opt/anaconda3/envs/jaffe-lab-0/lib/python3.13/site-packages/pyvisa/highlevel.py:3137](http://localhost:8888/lab/tree/OneDrive%20-%20Montana%20State%20University/Documents%20-%20Jaffe%20Lab/matt/dailys/24/12/~/opt/anaconda3/envs/jaffe-lab-0/lib/python3.13/site-packages/pyvisa/highlevel.py#line=3136), in ResourceManager.list_resources(self, query)
   3089 def list_resources(self, query: str = "?*::INSTR") -> Tuple[str, ...]:
   3090     r"""Return a tuple of all connected devices matching query.
   3091 
   3092     Notes
   (...)
   3134 
   3135     """
-> 3137     return self.visalib.list_resources(self.session, query)

File [~/opt/anaconda3/envs/jaffe-lab-0/lib/python3.13/site-packages/pyvisa_py/highlevel.py:487](http://localhost:8888/lab/tree/OneDrive%20-%20Montana%20State%20University/Documents%20-%20Jaffe%20Lab/matt/dailys/24/12/~/opt/anaconda3/envs/jaffe-lab-0/lib/python3.13/site-packages/pyvisa_py/highlevel.py#line=486), in PyVisaLibrary.list_resources(self, session, query)
    485 resources: List[str] = []
    486 for key, st in sessions.Session.iter_valid_session_classes():
--> 487     resources += st.list_resources()
    489 return rname.filter(resources, query)

File [~/opt/anaconda3/envs/jaffe-lab-0/lib/python3.13/site-packages/pyvisa_py/tcpip.py:90](http://localhost:8888/lab/tree/OneDrive%20-%20Montana%20State%20University/Documents%20-%20Jaffe%20Lab/matt/dailys/24/12/~/opt/anaconda3/envs/jaffe-lab-0/lib/python3.13/site-packages/pyvisa_py/tcpip.py#line=89), in TCPIPInstrSession.list_resources(wait_time)
     88 @staticmethod
     89 def list_resources(wait_time=1.0) -> List[str]:
---> 90     return TCPIPInstrVxi11.list_resources() + TCPIPInstrHiSLIP.list_resources()

File [~/opt/anaconda3/envs/jaffe-lab-0/lib/python3.13/site-packages/pyvisa_py/tcpip.py:404](http://localhost:8888/lab/tree/OneDrive%20-%20Montana%20State%20University/Documents%20-%20Jaffe%20Lab/matt/dailys/24/12/~/opt/anaconda3/envs/jaffe-lab-0/lib/python3.13/site-packages/pyvisa_py/tcpip.py#line=403), in TCPIPInstrVxi11.list_resources()
    402                 addr = snic.address
    403                 mask = snic.netmask
--> 404                 network = ipaddress.IPv4Network(addr + "[/](http://localhost:8888/)" + mask, strict=False)
    405                 broadcast_addr.append(str(network.broadcast_address))
    406 else:
    407     # If psutil unavailable fallback to default interface

TypeError: can only concatenate str (not "NoneType") to str
```

Here is the output of `pyvisa-info`:

```
Machine Details:
   Platform ID:    macOS-14.4-x86_64-i386-64bit-Mach-O
   Processor:      i386

Python:
   Implementation: CPython
   Executable:     /Users/jaffe_lab/opt/anaconda3/envs/jaffe-lab-0/bin/python3.13
   Version:        3.13.0
   Compiler:       Clang 17.0.6 
   Architecture:   ('x86', 64)
   Build:          Oct 17 2024 12:38:20 (#main)
   Unicode:        UCS4

PyVISA Version: 1.14.1

Backends:
   ivi:
      Version: 1.14.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.7.2
      ASRL INSTR: Available via PySerial (3.5)
      USB INSTR: Available via PyUSB (1.2.1). Backend: libusb1
      USB RAW: Available via PyUSB (1.2.1). Backend: libusb1
      TCPIP INSTR: Available 
         Resource discovery:
         - VXI-11: ok
         - hislip: disabled (zeroconf not installed)
      TCPIP SOCKET: Available 
      VICP INSTR:
         Please install PyVICP to use this resource type.
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'
      GPIB INTFC:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'
```

Any suggestions? I'm not sure how to proceed. Thanks in advance.

mattjaffe42, 2024-12-09:
i didn't get this error on a windows system, and `pyvisa` seems to be able to read and write to the device fine using `query('*IDN?')`. So maybe it's just some `None` /  `str` handling that's different on Mac when `list_resources()` doesn't find anything (even if connection by explicit IP address is possible).

MatthieuDartiailh, 2024-12-16:
Looking at psutil docs, it appears that the `.netmask` attribute can be None. we should probably skip such networks when discovering devices. I will make a PR, but I won't be able to test. Would you be able to test on your end ?

closed 2024-12-29 by MatthieuDartiailh