pyvisa/pyvisa-py #482: [COM] Communication issue with OWON DGE2000 using USB

mriscoc, 2025-01-15:
Instrument details
------------------
* Model: DGE2000 series
* Communication: USB
* Link to the documentation: https://www.owon.com.hk/products_owon_dge2000_series_arbitrary_waveform_generator

Output of `pyvisa-info`
-----------------------
```
Machine Details:
   Platform ID:    Windows-11-10.0.22631-SP0
   Processor:      Intel64 Family 6 Model 158 Stepping 13, GenuineIntel

Python:
   Implementation: CPython
   Executable:     W:\Python\dge2070\.venv\Scripts\python.exe
   Version:        3.13.1
   Compiler:       MSC v.1942 64 bit (AMD64)
   Architecture:   AMD64
   Build:          Dec  3 2024 19:06:28 (#tags/v3.13.1:0671451)
   Unicode:        UCS4

PyVISA Version: 1.14.1

Backends:
   ivi:
      Version: 1.14.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.7.3.dev45+g2937877
      ASRL INSTR: Available via PySerial (3.5)
      USB INSTR: Available via PyUSB (1.3.1). Backend: libusb0
      USB RAW: Available via PyUSB (1.3.1). Backend: libusb0
      TCPIP INSTR: Available
         Resource discovery:
         - VXI-11: ok
         - hislip: ok
      TCPIP SOCKET: Available
      VICP INSTR:
         Please install PyVICP to use this resource type.
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'
      GPIB INTFC:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'
```
Using this python script to look for devices:
```python
import usb.core
for dev in usb.core.find(find_all=True):
    print(dev)
```
Gives this: (Serial number replaced by XXXXXXXX)
```
DEVICE ID 5345:1234 on Bus 000 Address 001 =================
 bLength                :   0x12 (18 bytes)
 bDescriptorType        :    0x1 Device
 bcdUSB                 :  0x200 USB 2.0
 bDeviceClass           :    0x3
 bDeviceSubClass        :    0x0
 bDeviceProtocol        :    0x0
 bMaxPacketSize0        :   0x40 (64 bytes)
 idVendor               : 0x5345
 idProduct              : 0x1234
 bcdDevice              :  0x200 Device 2.0
 iManufacturer          :    0x1 System CPU
 iProduct               :    0x2 generator
 iSerialNumber          :    0x3 XXXXXXXX
 bNumConfigurations     :    0x1
  CONFIGURATION 1: 100 mA ==================================
   bLength              :    0x9 (9 bytes)
   bDescriptorType      :    0x2 Configuration
   wTotalLength         :   0x29 (41 bytes)
   bNumInterfaces       :    0x1
   bConfigurationValue  :    0x1
   iConfiguration       :    0x0
   bmAttributes         :   0xc0 Self Powered
   bMaxPower            :   0x32 (100 mA)
    INTERFACE 0: Human Interface Device ====================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x0
     bAlternateSetting  :    0x0
     bNumEndpoints      :    0x2
     bInterfaceClass    :    0x3 Human Interface Device
     bInterfaceSubClass :    0x0
     bInterfaceProtocol :    0x0
     iInterface         :    0x0
      ENDPOINT 0x81: Bulk IN ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x81 IN
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :  0x200 (512 bytes)
       bInterval        :    0x1
      ENDPOINT 0x1: Bulk OUT ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :    0x1 OUT
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :  0x200 (512 bytes)
       bInterval        :    0x1
```
Now using `usb` module to get information about the specific device:
```python
import usb.core
import usb.util

USB_VENDOR_ID = 0x5345
USB_PRODUCT_ID = 0x1234
DGE = ""

# Find the device
device = usb.core.find(idVendor=USB_VENDOR_ID, idProduct=USB_PRODUCT_ID)

if device is None:
    raise ValueError('Device not found')

# Get serial number string descriptor (index 3)
serial_number = usb.util.get_string(device, device.iSerialNumber)
print(f"Found USB with serial: {serial_number}")
DGE = f"USB::{hex(USB_VENDOR_ID)}::{hex(USB_PRODUCT_ID)}::{serial_number}::INSTR"
print(f"VISA resource string: {DGE}")
```
Gives this result: (Serial number replaced by XXXXXXXX)
```
Found USB with serial: XXXXXXXX
VISA resource string: USB::0x5345::0x1234::XXXXXXXX::INSTR
```
Using pyvisa to gain access to the device using the previously constructed resource chain:
``` python
import logging
logging.basicConfig(level=logging.DEBUG)

import pyvisa

# Initialize the VISA resource manager
rm = pyvisa.ResourceManager("@py")
instruments = rm.list_resources()
print("Available instruments:", instruments)

try:
  DGE2070 = rm.open_resource(DGE)
  DGE2070.read_termination = DGE2070.write_termination = '\n'
  idn_response = DGE2070.query('*IDN?')
# OWON,DGE2070,24240368,SCPI:99.0 FV:V1.0.3.3.100
  DGE2070.close()
  print(f"Instrument Identification: {idn_response}")
except Exception as e:
  print("Error pyvisa:", e)
```
Gives this error:
```
DEBUG:pyvisa:SerialSession was correctly imported.
DEBUG:pyvisa:USBSession and USBRawSession were correctly imported.
DEBUG:pyvisa:TCPIPSession was correctly imported.
DEBUG:pyvisa:GPIBSession was not imported No module named 'gpib'.
DEBUG:pyvisa:Created library wrapper for py
DEBUG:pyvisa:Created ResourceManager with session 8461463
DEBUG:asyncio:Using proactor: IocpProactor
Available instruments: ('ASRL1::INSTR',)
DEBUG:pyvisa:USB0::0x5345::0x1234::XXXXXXXX::0::INSTR - opening ...
Error pyvisa: No device found.
DEBUG:pyvisa:Closing ResourceManager (session: 8461463)
```

arr-ee, 2025-01-15:
I assume you are constructing the connection string manually because `list_resources` does not return the device.

The reason for that might be that, according to output presented, currently device enumerates as a HID device, not USBTMC. 

Manual suggests that you can change USB device mode in device settings — have you tried that?

mriscoc, 2025-01-15:
Thanks for your answer.  

For reference the relevant section in the manual is this:
[image]
I was trying to use the device in PC mode due to be able to use also the Waveform editor software but seems that in PC mode the protocol is "internal".  
  
After change the **USB Dev** to `USB TMC` the computer enumerates the generator as a **Unknown device** without a feasible driver. Using Zadig to force libusbK driver gives:
```
DEVICE ID 5345:1235 on Bus 000 Address 255 =================
 bLength                :   0x12 (18 bytes)
 bDescriptorType        :    0x1 Device
 bcdUSB                 :  0x200 USB 2.0
 bDeviceClass           :    0x0 Specified at interface
 bDeviceSubClass        :    0x0
 bDeviceProtocol        :    0x0
 bMaxPacketSize0        :   0x40 (64 bytes)
 idVendor               : 0x5345
 idProduct              : 0x1235
 bcdDevice              :  0x100 Device 1.0
 iManufacturer          :    0x1 System CPU
 iProduct               :    0x2 generator
 iSerialNumber          :    0x3 24240368
 bNumConfigurations     :    0x1
  CONFIGURATION 1: 500 mA ==================================
   bLength              :    0x9 (9 bytes)
   bDescriptorType      :    0x2 Configuration
   wTotalLength         :   0x20 (32 bytes)
   bNumInterfaces       :    0x1
   bConfigurationValue  :    0x1
   iConfiguration       :    0x5 CDC Abstract Control Model (ACM)
   bmAttributes         :   0xc0 Self Powered
   bMaxPower            :   0xfa (500 mA)
    INTERFACE 0: Application Specific ======================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x0
     bAlternateSetting  :    0x0
     bNumEndpoints      :    0x2
     bInterfaceClass    :   0xfe Application Specific
     bInterfaceSubClass :    0x3
     bInterfaceProtocol :    0x1
     iInterface         :    0x4 CDC ACM config
      ENDPOINT 0x81: Bulk IN ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x81 IN
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :  0x200 (512 bytes)
       bInterval        :    0x1
      ENDPOINT 0x1: Bulk OUT ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :    0x1 OUT
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :  0x200 (512 bytes)
       bInterval        :    0x1
```
So now using this Python code:
```python
import logging
logging.basicConfig(level=logging.DEBUG)

import pyvisa

USB_VENDOR_ID = 0x5345
USB_PRODUCT_ID = 0x1235

# Initialize the VISA resource manager
rm = pyvisa.ResourceManager("@py")
instruments = rm.list_resources(f"USB?::{str(USB_VENDOR_ID)}::{str(USB_PRODUCT_ID)}::*")
print("Available instruments:", instruments)
if len(instruments) > 0:
  DGE = instruments[0]
  print(f"Using instrument: {DGE}")
else:
  print("No instruments found")
  exit()

try:
  DGE2070 = rm.open_resource(DGE, open_timeout=10000)
  DGE2070.read_termination = DGE2070.write_termination = '\n'
  idn_response = DGE2070.query('*IDN?')
  DGE2070.close()
  print(f"Instrument Identification: {idn_response}")
except Exception as e:
  print("Error pyvisa:", e)
```
Gives this result:
```
DEBUG:pyvisa:SerialSession was correctly imported.
DEBUG:pyvisa:USBSession and USBRawSession were correctly imported.
DEBUG:pyvisa:TCPIPSession was correctly imported.
DEBUG:pyvisa:GPIBSession was not imported No module named 'gpib'.
DEBUG:pyvisa:Created library wrapper for py
DEBUG:pyvisa:Created ResourceManager with session 3621679
DEBUG:asyncio:Using proactor: IocpProactor
Available instruments: ('USB0::21317::4661::24240368::0::INSTR',)
Using instrument: USB0::21317::4661::24240368::0::INSTR
DEBUG:pyvisa:USB0::21317::4661::24240368::0::INSTR - opening ...
Error pyvisa: [Errno None] b'libusb0-dll:err [control_msg] sending control message failed, win error: Se agot\xf3 el tiempo de espera del sem\xe1foro.\r\n\n'
DEBUG:pyvisa:Closing ResourceManager (session: 3621679)
```

mriscoc, 2025-01-15:
If I use the app `WinUsbTmc` from https://github.com/xyphro/WinUsbTmc:
```shell
>  .\WinUsbTmc.exe /R 0 "*IDN?"
OWON,DGE2070,XXXXXXXX,SCPI:99.0 FV:V1.0.3.3.100
```

mriscoc, 2025-01-19:
Any idea how to solve it?