pyvisa/pyvisa-py #495: [COM] Communication issue with Tektronix 2461 Source meter using USB/TCPIP

KendallIsWelsh, 2025-03-27:
Hello, I'm trying to run a test script from a raspberry pi for long term testing of power consumption. I've run this script using my windows surface using the NI backend with no issues. Running from the pi I seem to always get an empty list from list_resources() I've used both TCPIP and USB with the same results. I'm pretty new to python and raspberry pi in general, so any help would be greatly appreciated.

Instrument details
------------------
* Model: Tektronix 2461 Source Measurement Unit
* Communication: TCPIP & USB
* Link to the documentation (if available): https://www.tek.com/en/datasheet/2461-graphical-source-measure-unit

Output of `pyvisa-info`
-----------------------

Machine Details:
   Platform ID:    Linux-6.6.74+rpt-rpi-v8-aarch64-with-glibc2.36
   Processor:      

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.11.2
   Compiler:       GCC 12.2.0
   Bits:           64bit
   Build:          Nov 30 2024 21:22:50 (#main)
   Unicode:        UCS4

PyVISA Version: 1.11.3

Backends:
   ivi:
      Version: 1.11.3 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.5.1
      ASRL INSTR: Available via PySerial (3.5)
      USB INSTR: Available via PyUSB (1.2.1-2). Backend: libusb1
      USB RAW: Available via PyUSB (1.2.1-2). Backend: libusb1
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
         No module named 'gpib'

Here is the output of lsusb:
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 05e6:2461 Keithley Instruments Source Measure Unit
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

And for reference if it is needed here is the program I am trying to run:

import pyvisa
import sys
import csv
from datetime import *
import time
rm = pyvisa.ResourceManager()
count = 0

#Loop for trying to find connected instrument,set maxcount to desired
#of time. Every ~1000 = ~1 second, recommend 60,000 for about a minute wait.
#Program will stop executing if timeout occurs
max_count = 60000

print("Detecting instrument...")
while not rm.list_resources(): 
    
    if (count < max_count):
        count += 1
    else:
        print("Timeout, no instrument detected")
        sys.exit()

       
print("Resources detected\n{}\n".format(rm.list_resources()))
resources = rm.list_resources()

smu=rm.open_resource(''+resources[0])

results = open("power_consumption_long.csv","w",)
wr=csv.writer(results, dialect='excel')
wr.writerow(["Date and Time", "Current(mA)","Current Units",'Voltage','Voltage Units']) 
results.close()
current_date = date.strftime(datetime.now(),'%Y %m %d %H %M %S')
smu_time = current_date.split(" ")
smu.write(f'SYST:TIME {smu_time[0]}, {smu_time[1]},{smu_time[2]},{smu_time[3]},{smu_time[4]},{smu_time[5]}')
#Set up time for test <weekday> <month> <day> <hour>:<minute>:<second> <year>
# Example: 'Fri Jan 10 16:00:00 2025'
end_test_requested = 'Tue Mar 18 10:30:00 2025'

end_test = end_test_requested.split()
end_test_time = end_test[3].replace(':','')

current_date_smu = smu.query('SYSTem:TIME? 1')
current_date_split = current_date_smu.split()
current_time = current_date_split[3].replace(':','')

smu.write('ROUTe:TERMinals FRONt')
smu.write(':SOURce:FUNCtion VOLTage')
smu.write(':SOURce:VOLTage 3.7')
smu.write('SOURCe:VOLTage:ILIMit 1')
smu.write(':SENSe:CURRent:NPLCycles 10')
smu.write(':OUTPut:STATe ON')

#Trigger stuff, will continue looping until *TRG is sent to instrument
smu.write('TRIG:LOAD "LOOPUNTILEVENT", COMM, 100')
smu.write(':INIT:IMM')

while float(end_test[2]) >= float(current_date_split[2]) and float(end_test_time) > float(current_time):
    test_count = 0

     
        
    while (test_count < 10):
        time.sleep(20)
        results = open("power_consumption_long.csv","a",)
        wr=csv.writer(results, dialect='excel')
        measurement = smu.query\
               (':FETC? "defbuffer1", DATE,TIME,FRAC,READ, UNIT, SOUR, SOURUNIT')#,DATE,FORM,FRAC,READ,REL,SEC,SOUR, SOURFORM,SOURSTAT, SOURUNIT, STAT, TIME, TST,UNIT')
        measurements=measurement.split(',')
        wr.writerow(measurements)    
        test_count += 1
        print(measurement)
    else:
        results.close()
        current_date_smu = smu.query('SYSTem:TIME? 1')
        current_date_split = current_date_smu.split()
        current_time = current_date_split[3].replace(':','')
        time.sleep(10)
        
else:
    
    smu.write('*TRG')   
    smu.close() 
    print("Test is Complete")

MatthieuDartiailh, 2025-03-30:
Couple of questions:
- even though you cannot list the resource did you try to connect to it ? If yes, what happens then ? Otherwise could you try ?
- did you follow the documentation regarding adding the proper authorizations to access usb devices ?

Jimmyvandenbergh, 2025-05-13:
Did you switch the driver backend from 'ni-visa' to 'py' also. As py will not detect/ use a ni-visa device.